openai/mujoco-py #390: render mode="human" works, but mode="rgb_array" gives GLEW initialization error

henrycharlesworth, 2019-04-17:
**Describe the bug**
When I run `env.render(mode="human")` everything works as it should. However, if instead I run `env.render(mode="rgb_array")` I get the error "ERROR: GLEW initalization error: Missing GL version". I am aware of the issue [here](https://github.com/openai/mujoco-py/issues/268), and I have added export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so (actually before doing this, I don't think mode="human" worked either), however it doesn't help fix this.

Strangely, if I first call `env.render(mode="human")` I am able to call `env.render(mode="rgb_array")` afterwards without getting this error. So, for example, the following code works:

`from gym.envs.mujoco.half_cheetah_v3 import HalfCheetahEnv
import numpy as np

env = HalfCheetahEnv()

test = env.reset()
env.render(mode="human")

for i in range(10000):
    a,b,c,d = env.step(np.random.randn(6).clip(-1,1))
    img=env.render(mode="rgb_array")`

But if I miss out the initial `env.render(mode="human")` I get the error.

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04.2 LTS
 - Python Version 3.5.2
 - Mujoco Version 2.0
 - mujoco-py version 2.0.2.0

**Environment**
 - output of: `echo $LD_LIBRARY_PATH` - /home/henry/.mujoco/mujoco200/bin
 - output of: `echo $HOME` /home/henry
 - output of: `echo $USER` - henry

jendelel, 2019-08-27:
As someone described in a thread for gym. If you render once as 'human' it works afterwards. After trying that, I figured a workaround:
Add this line at the beginning of your script:
```
from mujoco_py import GlfwContext
GlfwContext(offscreen=True)  # Create a window to init GLFW.
```

That solves the issue for me.

khedd, 2020-01-04:
Using mujoco_py==1.50.1.1, I had the same problem, but in this version, GlfwContext was not being exported. Simply exporting it and following the workaround proposed by @jendelel solved the issue.

zoharri, 2020-03-20:
I am having the exact same problem, but I am using a windows machine.
Do you have an idea how to solve it on windows?

yijiew, 2020-06-25:
@zoharri you can use the same method @jendelel and @khedd mentioned to fix the issue on Windows.

xixinzhang, 2021-01-31:
```
mujoco_py.MjRenderContextOffscreen(self.sim, -1)
```
In mujoco_env.py change -1 to be 0 works for me