openai/mujoco-py #391: RuntimeError: Failed to set buffer size when trying to take video record or screenshot

HiPPoTUS, 2019-04-17:
> Creating window glfw
Buffer size too big
Traceback (most recent call last):
  File "model_test.py", line 22, in <module>
    env.render('human')
  File "/home/mike/Desktop/notebook/gym/gym/core.py", line 249, in render
    return self.env.render(mode, **kwargs)
  File "/home/mike/Desktop/notebook/gym/gym/envs/mujoco/mujoco_env.py", line 123, in render
    self._get_viewer(mode).render()
  File "/home/mike/anaconda3/lib/python3.7/site-packages/mujoco_py/mjviewer.py", line 202, in render
    render_inner_loop(self)
  File "/home/mike/anaconda3/lib/python3.7/site-packages/mujoco_py/mjviewer.py", line 180, in render_inner_loop
    frame = self._read_pixels_as_in_window()
  File "/home/mike/anaconda3/lib/python3.7/site-packages/mujoco_py/mjviewer.py", line 218, in _read_pixels_as_in_window
    self.sim.render(resolution[0], resolution[1])
  File "mujoco_py/mjsim.pyx", line 156, in mujoco_py.cymj.MjSim.render
  File "mujoco_py/mjsim.pyx", line 163, in mujoco_py.cymj.MjSim.render
  File "mujoco_py/mjrendercontext.pyx", line 153, in mujoco_py.cymj.MjRenderContext.render
  File "mujoco_py/opengl_context.pyx", line 140, in mujoco_py.cymj.OffscreenOpenGLContext.set_buffer_size
RuntimeError: Failed to set buffer size

It works well till I press V in mujoco-py window

HiPPoTUS, 2019-04-17:
Please, comment line 153 in mjrendercontext.pyx, because this code is incompatible.

> eglshim.c

```
int setOpenGLBufferSize(int device_id, int width, int height) {
    // Noop since we don't need to change buffer here.
    return 1;
}
```

HiPPoTUS, 2019-04-17:
add any changes before cytonizing!

closed 2019-04-17 by HiPPoTUS

cmacw, 2019-06-17:
Hi, I commented the line as said and still get the error. Do I need to reinstall or refresh in some way? I am new to using Linux and detail steps would be very helpful.

Since I had problem installing mujoco-py, I had to clone the repository in `~/mujoco-py/mujoco_py` and install with `sudo pip3 install -e . --no-cache`

Then I commented the line 153 in `~/mujoco-py/mujoco_py/mjrendercontext.pyx`
```
def render(self, width, height, camera_id=None):
        cdef mjrRect rect
        rect.left = 0
        rect.bottom = 0
        rect.width = width
        rect.height = height

        if self.sim.render_callback is not None:
            self.sim.render_callback(self.sim, self)

        # Sometimes buffers are too small.
        if width > self._con.offWidth or height > self._con.offHeight:
            new_width = max(width, self._model_ptr.vis.global_.offwidth)
            new_height = max(height, self._model_ptr.vis.global_.offheight)
            self.update_offscreen_size(new_width, new_height)

        if camera_id is not None:
            if camera_id == -1:
                self.cam.type = const.CAMERA_FREE
            else:
                self.cam.type = const.CAMERA_FIXED
            self.cam.fixedcamid = camera_id

        #comment line 153
        # self.opengl_context.set_buffer_size(width, height)

```

The error after commented the line is as below:
```
Creating window glfw
Buffer size too big
Traceback (most recent call last):
  File "/home/maco/Documents/MSc_CompSci/Individual_Project/Code/Python_Code/main.py", line 33, in <module>
    viewer.render()
  File "/home/maco/mujoco-py/mujoco_py/mjviewer.py", line 190, in render
    render_inner_loop(self)
  File "/home/maco/mujoco-py/mujoco_py/mjviewer.py", line 178, in render_inner_loop
    super().render()
  File "/home/maco/mujoco-py/mujoco_py/mjviewer.py", line 57, in render
    glfw.poll_events()
  File "/usr/local/lib/python3.6/dist-packages/glfw/__init__.py", line 1731, in poll_events
    _glfw.glfwPollEvents()
  File "/usr/local/lib/python3.6/dist-packages/glfw/__init__.py", line 609, in errcheck
    _reraise(exc[1], exc[2])
  File "/usr/local/lib/python3.6/dist-packages/glfw/__init__.py", line 45, in _reraise
    raise exception.with_traceback(traceback)
  File "/usr/local/lib/python3.6/dist-packages/glfw/__init__.py", line 588, in callback_wrapper
    return func(*args, **kwargs)
  File "/home/maco/mujoco-py/mujoco_py/mjviewer.py", line 326, in key_callback
    img = self._read_pixels_as_in_window()
  File "/home/maco/mujoco-py/mujoco_py/mjviewer.py", line 218, in _read_pixels_as_in_window
    self.sim.render(resolution[0], resolution[1])
  File "mujoco_py/mjsim.pyx", line 156, in mujoco_py.cymj.MjSim.render
  File "mujoco_py/mjsim.pyx", line 163, in mujoco_py.cymj.MjSim.render
  File "mujoco_py/mjrendercontext.pyx", line 153, in mujoco_py.cymj.MjRenderContext.render
  File "mujoco_py/opengl_context.pyx", line 140, in mujoco_py.cymj.OffscreenOpenGLContext.set_buffer_size
RuntimeError: Failed to set buffer size
```

HiPPoTUS, 2019-06-18:
Hello!
You should change the code before first mujoco launch. 
Couse of this is cytonizing the library. (when program is translated from python script to another format which doesn't provide any changes...if I understand correctly :) ) 
1. Reinstal mujoco-py (don't launch any code which requires mujoco-py)
2. Comment line
3. Launch!