openai/mujoco-py #394: Building fails on Ubuntu 18.04 with Python 3.7: Unrecognized command line option: '-R'

alok, 2019-04-27:
**Describe the bug**

Attempting to build with `pip3.7 install mujoco-py' leads to the following [error log](https://gist.github.com/a1340712f33b3f395bc3680a32167c51).

**To Reproduce**
Attempt to build with 3.7. 3.6 worked on my system.

**Error Messages**
See [gist of log](https://gist.github.com/a1340712f33b3f395bc3680a32167c51).

**Desktop (please complete the following information):**
 - OS: [e.g. macOS 10.13.6]: ubuntu 18.04
 - Python Version [e.g. 3.6.6]: 3.7.3
 - Mujoco Version [e.g. 1.50]: 2.0
 - mujoco-py version [e.g. 1.50.1.59]: 2.0.22

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`: `/usr/local/cuda-9.2/lib64:/usr/local/cuda-10.0/lib64:/home/alok/.mujoco/mujoco200/bin:/home/alok/.config/mujoco/mujoco/bin:`
 - output of: `echo $HOME`: `/home/alok`
 - output of: `echo $USER`: `alok`

MillionIntegrals, 2019-05-13:
Looks like an incompatible version of a C compiler. What's your `cc --version`?

alok, 2019-05-14:
7.4.0. I tried with gcc 8.2.0 as well by setting `CC=gcc=8`.

--- Alok

samuela, 2019-09-04:
Same thing here. I'm on `Ubuntu 18.04.3 LTS` and Python 3.7.4.

samuela, 2019-09-04:
I was on gcc version 7.4.0. 

I found out that clang does seem to work. Unfortunately the installer doesn't respect the `CC` environment variable though so you have to use `update-alternatives` instead:
```
sudo apt install clang
sudo update-alternatives --config cc
```