openai/mujoco-py #407: Render failed when including multiple composite object

kracon7, 2019-05-21:
Hi,
I was trying to simulate deformable object with mujoco200 and mujoco-py.
There are two composite bodies in the XML file. Mujoco-py failed to render objects that don't include composite bodies. However, if I remove one composite body and left with only one, everything is fine.

Here's the XML file I used:
```xml
<mujoco>
   <compiler inertiafromgeom='auto' angle='radian'/>
   <size njmax="10000" nconmax="3000"/>
   <visual>
        <map fogstart="3" fogend="5" force="0.1" znear="0.1"/>
        <quality shadowsize="2048"/>
        <global offwidth="800" offheight="800"/>
    </visual>
   <worldbody>
      <light diffuse=".5 .5 .5" pos="0 0 3" dir="0 0 -1"/>    
      <geom type="plane" size="1.5 1.5 0.1" rgba=".9 0 0 1"/>
      <body name='gripper_palm' pos='0 -0.2 0.6'>
         <joint name='g_1' type='slide' pos='0 0 0' axis='0 0 1 ' damping='.2' />
         <joint name='g_2' type='slide' pos='0 0 0' axis='0 1 0 ' damping='.2' />
         <geom pos='0 0 0' type='box' size='0.1 0.03 0.03' quat='1 0 0 0' condim='1' />
         <body name='left_finger' pos='-0.08 0 0' quat='0.7648422 0 0 0.6442177'>
            <joint name='l_f' type='hinge' pos='0 0 0' axis='0 0 -1 ' range='0 1.57' damping='.2' limited='true'/>
            <geom pos='0 0.1 0' type='box' size='0.02 0.1 0.02' condim='1' />          
               <body name='left_fingertip' pos='0 0.25 0' quat='0.7071068 0 0.7071068 0'>
                  <geom pos='0 0 0' type='cylinder' size='0.07 0.02' condim='3' />
                  <body name='B_0' pos='0 0 0.04'  >
                     <composite type="cylinder" count='14 14 4' spacing='0.01' prefix="C">
                        <geom type="sphere" size=".006" rgba=".2 .8 .8 1" condim='3' />
                     </composite>
                  </body>
               </body>
         </body>
         <body name='right_finger' pos='0.08 0 0' quat='0.7648422 0 0 -0.6442177'>
            <joint name='r_f' type='hinge' pos='0 0 0' axis='0 0 1 ' range='0 1.57' damping='.2' limited='true'/>
            <geom pos='0 0.1 0' type='box' size='0.02 0.1 0.02' condim='1' />
               <body name='right_fingertip' pos='0 0.25 0' quat='-0.7071068 0 0.7071068 0'>
                  <geom pos='0 0 0' type='cylinder' size='0.07 0.02' condim='3' />
                  <body name='B_1' pos='0 0 0.04' >
                     <composite type="cylinder" count='14 14 4' spacing='0.01' prefix="D" >
                        <geom type="sphere" size=".006" rgba=".2 .8 .8 1" condim='3' />
                     </composite>
                  </body>
               </body>
         </body>
      </body>
      <body name='table' pos='0 0.4 0'>
         <geom pos='0 0 0.2' type='box' size='.1 .1 .2'  quat="1 0 0 0  "/>
         <geom pos='0 0.5 0.4' type='box' size='.1 .1 .4'  quat="1 0 0 0  "/>
      </body>
      <body name='objCylinder' pos='0 0.4 0.5'>
            <freejoint/>
            <geom type='cylinder' mass='0.1' pos='0 0 0' rgba='0 0 1 1' size='0.07 0.1' />
      </body>
   </worldbody>
   <actuator>
      <motor name='g_1' gear='35' joint='g_1' />
      <motor name='g_2' gear='35' joint='g_2' />
      <motor name='l_f' gear='35' joint='l_f' />
      <motor name='r_f' gear='35' joint='r_f' />
   </actuator>
</mujoco>
```

And the python scripts to test the render issue:
```python
#!/usr/bin/env python3
import random
from mujoco_py import load_model_from_path, MjSim, MjViewer
import os

model = load_model_from_path("2dGripper.xml")
sim = MjSim(model)
viewer = MjViewer(sim)
sim_state = sim.get_state()

while True:
    sim.set_state(sim_state)
    for i in range(500): # number of iteration steps
        if i < 250:
            sim.data.ctrl[:] = 0.1
        else:
            sim.data.ctrl[:] = -0.1
        sim.step()
        viewer.render()
    if os.getenv('TESTING') is not None:
        break
```

By commenting out one of the composite body, the render is back to normal.

 - OS: Ubuntu 16.04
 - Python Version 3.6.8
 - Mujoco 200
 - mujoco-py 2.0.2.2

**Environment**
 - output of: `echo $LD_LIBRARY_PATH` : /home/jc/catkin_ws/devel/lib:/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu:/home/jc/.mujoco/mujoco200/bin:/usr/lib/nvidia-410

 - output of: `echo $HOME` : /home/jc

 - output of: `echo $USER` : jc

wutter, 2019-05-22:
Same for me on macos (python 3.7.3, mujoco200, mujoco-py 2.0.2.2), possibly
Duplicate of #373 ?

JadeCong, 2021-06-30:
Maybe you should add a prefix for every composite object when you construct multiple composite objects.

Ekanshh, 2022-07-19:
I also faced a similar issue while working with multiple soft bodies in the simulation. 

Below is an example on how to add prefix for multiple composite bodies, as mentioned by @JadeCong 
``` 
       <body name="A" pos="0. 0. 0.075" quat="0.5 0.5 -0.5 -0.5">
          <composite prefix="A" type="box" count="6 4 4" spacing="0.05" solrefsmooth="-1324.17 -17.59">
            <geom type="capsule" size="0.0075 0.025" rgba=".8 .2 .1 1" mass="0.1" friction="0.01 0.005 0.0001" contype="0" group="0"/>
            <skin material="skin-material" texcoord="true"/>
          </composite>
        </body> 
        <body name="B" pos="0.15 0. 0.075" quat="0.5 0.5 -0.5 -0.5">
          <composite prefix="B" type="box" count="6 4 4" spacing="0.05" solrefsmooth="-1324.17 -17.59">
            <geom type="capsule" size="0.0075 0.025" rgba=".8 .2 .1 1" mass="0.001" friction="0.01 0.005 0.0001" contype="0" group="0"/>
            <skin material="skin-material" texcoord="true"/> 
          </composite>
        </body>
        <body name="C" pos="-0.15 0. 0.075" quat="0.5 0.5 -0.5 -0.5">
          <composite prefix="C" type="box" count="6 4 4" spacing="0.05" solrefsmooth="-1324.17 -17.59">
            <geom type="capsule" size="0.0075 0.025" rgba=".8 .2 .1 1" mass="0.001" friction="0.01 0.005 0.0001" contype="0" group="0"/>
            <skin material="skin-material" texcoord="true"/> 
          </composite>
        </body>
```