openai/mujoco-py #408: Offscreen render does not work (GLEW initalization error: Missing GL version)

caprest, 2019-05-23:
**Describe the bug**
Rendering with off screen fails to initialize.

**To Reproduce**
```
import mujoco_py
model = """
<mujoco model="cartpole">
  <worldbody>
    <geom name="floor" pos="0 0 -3.0" size="40 40 40" type="plane"/>
  </worldbody>
</mujoco>
"""

model = mujoco_py.load_model_from_xml(model)
sim = mujoco_py.MjSim(model)
mujoco_py.MjRenderContextOffscreen(sim, device_id=0)
```

**Error Messages**
`ERROR: GLEW initalization error: Missing GL version`

It seems like the error occurs at line 81 of  `mujoco_py/mjrendercontext`
        
`mjr_makeContext(self._model_ptr, &self._con, mjFONTSCALE_150)`
(found by print debug)

Furthermore when I add "None" argument at mysterious position
`mujoco_py.MjRenderContextOffscreen(sim, None,device_id=0)`
images are correctly rendered, though black winodow appears.

**Desktop (please complete the following information):**
 - OS: [Ubuntu18.04 ]
 - Python Version [3.7.3]
 - Mujoco Version [2.0.0]
 - mujoco-py version [2.0.2.2]

**Environment**
 - output of: `echo $LD_LIBRARY_PATH` `/home/user/.mujoco/mujoco200/bin`

**Additional context**
OpenGL version: 4.6.0
Nvidia Driver: 418.67

katadh, 2019-06-04:
I am having the same problem.

gitanshu, 2019-06-04:
Have you already looked at #268 ?

caprest, 2019-06-05:
At least, I already tried the solution for `LD_PRELOAD` and seems like to have correct output

```
>>> export LD_PRELOAD
/usr/lib/x86_64-linux-gnu/libGLEW.so
```

ZiwenZhuang, 2019-07-05:
It seems like if you have it render in 'window' mode for the first time, there will be no error. But the following offscreen render will be all zeros (black) matrix. (So, I still didn't fix it.)

ZiwenZhuang, 2019-07-07:
After gained some insight from my several tests, I found that letting the system render on screen or make a 'window' mode render context will get rid of the problem. But that would lead to a serious buffer problem and get only all zero numpy arrays if you keep using that context viewer, which might due to some OpenGL buffer problems.
Also, I've noticed that PyOpenGL supports OpenGL only from v1.1 to 4.4 (referring to [here](http://pyopengl.sourceforge.net/)). But you reported your OpenGL version is 4.6 :)

**Solution**

1. Theoretically, you can downgrade your OpenGL version via reinstall a lower version of Nvidia driver. But that would lead to a series of re-installation of Cuda, or more. So I didn't try that.

2. Let the entire program initialize an on-screen render context ('window' mode) and abandon it. Then, make your off-screen render context as normal, which will ensure you get the image you expected. Although this will leave a black window on your screen and system keep reporting 'mujoco-py not responding'.

dingchenghu, 2019-09-01:
I face exactly the same issue. 
Any other idea on how to solve that? Thanks

RajK853, 2019-10-28:
This solved my issue in Ubuntu 18.04.3 LTS:
´sudo mkdir -p /use/lib/nvidia-000´
Then add this in ~/.bashrc file:
´export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia-000´

Rowing0914, 2019-12-17:
I am having the same problem.

RajK853, 2019-12-17:
Recently while installing mujoco-py in a new conda environment to use tensorflow-gpu 1.13.1, I faced the same problem again. My problem was resolved by installing tensorflow-gpu and glew using conda instead of pip. This will also install the supported cuda version in the conda and it might have solved my issue.

Rowing0914, 2019-12-17:
> This solved my issue in Ubuntu 18.04.3 LTS:
> `sudo mkdir -p /use/lib/nvidia-000`
> Then add this in ~/.bashrc file:
> `export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia-000`

@RajK853 
Thank you for your sharing! In my case, which is described above, your advice worked well and solved my issue.

So, after I created the empty dir named `nvidia-OOO`(`OOO == 430` in my case), at the first execution of `import mujoco_py` it started compiling the library again though, everything seems to settle now!

wangcongrobot, 2020-01-27:
Hi, I have the folder `/usr/lib/nvidia-410` but it's not empty. How to deal with it?
```bash
$ ll /usr/lib/nvidia-410
total 196224
drwxr-xr-x   6 root root     4096 Jul 30 11:34 ./
drwxr-xr-x 188 root root    20480 Jan 13 23:30 ../
-rw-r--r--   1 root root        0 Nov 15  2018 alt_ld.so.conf
drwxr-xr-x   2 root root     4096 Jul 30 11:34 bin/
-rw-r--r--   1 root root       42 Nov 15  2018 ld.so.conf
lrwxrwxrwx   1 root root       23 Nov 15  2018 libEGL_nvidia.so.0 -> libEGL_nvidia.so.410.78
-rw-r--r--   1 root root  1031552 Nov 11  2018 libEGL_nvidia.so.410.78
lrwxrwxrwx   1 root root       16 Nov 15  2018 libEGL.so -> libEGL.so.410.78
lrwxrwxrwx   1 root root       16 Nov 15  2018 libEGL.so.1 -> libEGL.so.410.78
-rw-r--r--   1 root root    73328 Nov 11  2018 libEGL.so.1.1.0
-rw-r--r--   1 root root    23392 Nov 11  2018 libEGL.so.410.78
-rw-r--r--   1 root root   732408 Nov 11  2018 libGLdispatch.so.0
lrwxrwxrwx   1 root root       29 Nov 15  2018 libGLESv1_CM_nvidia.so.1 -> libGLESv1_CM_nvidia.so.410.78
-rw-r--r--   1 root root    60200 Nov 11  2018 libGLESv1_CM_nvidia.so.410.78
lrwxrwxrwx   1 root root       17 Nov 15  2018 libGLESv1_CM.so -> libGLESv1_CM.so.1
lrwxrwxrwx   1 root root       21 Jul 30 11:34 libGLESv1_CM.so.1 -> libGLESv1_CM.so.1.2.0
-rw-r--r--   1 root root    43696 Nov 11  2018 libGLESv1_CM.so.1.2.0
lrwxrwxrwx   1 root root       26 Nov 15  2018 libGLESv2_nvidia.so.2 -> libGLESv2_nvidia.so.410.78
-rw-r--r--   1 root root   111400 Nov 11  2018 libGLESv2_nvidia.so.410.78
lrwxrwxrwx   1 root root       14 Nov 15  2018 libGLESv2.so -> libGLESv2.so.2
lrwxrwxrwx   1 root root       18 Jul 30 11:34 libGLESv2.so.2 -> libGLESv2.so.2.1.0
-rw-r--r--   1 root root    83280 Nov 11  2018 libGLESv2.so.2.1.0
lrwxrwxrwx   1 root root       10 Nov 15  2018 libGL.so -> libGL.so.1
lrwxrwxrwx   1 root root       15 Nov 15  2018 libGL.so.1 -> libGL.so.410.78
-rw-r--r--   1 root root   685848 Nov 11  2018 libGL.so.1.7.0
-rw-r--r--   1 root root  1270576 Nov 11  2018 libGL.so.410.78
lrwxrwxrwx   1 root root       23 Nov 15  2018 libGLX_indirect.so.0 -> libGLX_nvidia.so.410.78
lrwxrwxrwx   1 root root       23 Nov 15  2018 libGLX_nvidia.so.0 -> libGLX_nvidia.so.410.78
-rw-r--r--   1 root root  1270576 Nov 11  2018 libGLX_nvidia.so.410.78
lrwxrwxrwx   1 root root       11 Nov 15  2018 libGLX.so -> libGLX.so.0
-rw-r--r--   1 root root    65168 Nov 11  2018 libGLX.so.0
lrwxrwxrwx   1 root root       15 Nov 15  2018 libnvcuvid.so -> libnvcuvid.so.1
lrwxrwxrwx   1 root root       20 Nov 15  2018 libnvcuvid.so.1 -> libnvcuvid.so.410.78
-rw-r--r--   1 root root  2807440 Nov 11  2018 libnvcuvid.so.410.78
-rw-r--r--   1 root root   370984 Nov 11  2018 libnvidia-cbl.so.410.78
lrwxrwxrwx   1 root root       18 Nov 15  2018 libnvidia-cfg.so -> libnvidia-cfg.so.1
lrwxrwxrwx   1 root root       23 Nov 15  2018 libnvidia-cfg.so.1 -> libnvidia-cfg.so.410.78
-rw-r--r--   1 root root   179528 Nov 11  2018 libnvidia-cfg.so.410.78
lrwxrwxrwx   1 root root       23 Nov 15  2018 libnvidia-compiler.so -> libnvidia-compiler.so.1
lrwxrwxrwx   1 root root       28 Nov 15  2018 libnvidia-compiler.so.1 -> libnvidia-compiler.so.410.78
-rw-r--r--   1 root root 47842480 Nov 11  2018 libnvidia-compiler.so.410.78
-rw-r--r--   1 root root 25215360 Nov 11  2018 libnvidia-eglcore.so.410.78
lrwxrwxrwx   1 root root       30 Nov 15  2018 libnvidia-egl-wayland.so.1 -> libnvidia-egl-wayland.so.1.1.0
-rw-r--r--   1 root root    32904 Nov 11  2018 libnvidia-egl-wayland.so.1.1.0
lrwxrwxrwx   1 root root       21 Nov 15  2018 libnvidia-encode.so -> libnvidia-encode.so.1
lrwxrwxrwx   1 root root       26 Nov 15  2018 libnvidia-encode.so.1 -> libnvidia-encode.so.410.78
-rw-r--r--   1 root root   168184 Nov 11  2018 libnvidia-encode.so.410.78
-rw-r--r--   1 root root   292840 Nov 11  2018 libnvidia-fatbinaryloader.so.410.78
lrwxrwxrwx   1 root root       18 Nov 15  2018 libnvidia-fbc.so -> libnvidia-fbc.so.1
lrwxrwxrwx   1 root root       23 Nov 15  2018 libnvidia-fbc.so.1 -> libnvidia-fbc.so.410.78
-rw-r--r--   1 root root   119048 Nov 11  2018 libnvidia-fbc.so.410.78
-rw-r--r--   1 root root 27018280 Nov 11  2018 libnvidia-glcore.so.410.78
-rw-r--r--   1 root root   580856 Nov 11  2018 libnvidia-glsi.so.410.78
-rw-r--r--   1 root root 13792536 Nov 11  2018 libnvidia-glvkspirv.so.410.78
lrwxrwxrwx   1 root root       18 Nov 15  2018 libnvidia-ifr.so -> libnvidia-ifr.so.1
lrwxrwxrwx   1 root root       23 Nov 15  2018 libnvidia-ifr.so.1 -> libnvidia-ifr.so.410.78
-rw-r--r--   1 root root   206888 Nov 11  2018 libnvidia-ifr.so.410.78
lrwxrwxrwx   1 root root       17 Nov 15  2018 libnvidia-ml.so -> libnvidia-ml.so.1
lrwxrwxrwx   1 root root       22 Nov 15  2018 libnvidia-ml.so.1 -> libnvidia-ml.so.410.78
-rw-r--r--   1 root root  1528152 Nov 11  2018 libnvidia-ml.so.410.78
lrwxrwxrwx   1 root root       34 Jul 30 11:34 libnvidia-ptxjitcompiler.so.1 -> libnvidia-ptxjitcompiler.so.410.78
-rw-r--r--   1 root root 12129192 Nov 11  2018 libnvidia-ptxjitcompiler.so.410.78
-rw-r--r--   1 root root 27151728 Nov 11  2018 libnvidia-rtcore.so.410.78
-rw-r--r--   1 root root    13080 Nov 11  2018 libnvidia-tls.so.410.78
lrwxrwxrwx   1 root root       23 Jul 30 11:34 libnvidia-wfb.so.1 -> libnvidia-wfb.so.410.78
-rw-r--r--   1 root root   295416 Jun  4  2018 libnvidia-wfb.so.410.78
[... 6 lines cut ...]
drwxr-xr-x   2 root root     4096 Jul 30 11:34 xorg/
```
And 
```bash
echo $LD_LIBRARY_PATH
/home/cong/ros_ws/dual_ws/devel/lib:/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu:/usr/local/cuda-10.0/lib64:/home/cong/.mujoco/mujoco200/bin:/usr/lib/nvidia-410
```
When I run the code
```python
import gym
from gym.wrappers import Monitor

env = gym.make('HalfCheetah-v2')
env = Monitor(env, './video', force=True)
env.reset()
while True:
    obs, r, done, info = env.step([0, 0, 0, 0, 0, 0])
    if done: break
```
I also get the error:
```bash
ERROR: GLEW initalization error: Missing GL version

Press Enter to exit ...
```

seyeeet, 2020-08-29:
@wangcongrobot did you find the solution?

wangcongrobot, 2020-09-02:
@seyeeet I use Unity3d to render.

hermanjakobsen, 2020-10-09:
Has anyone been able to come up with a solution to this problem?

Vigilans, 2020-11-30:
To sum up: do not link the libraries provided by system packages (e.g., libglew-dev and libegl1-mesa)
```
/usr/lib/x86_64-linux-gnu/libGLEW.so
/usr/lib/x86_64-linux-gnu/mesa-egl/libEGL.so.1
```
Instead, link the libraries provided by mujoco and nvidia:
```
$HOME/.mujoco/mujoco200/bin/libglewegl.so
/usr/lib/nvidia-xxx/libEGL.so.1
```
A proper environment setup may look like this:
```
export LD_LIBRARY_PATH=/usr/lib/nvidia-<xxx>:$LD_LIBRARY_PATH
export LD_LIBRARY_PATH=$HOME/.mujoco/mujoco200/bin:$LD_LIBRARY_PATH
export LD_PRELOAD=/usr/lib/nvidia-410/libEGL.so.1:$LD_PRELOAD
export LD_PRELOAD=$HOME/.mujoco/mujoco200/bin/libglewegl.so:$LD_PRELOAD
```

#### Explanation

Inside `cymj.OffscreenOpenGLContext`, `initOpenGL` function (EGL version) will be called, which first initializes EGL, then intializes glew by calling `glewInit()`:
https://github.com/openai/mujoco-py/blob/4830435a169c1f3e3b5f9b58a7c3d9c39bdf4acb/mujoco_py/gl/eglshim.c#L113-L119

If the glew library provided by system is linked, the `glewInit()` call will be directed to an implementation which invokes `glGetString(GL_VERSION)` (sourced from `apt source libglew-dev` 1.13.0):
```c
  /* query opengl version */
  s = glGetString(GL_VERSION);
  dot = _glewStrCLen(s, '.');
  if (dot == 0)
    return GLEW_ERROR_NO_GL_VERSION;
```
Which results in the error in the issue.  It seems that the EGL is correctly initialized, but the system glew library cannot correctly detect it.

However, mujoco's `libglewegl.so` seems to provide an implementation of glew with special support for EGL. Linking with this library will let `glewInit()` correctly executed without error.

Also, `libEGL.so.1` should be linked with the nvidia provided one, linking with the `libegl1-mesa` one will result in segment fault from my tested machine.

---

Above code comes from `cymj`, a cython package which will be compiled locally after installing mujoco-py. You make check the library dependency of the generated .so file in this way:
```
$ python
>>> import mujoco_py
>>> mujoco_py.__file__
'<some-path>/site-packages/mujoco_py/__init__.py'

$ ldd <some-path>/site-packages/mujoco_py/generated/cymj_2.0.2.<xxxxxxxx>.so
```
A working machine reports the dependency like this:
```
        linux-vdso.so.1 =>  (0x00007fffb65f8000)
        libglewegl.so => /home/<user>/.mujoco/mujoco200/bin/libglewegl.so (0x00007f1a0a1c1000)
        libmujoco200.so => /home/<user>/.mujoco/mujoco200/bin/libmujoco200.so (0x00007f1a09e0e000)
        libEGL.so.1 => /usr/lib/nvidia-418/libEGL.so.1 (0x00007f1a09c08000)
        libOpenGL.so.0 => /usr/lib/nvidia-418/libOpenGL.so.0 (0x00007f1a099d3000)
        libgomp.so.1 => /usr/lib/x86_64-linux-gnu/libgomp.so.1 (0x00007f1a097b1000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f1a09594000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f1a091ca000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f1a08e48000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f1a08b3f000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f1a08929000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f1a08725000)
        libGLdispatch.so.0 => /usr/lib/nvidia-418/libGLdispatch.so.0 (0x00007f1a08452000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f1a0a69a000)
```
A non-working machine will typically include dependencies like:
```
/usr/lib/x86_64-linux-gnu/libGLEW.so (0x00007fa44cf82000)
```
And once you added `libGLEW.so` to the `LD_PRELOAD`, the `glewInit()` may be forwarded to the implementation of it and the error happens thereafter.

wangcongrobot, 2020-12-20:
> @wangcongrobot did you find the solution?

When I reinstall my OS system from Ubuntu16.04 to Ubuntu18.04, the solution above works fine.

wangcongrobot, 2020-12-20:
> @seyeeet did you find the solution?

When I reinstall my OS system from Ubuntu16.04 to **Ubuntu18.04**, the solution above works fine.

hermanjakobsen, 2021-01-18:
Hi @Vigilans! 

Thank you for coming up with a what seems to be viable solution. :smile: However, I am not very proficient at modifying linked packages etc on Ubuntu and I am therefore terrified of messing something up. I would therefore be thrilled if you could make a step-by-step guide for us dummies :sweat_smile: 

What I have previously tried is
```
sudo mkdir -p /usr/lib/nvidia-455
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia-455
```
When I then ran 
```
python 
>> import mujoco_py
```
the package recompiled. Trying to use offscreen rendering with the [robosuite](https://github.com/ARISE-Initiative/robosuite) framework then yields 
```
Found 4 GPUs for rendering. Using device 0.
Traceback (most recent call last):
  File "main.py", line 28, in <module>
    controller_demo('UR5e', save_data=False)
  File "/home/hermankj/Documents/masters_thesis/src/demos.py", line 21, in controller_demo
    env = suite.make(
  File "/home/hermankj/Documents/masters_thesis/venv/lib/python3.8/site-packages/robosuite/environments/base.py", line 41, in make
    return REGISTERED_ENVS[env_name](*args, **kwargs)
  File "/home/hermankj/Documents/masters_thesis/src/my_environments/fetchpush.py", line 140, in __init__
    super().__init__(
  File "/home/hermankj/Documents/masters_thesis/venv/lib/python3.8/site-packages/robosuite/environments/manipulation/manipulation_env.py", line 147, in __init__
    super().__init__(
  File "/home/hermankj/Documents/masters_thesis/venv/lib/python3.8/site-packages/robosuite/environments/robot_env.py", line 186, in __init__
    super().__init__(
  File "/home/hermankj/Documents/masters_thesis/venv/lib/python3.8/site-packages/robosuite/environments/base.py", line 147, in __init__
    self._reset_internal()
  File "/home/hermankj/Documents/masters_thesis/src/my_environments/fetchpush.py", line 325, in _reset_internal
    super()._reset_internal()
  File "/home/hermankj/Documents/masters_thesis/venv/lib/python3.8/site-packages/robosuite/environments/robot_env.py", line 390, in _reset_internal
    super()._reset_internal()
  File "/home/hermankj/Documents/masters_thesis/venv/lib/python3.8/site-packages/robosuite/environments/base.py", line 280, in _reset_internal
    render_context = MjRenderContextOffscreen(self.sim, device_id=self.render_gpu_device_id)
  File "venv/lib/python3.8/site-packages/mujoco_py/mjrendercontext.pyx", line 43, in mujoco_py.cymj.MjRenderContext.__init__
  File "venv/lib/python3.8/site-packages/mujoco_py/mjrendercontext.pyx", line 121, in mujoco_py.cymj.MjRenderContext._setup_opengl_context
  File "venv/lib/python3.8/site-packages/mujoco_py/opengl_context.pyx", line 130, in mujoco_py.cymj.OffscreenOpenGLContext.__init__
RuntimeError: Failed to initialize OpenGL
```
**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
 ```
 /usr/lib/cuda/include:/usr/lib/cuda/lib64:/home/hermankj/webots/lib:/home/hermankj/.mujoco/mujoco200/bin:/usr/lib/nvidia-455
 ```
 - output of: `ldd <some-path>/site-packages/mujoco_py/generated/cymj_2.0.2.<xxxxxxxx>.so`
 ```
 linux-vdso.so.1 (0x00007ffeb15ff000)
	/usr/lib/x86_64-linux-gnu/libGLEW.so (0x00007f4f715e4000)
	/usr/lib/x86_64-linux-gnu/libGL.so (0x00007f4f7155c000)
	libglewosmesa.so => /home/hermankj/.mujoco/mujoco200/bin/libglewosmesa.so (0x00007f4f712c3000)
	libmujoco200.so => /home/hermankj/.mujoco/mujoco200/bin/libmujoco200.so (0x00007f4f70f10000)
	libOSMesa.so.8 => /usr/lib/x86_64-linux-gnu/libOSMesa.so.8 (0x00007f4f703ed000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f4f701f9000)
	libGLdispatch.so.0 => /usr/lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007f4f70141000)
	libGLX.so.0 => /usr/lib/x86_64-linux-gnu/libGLX.so.0 (0x00007f4f7010d000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f4f6ff2c000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f4f6fddd000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f4f6fdc2000)
	libglapi.so.0 => /usr/lib/x86_64-linux-gnu/libglapi.so.0 (0x00007f4f6fd87000)
	libLLVM-11.so.1 => /usr/lib/x86_64-linux-gnu/libLLVM-11.so.1 (0x00007f4f6ade3000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f4f6adc7000)
	libzstd.so.1 => /usr/lib/x86_64-linux-gnu/libzstd.so.1 (0x00007f4f6ad1e000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f4f6ad18000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f4f6acf5000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f4f71cb5000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f4f6abb6000)
	libffi.so.7 => /usr/lib/x86_64-linux-gnu/libffi.so.7 (0x00007f4f6abaa000)
	libedit.so.2 => /usr/lib/x86_64-linux-gnu/libedit.so.2 (0x00007f4f6ab72000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f4f6ab67000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007f4f6ab37000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f4f6ab0b000)
	libbsd.so.0 => /usr/lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f4f6aaf1000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f4f6aaeb000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f4f6aae3000)
 ```

RajK853, 2021-11-23:
> > > This solved my issue in Ubuntu 18.04.3 LTS:
> > > `sudo mkdir -p /use/lib/nvidia-000`
> > > Then add this in ~/.bashrc file:
> > > `export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia-000`
> > 
> > 
> > @RajK853 Thank you for your sharing! In my case, which is described above, your advice worked well and solved my issue.
> > So, after I created the empty dir named `nvidia-OOO`(`OOO == 430` in my case), at the first execution of `import mujoco_py` it started compiling the library again though, everything seems to settle now!
> > Edit: I am running scripts on PyCharm, which requires some tricky bit in terms of `environment variables`....
> 
> and how to know the 000?

Maybe it's the nvidia driver version installed in your Ubuntu.

RayYoh, 2021-11-24:
> > > > 这解决了我在 Ubuntu 18.04.3 LTS 中的问题：
> > > > `sudo mkdir -p /use/lib/nvidia-000`
> > > > 然后将其添加到 ~/.bashrc 文件中：
> > > > `export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia-000`
> > > 
> > > 
> > > @RajK853感谢您的分享！就我而言，如上所述，您的建议效果很好并解决了我的问题。
> > > 因此，在我创建了名为`nvidia-OOO`（`OOO == 430`在我的例子中）的空目录之后，在它第一次执行时`import mujoco_py`再次开始编译库，现在一切似乎都解决了！
> > > 编辑：我在 PyCharm 上运行脚本，这需要一些棘手的`environment variables`......
> > 
> > 
> > 以及如何知道000？
> 
> 也许这是您的 Ubuntu 中安装的 nvidia 驱动程序版本。

Sad, it does not work for me.

RajK853, 2021-11-24:
> Sad, it does not work for me.

Sad to hear that. I am not sure if this will help but recently I reinstalled the packages again in my conda environment and I was able to solve the issue with the mujoco with these settings. Hope it helps  

[image]

RayYoh, 2021-11-25:
> 

I will try it.

RayYoh, 2021-11-25:
> Sad to hear that. I am not sure if this will help but recently I reinstalled the packages again in my conda environment and I was able to solve the issue with the mujoco with these settings. Hope it [...]

I will try it. It is so difficult for me.