openai/mujoco-py #410: Error in Building Wheels and "You appear to be missing MuJoCo" for Mac mujoco-py version 1.50.2

weijiafeng, 2019-05-29:
Hi all~,

When I tried to install Mujoco-py it keeps on getting the error: "**Building wheel for mujoco-py (setup.py) ...**" error and "**You appear to be missing MuJoCo. We expected to find the file here: /Users/maggiefeng/.mujoco/mujoco200**". but I have clearly put the mujoco files in the described folder:
[image]

I've tried most of the solutions from past posts but haven't succeeded yet with version 1.50.2. When I install Mujoco-py 0.5.7 it can be installed successfully, but for some tasks, 0.5.7 is not enough.

FYI - My environment is:

MacOS Mojave version 10.14.1
Python 3.5.2
Gym version I am not sure but I tried version 0.9 (but never succeeded overall)

Note: in the screenshot below, though I downgraded to python 3.5.2, the installed Mujoco-py 0.5.7 is still in the Python 3.7 library, not sure if this is an issue? Thank you very much for your help!

[image]

caprest, 2019-05-30:
it seems like you have to put the mujoco library at `.mujoco` folder instead of `mujoco` folder.
Check the output of  `ls /Users/maggiefeng/.mujoco` to verify the library is properly placed.

weijiafeng, 2019-05-30:
@caprest Yep it works following what you said, thank you very much! :-)

closed 2019-05-30 by weijiafeng

zueting, 2020-02-27:
Hello,
I encounter the same problem. But I am having trouble create the .folder in mac. How did you do it?

weijiafeng, 2020-02-28:
@zueting use Mac terminal's command line to create the folder, since folders starting with "." are "invisible" in the folder view, you need to create and view details in the folder via command line

zueting, 2020-02-28:
Thanks for your help

CS309, 2021-04-05:
@weijiafeng I think I am having the same problem.
How do I create and view details in the folder via command line, like you say? I tried renaming the folder in terminal using "mv folder .folder" but I got this back: "mv: rename folder to .folder: Not a directory". 

Next, I tried unzipping my file into .folder (I had already unzipped it into the original folder), and got this back: 
"Archive:  /Users/myname/Downloads/myfile.zip
checkdir:  cannot create extraction directory: .folder
           File exists"

Finally, I tried creating .folder with "mkdir .folder" and got back 
"mkdir: .folder: File exists"

Please help!

CS309, 2021-04-05:
When I try "mkdir Users/myname/.folder"

I get back "mkdir: Users/myname: No such file or directory"

As background info, I am on macos.

IndupriyaKompi, 2021-05-06:
I tried this and it worked.

1. mkdir ./mujoco
2. cd ./mujoco
3. download/copy/move mujoco200 here