openai/mujoco-py #412: Failed building wheel for mujoco-py

cmacw, 2019-05-30:
I am really new to both Linux and MuJoCo and I am really sorry to bother you guys.

I follow the tutorial from the readme in git. 

Put the unzipped folder here:
`tree .mujoco/
.mujoco/
├── mjkey.txt
└── mujoco200
    ├── bin
    │   ├── basic
    │   ├── compile
    │   ├── derivative
    │   ├── libglewegl.so
    │   ├── libglewosmesa.so
    │   ├── libglew.so
    │   ├── libglfw3.a
    │   ├── libglfw.so.3
    │   ├── libmujoco200nogl.so
    │   ├── libmujoco200.so
    │   ├── record
    │   ├── simulate
    │   ├── testspeed
    │   └── testxml
    ├── doc
    │   ├── README.txt
    │   └── REFERENCE.txt
    ├── include
    │   ├── glfw3.h
    │   ├── mjdata.h
    │   ├── mjmodel.h
    │   ├── mjrender.h
    │   ├── mjui.h
    │   ├── mjvisualize.h
    │   ├── mjxmacro.h
    │   ├── mujoco.h
    │   ├── uitools.c
    │   └── uitools.h
    ├── model
    │   ├── arm26.xml
    │   ├── carpet.png
    │   ├── cloth.xml
    │   ├── grid1pin.xml
    │   ├── grid1.xml
    │   ├── grid2pin.xml
    │   ├── grid2.xml
    │   ├── hammock.xml
    │   ├── humanoid100.xml
    │   ├── humanoid.xml
    │   ├── loop.xml
    │   ├── marble.png
    │   ├── particle.xml
    │   ├── rope.xml
    │   ├── scene.xml
    │   ├── softbox.xml
    │   ├── softcylinder.xml
    │   ├── softellipsoid.xml
    │   └── sponge.png
    └── sample
        ├── basic.cpp
        ├── compile.cpp
        ├── derivative.cpp
        ├── makefile
        ├── mjx.cpp
        ├── mjxmake.m
        ├── record.cpp
        ├── simulate.cpp
        ├── testspeed.cpp
        └── testxml.cpp
`
[... 35 lines cut ...]
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/tmp/pip-install-cby_7arg/mujoco-py/mujoco_py -I/home/maco/.mujoco/mujoco200/include -I/usr/local/lib/python3.6/dist-packages/numpy/core/include -I/usr/include/python3.6m -c /tmp/pip-install-cby_7arg/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-install-cby_7arg/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/tmp/pip-install-cby_7arg/mujoco-py/mujoco_py/cymj.o -fopenmp -w
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/tmp/pip-install-cby_7arg/mujoco-py/mujoco_py -I/home/maco/.mujoco/mujoco200/include -I/usr/local/lib/python3.6/dist-packages/numpy/core/include -I/usr/include/python3.6m -c /tmp/pip-install-cby_7arg/mujoco-py/mujoco_py/gl/osmesashim.c -o /tmp/pip-install-cby_7arg/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/tmp/pip-install-cby_7arg/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
  creating /tmp/pip-install-cby_7arg/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_36_linuxcpuextensionbuilder/lib.linux-x86_64-3.6
  creating /tmp/pip-install-cby_7arg/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_36_linuxcpuextensionbuilder/lib.linux-x86_64-3.6/mujoco_py
  x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 /tmp/pip-install-cby_7arg/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/tmp/pip-install-cby_7arg/mujoco-py/mujoco_py/cymj.o /tmp/pip-install-cby_7arg/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/tmp/pip-install-cby_7arg/mujoco-py/mujoco_py/gl/osmesashim.o -L/home/maco/.mujoco/mujoco200/bin -Wl,--enable-new-dtags,-R/home/maco/.mujoco/mujoco200/bin -lmujoco200 -lglewosmesa -lOSMesa -lGL -o /tmp/pip-install-cby_7arg/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_36_linuxcpuextensionbuilder/lib.linux-x86_64-3.6/mujoco_py/cymj.cpython-36m-x86_64-linux-gnu.so -fopenmp
  error: [Errno 2] No such file or directory: 'patchelf': 'patchelf'
  
  ----------------------------------------
  Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
Failed to build mujoco-py
Installing collected packages: mujoco-py
  Running setup.py install for mujoco-py ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-cby_7arg/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-ucvbjd8a/install-record.txt --single-version-externally-managed --compile --user --prefix=:
    running install
    running build
    Removing old mujoco_py cext /tmp/pip-install-cby_7arg/mujoco-py/mujoco_py/generated/cymj_2.0.2.2_36_linuxcpuextensionbuilder_36.so
    running build_ext
    error: [Errno 2] No such file or directory: 'patchelf': 'patchelf'
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-cby_7arg/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-ucvbjd8a/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-install-cby_7arg/mujoco-py/
`

Thank you a lot!!

closed 2019-05-30 by cmacw

DanielTakeshi, 2019-06-14:
@cmacw  What was the solution?

lbertge, 2019-10-10:
not OP but I ran into the same error, given the same initial setup (`tree .mujoco` gives the equivalent output). 

[This comment](https://github.com/openai/mujoco-py/issues/147#issuecomment-361417560) onwards shows that you can install the missing utility `patchelf` through the package manager of your OS distribution. Then retry the pip install command.

Hope that helps anyone else who stumbles on this issue