openai/mujoco-py #414: How to get document of mujoco-py

1576012404, 2019-06-03:
i want to get document of usage of API of mujoco-py
for example,the meaning of model.data.com_subtree   model.data.qpos model.data.qvel

MillionIntegrals, 2019-06-05:
Most of these things are very thing wrappers on top of underlying C structures from mujoco itself. Documentation is here. http://mujoco.org/book/APIreference.html

wxplovehlt, 2019-06-15:
> Most of these things are very thing wrappers on top of underlying C structures from mujoco itself. Documentation is here. http://mujoco.org/book/APIreference.html

what does mean"To include mujoco-py in your own package",where mujoco-py should be located?

tuonan, 2019-07-08:
@MillionIntegrals  Thanks . Can you understand sim.data.qpos ,sim.data.qvel?I watch it,butBut its explanation is too little and too vague for me.

MillionIntegrals, 2019-07-08:
`qpos` is basically a position of each joint in the simulation and `qvel` are the velocities of these joints.

wxplovehlt, 2019-07-09:
what  about the action[3]  in  the env  Fetchpickandplace.the value of it means？