openai/mujoco-py #418: Error while importing mujoco in python 3.6.8: mjmodel.h not found

adityagupte95, 2019-06-14:
**Describe the bug**
The mjmodel.h file cannot be found while importing the mujoco_py.

But it does exist at the prescribed location

_**ls ~/.mujoco/mujoco200/mujoco200_linux/include
glfw3.h  mjdata.h  mjmodel.h  mjrender.h  mjui.h  mjvisualize.h  mjxmacro.h  mujoco.h  uitools.c  uitools.h**_

on request I have also added the following line to my .bashrc the 
 **_Please add following line to .bashrc:
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/aditya/.mujoco/mujoco200/bin_**

**To Reproduce**
Followed the same exact steps as prescribed in the readme file on Github and got the same error.
Also git cloned and tried installing the library with the following command:
**_sudo pip3 install -e . --no-cache_**

**Expected behavior**
The expected behaviour would have been that the the library gets imported.

**Error Messages**
This the output of my terminal as the error occurs

$python
Python 3.6.8 (default, Dec 24 2018, 19:24:27) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
running build_ext
building 'mujoco_py.cymj' extension
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco-py/mujoco-py/mujoco_py -I/home/aditya/mujoco-py/mujoco-py/mujoco_py -I/home/aditya/.mujoco/mujoco200/include -I/home/aditya/.local/lib/python3.6/site-packages/numpy/core/include -I/home/aditya/mujoco-py/mujoco-py/mujoco_py/vendor/egl -I/usr/include/python3.6m -c /home/aditya/mujoco-py/mujoco-py/mujoco_py/cymj.c -o /home/aditya/mujoco-py/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/home/aditya/mujoco-py/mujoco-py/mujoco_py/cymj.o -fopenmp -w
/home/aditya/mujoco-py/mujoco-py/mujoco_py/cymj.c:641:21: fatal error: mjmodel.h: No such file or directory
compilation terminated.
Traceback (most recent call last):
  File "/usr/lib/python3.6/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/usr/lib/python3.6/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/usr/lib/python3.6/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/usr/lib/python3.6/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'x86_64-linux-gnu-gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/aditya/mujoco-py/mujoco-py/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/aditya/mujoco-py/mujoco-py/mujoco_py/builder.py", line 503, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/aditya/mujoco-py/mujoco-py/mujoco_py/builder.py", line 105, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/aditya/mujoco-py/mujoco-py/mujoco_py/builder.py", line 220, in build
    built_so_file_path = self._build_impl()
  File "/home/aditya/mujoco-py/mujoco-py/mujoco_py/builder.py", line 290, in _build_impl
    so_file_path = super()._build_impl()
  File "/home/aditya/mujoco-py/mujoco-py/mujoco_py/builder.py", line 243, in _build_impl
    dist.run_commands()
  File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/aditya/.local/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/usr/lib/python3.6/distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/home/aditya/mujoco-py/mujoco-py/mujoco_py/builder.py", line 143, in build_extensions
    build_ext.build_extensions(self)
  File "/home/aditya/.local/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 194, in build_extensions
    self.build_extension(ext)
  File "/usr/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
    depends=ext.depends)
  File "/usr/lib/python3.6/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/usr/lib/python3.6/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command 'x86_64-linux-gnu-gcc' failed with exit status 1

**Desktop (please complete the following information):**
 - OS: UBUNTU 16.04 LTS
 - Python Version : 3.6.8
 - Mujoco Version: 200
 - mujoco-py version 2.1

**Environment**
 - output of: `echo $LD_LIBRARY_PATH'
/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu:/home/aditya/.mujoco/mujoco200/bin:/usr/lib/nvidia-410

 - output of: `echo $HOME`
/home/aditya
 
- output of: `echo $USER`
aditya

**Additional context**
I had the same problem with python 3.5.2.

closed 2019-06-14 by adityagupte95

xiapengchng, 2019-08-26:
how do you fix this, I meet the totally same question?

adityagupte95, 2019-08-26:
Did you sort the problem? If not could you please paste the directory path