openai/mujoco-py #419: MacOS ERROR: Could not read activation key

mfeng9, 2019-06-15:
`./simulate` under mujoco200/bin could successfully launch the Mujoco GUI window, however, the mujoco-py could NOT be imported with the following error:

``` 
import mujoco_py

ERROR: Could not read activation key

Press Enter to exit ...
```

**To Reproduce**
Steps to reproduce the behavior.

SEE FULL TERMINAL HISTORY BELOW

**Expected behavior**
A clear and concise description of what you expected to happen.

import should simply work, but at the moment it does not

**Error Messages**
Including more/longer error messages gives us more information to work with.

Full Terminal Output:
```
➜  /Users/sam/.mujoco pip3 install 'mujoco-py<2.1,>=2.0'
Collecting mujoco-py<2.1,>=2.0
Requirement already satisfied: lockfile>=0.12.2 in /usr/local/lib/python3.6/site-packages (from mujoco-py<2.1,>=2.0) (0.12.2)
Requirement already satisfied: numpy>=1.11 in /usr/local/lib/python3.6/site-packages (from mujoco-py<2.1,>=2.0) (1.16.4)
Requirement already satisfied: Cython>=0.27.2 in /usr/local/lib/python3.6/site-packages (from mujoco-py<2.1,>=2.0) (0.29.10)
Requirement already satisfied: imageio>=2.1.2 in /usr/local/lib/python3.6/site-packages (from mujoco-py<2.1,>=2.0) (2.5.0)
Requirement already satisfied: glfw>=1.4.0 in /usr/local/lib/python3.6/site-packages (from mujoco-py<2.1,>=2.0) (1.8.1)
Requirement already satisfied: cffi>=1.10 in /usr/local/lib/python3.6/site-packages (from mujoco-py<2.1,>=2.0) (1.12.3)
Requirement already satisfied: pillow in /usr/local/lib/python3.6/site-packages (from imageio>=2.1.2->mujoco-py<2.1,>=2.0) (6.0.0)
Requirement already satisfied: pycparser in /usr/local/lib/python3.6/site-packages (from cffi>=1.10->mujoco-py<2.1,>=2.0) (2.19)
Installing collected packages: mujoco-py
Successfully installed mujoco-py-2.0.2.2
You are using pip version 10.0.1, however version 19.1.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
➜  /Users/sam/.mujoco pip3 list
Package        Version
-------------- -------
cffi           1.12.3
Cython         0.29.10
glfw           1.8.1
imageio        2.5.0
lockfile       0.12.2
mujoco-py      2.0.2.2
numpy          1.16.4
Pillow         6.0.0
pip            10.0.1
pip-autoremove 0.9.1
pycparser      2.19
setuptools     39.2.0
wheel          0.31.1
You are using pip version 10.0.1, however version 19.1.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
➜  /Users/sam/.mujoco python3
Python 3.6.5 (default, Jun 17 2018, 12:13:06)
[GCC 4.2.1 Compatible Apple LLVM 9.1.0 (clang-902.0.39.2)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
Import error. Trying to rebuild mujoco_py.
Compiling /usr/local/lib/python3.6/site-packages/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /usr/local/lib/python3.6/site-packages/mujoco_py/cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
creating /usr/local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.2_36_macextensionbuilder
creating /usr/local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.2_36_macextensionbuilder/temp.macosx-10.13-x86_64-3.6
creating /usr/local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.2_36_macextensionbuilder/temp.macosx-10.13-x86_64-3.6/usr
creating /usr/local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.2_36_macextensionbuilder/temp.macosx-10.13-x86_64-3.6/usr/local
creating /usr/local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.2_36_macextensionbuilder/temp.macosx-10.13-x86_64-3.6/usr/local/lib
creating /usr/local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.2_36_macextensionbuilder/temp.macosx-10.13-x86_64-3.6/usr/local/lib/python3.6
creating /usr/local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.2_36_macextensionbuilder/temp.macosx-10.13-x86_64-3.6/usr/local/lib/python3.6/site-packages
creating /usr/local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.2_36_macextensionbuilder/temp.macosx-10.13-x86_64-3.6/usr/local/lib/python3.6/site-packages/mujoco_py
creating /usr/local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.2_36_macextensionbuilder/temp.macosx-10.13-x86_64-3.6/usr/local/lib/python3.6/site-packages/mujoco_py/gl
/usr/local/bin/gcc-7 -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -DONMAC -I/usr/local/lib/python3.6/site-packages/mujoco_py -I/Users/sam/.mujoco/mujoco200/include -I/usr/local/lib/python3.6/site-packages/numpy/core/include -I/usr/local/Cellar/python/3.6.5_1/Frameworks/Python.framework/Versions/3.6/include/python3.6m -c /usr/local/lib/python3.6/site-packages/mujoco_py/cymj.c -o /usr/local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.2_36_macextensionbuilder/temp.macosx-10.13-x86_64-3.6/usr/local/lib/python3.6/site-packages/mujoco_py/cymj.o -fopenmp -w
/usr/local/bin/gcc-7 -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -DONMAC -I/usr/local/lib/python3.6/site-packages/mujoco_py -I/Users/sam/.mujoco/mujoco200/include -I/usr/local/lib/python3.6/site-packages/numpy/core/include -I/usr/local/Cellar/python/3.6.5_1/Frameworks/Python.framework/Versions/3.6/include/python3.6m -c /usr/local/lib/python3.6/site-packages/mujoco_py/gl/dummyshim.c -o /usr/local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.2_36_macextensionbuilder/temp.macosx-10.13-x86_64-3.6/usr/local/lib/python3.6/site-packages/mujoco_py/gl/dummyshim.o -fopenmp -w
creating /usr/local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.2_36_macextensionbuilder/lib.macosx-10.13-x86_64-3.6
creating /usr/local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.2_36_macextensionbuilder/lib.macosx-10.13-x86_64-3.6/mujoco_py
/usr/local/bin/gcc-7 -bundle -undefined dynamic_lookup /usr/local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.2_36_macextensionbuilder/temp.macosx-10.13-x86_64-3.6/usr/local/lib/python3.6/site-packages/mujoco_py/cymj.o /usr/local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.2_36_macextensionbuilder/temp.macosx-10.13-x86_64-3.6/usr/local/lib/python3.6/site-packages/mujoco_py/gl/dummyshim.o -L/Users/sam/.mujoco/mujoco200/bin -L/Users/sam/.mujoco/mujoco200/bin -lmujoco200 -lglfw.3 -o /usr/local/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.2_36_macextensionbuilder/lib.macosx-10.13-x86_64-3.6/mujoco_py/cymj.cpython-36m-darwin.so -fopenmp
ERROR: Could not read activation key

Press Enter to exit ...
```
**Desktop (please complete the following information):**
 - OS:  macOS 10.14.3 (laptop)
 - Python Version 3.6.5
 - Mujoco Version : 2
 - mujoco-py version: 2.0.2.2

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`: `/Users/sam/.mujoco/mujoco200/bin:`
 - output of: `echo $HOME`: `/Users/sam`
 - output of: `echo $USER`: `sam`

**Additional context**
mjkey.txt file structure

```
.mujuco
  - mjkey.txt
  - mujoco200
     - bin
        - mjkey.txt
```

I also tried anaconda python 3.6, 3.5, and 3.7, none worked.

mfeng9, 2019-06-19:
I did a fresh reinstallation of my system but still got the same error. I tried version 1.5, and it worked with no problem.

caprest, 2019-06-20:
Probably, your `MUJOCO_PY_MJKEY_PATH` is wrong. Deleting the variable or changing it to the direct key path might work. Like `export MUJOCO_PY_MJKEY_PATH=/Users/sam/.mujoco/
mjkey.txt`
> If you want to specify a nonstandard location for the key and package, use the env variables MUJOCO_PY_MJKEY_PATH and MUJOCO_PY_MUJOCO_PATH.

Following is how mujoco_py activate with your key.
https://github.com/openai/mujoco-py/blob/1aeb02e2df429860aaba6783d61fc382c58bf885/mujoco_py/builder.py#L498-L503
https://github.com/openai/mujoco-py/blob/1aeb02e2df429860aaba6783d61fc382c58bf885/mujoco_py/utils.py#L72-L99
It prioritize the `MUJOCO_PY_MJKEY_PATH ` setting and don't look for other possible paths.

mfeng9, 2019-06-26:
Thank you all for the great help! I tried to append mjkey.txt to the MUJOCO_PY_MJKEY_PATH and it did look like it did the trick!

closed 2019-06-26 by mfeng9

HareshMiriyala, 2020-01-31:
^^^^^^^^^ this worked for me !