openai/mujoco-py #423: Add a possibility to remove a marker.

jendelel, 2019-06-25:
Hi, 

I want to record a video from the viewer that contains a lot of markers (4 per render). I am using visual markers to display intermediate goals set by the RL policy. Everything works great until I run out of geoms. 
If I add a marker and render, then the next time I render again, the marker isn't there anymore. However, I seems that the marker counter (`self._scn.ngeom`) does not reset. Thus, I run out of the geoms. 

Is there a possibility to either reset the counter or even better add a function to remove a marker?

To reproduce the issue, run this code

```
#!/usr/bin/env python
# demonstration of markers (visual-only geoms)

import math
import time
import os
import numpy as np
from mujoco_py import load_model_from_xml, MjSim, MjRenderContextOffscreen

MODEL_XML = """
<?xml version="1.0" ?>
<mujoco>
    <worldbody>
        <body name="box" pos="0 0 0.2">
            <geom size="0.15 0.15 0.15" type="box"/>
            <joint axis="1 0 0" name="box:x" type="slide"/>
            <joint axis="0 1 0" name="box:y" type="slide"/>
        </body>
        <body name="floor" pos="0 0 0.025">
            <geom size="1.0 1.0 0.02" rgba="0 1 0 1" type="box"/>
        </body>
    </worldbody>
</mujoco>
"""
model = load_model_from_xml(MODEL_XML)
sim = MjSim(model)
viewer = MjRenderContextOffscreen(sim, -1)
step = 0
while True:
    t = time.time()
    x, y = math.cos(t), math.sin(t)
    viewer.add_marker(pos=np.array([x, y, 1]),
                      label=str(t))
    viewer.render(500, 500)
    print(step)

    step += 1
```
It crashes when step reaches 1000.

Thank you, 

Lukas

jendelel, 2019-07-22:
I figured out a workaround.

According to MuJoCo forum, update_scene resets all virtual geoms, i.e. all markers. Unfortunately, mujoco-py does not clear the _markers list in the rgb_array (offscreen=True) mode. 

Calling `del viewer._markers[:]` after `render()` fixes the issue.

SmileLab-technion, 2019-08-08:
Thanks for sharing very helpful.

alexpostnikov, 2019-12-09:
@jendelel  thanks for the discovered way for removing old markers.
i am wondering where can one find the parameters of markers?
Can one change size, type color etc...?

jonasjuerss, 2021-05-10:
@alexpostnikov Even though you might not need the answer anymore, I'll post what I figured out for future reference:
It seems like you can pass the attributes from http://www.mujoco.org/book/APIreference.html#mjvGeom as arguments. e.g.:
`viewer.add_marker(pos=np.array([0.1, 0.2, 0.3]), type=2, label="test", size=np.array([0.01, 0.01, 0.01]), rgba=np.array([0.0, 0.0, 1.0, 1.0]))`

happygaoxiao, 2022-09-03:
@jendelel  Hello, thank you a lot for your post. I am also facing this error `RuntimeError: Ran out of geoms. maxgeom: 1000`. I tried to remove this line from `mjrendercontext.pyx`, reinstalled `mujoco-py` but it does not work. How could you remove markers in offscreen mode? I checked the data under`viewer`. There is no `_markers` attribute.

happygaoxiao, 2022-09-04:
In case someone might need it. Because the `_markers` attribute is read-only in the `mjrendercontext.pyx`. In order to reset it, we need to do the following steps.
1. Uninstall mujoco_py by `pip uninstall mujoco_py`.
2. Delete all `cdef readonly list _markers` lines in the `mujoco_py/cymj.c`. So that we can re-write it.
3. Delete all `.so` files and cache.
4. `pip setup.py install` to install the package.
5. In your main script, after `render()`, add `viewer._markers = []` to clear all markers.

A more elegant way should be to add a reset directly in`cymj.c`. But anyway, the above solution works well.