openai/mujoco-py #424: Issues with lock file

hartikainen, 2019-06-26:
**Describe the bug**
When a mujoco-py program is killed inside the `with LockFile(lockpath)` in https://github.com/openai/mujoco-py/blob/d98011924f25466ea5c3819e0460ff8de90e24a2/mujoco_py/builder.py#L89)
the next run of mujoco-py hangs indefinitely.

Currently, builder doesn't give any warning about the lock file being waited forever. I think the lock acquisition should have a timeout or at the very least a warning message to describe how to remove the stale lock file.

**To Reproduce**
1. Run python program that imports mujoco-py
2. Kill the process when it's inside the `with LockFile` block
3. Run python program that imports mujoco-py
4. Wait forever

**Expected behavior**
Either better cleaning of lock file to avoid situtations where stale lock files are left behind or better warning messages.

As a side note, it seems like the [lockfile packages is deprecated](https://pypi.org/project/lockfile/).

MillionIntegrals, 2019-07-08:
I think there is a PR in progress that tries to solve this problem, but I don't know what's the status of that is. It's definitely a problem we are aware of.

hartikainen, 2019-07-08:
Ah, thanks @MillionIntegrals, I totally missed the PR. Would love to see that getting merged soon. I left a couple of minor comments in the PR, hopefully those were useful.

samuela, 2019-09-25:
Just ran into this exact same deadlock. Any known workarounds?

samuela, 2019-09-25:
It appears lockfile is stored in
```
your-virtualenv/lib/python3.7/site-packages/mujoco_py/generated/mujocopy-buildlock.lock
```
Deleting it manually seems to do the trick.

gokhansolak, 2019-10-25:
For me, deleting the buildlock file manually did not solve the problem. Now there is no buildlock file anymore (searching linux root for "buildlock" doesn't produce any results), but the error is the same.

I was able to import the mujoco_py and run the few interactive lines from the README.md. Then, I quitted that python session and I was not able to run any mujoco-py programs anymore. The error is raised when the mujoco_py module is imported:

> ...   File "/usr/local/lib/python3.5/dist-packages/fasteners-0.15-py3.5.egg/fasteners/process_lock.py", line 156, in acquire
    self._do_open()
  File "/usr/local/lib/python3.5/dist-packages/fasteners-0.15-py3.5.egg/fasteners/process_lock.py", line 128, in _do_open
    self.lockfile = open(self.path, 'a')
PermissionError: [Errno 13] Permission denied: b'/usr/local/lib/python3.5/dist-packages/mujoco_py-2.0.2.7-py3.5.egg/mujoco_py/generated/mujocopy-buildlock'

I suspect that it may be related to installing the package from source instead of using _pip_. I run the following under _mujoco-py/_ to install:
`sudo python3 setup.py install`

System: Ubuntu 16.04, Python 3.5.2, mujoco200, mujoco-py current master branch.

gokhansolak, 2019-10-25:
Installing the package with pip3 instead of building the repo solved the issue.

oluwayetty, 2019-11-19:
Is there any progess on this yet? I installed mine with pip3 and I can't even find the "mujocopy-buildlock" in the generated folder. I'm using python version 3.7.5.

SaminYeasar, 2020-05-10:
still having trouble using image on the server due to this issue.