openai/mujoco-py #431: installation error - cannot find -lmujoco200 -lglewosmesa

MishaLaskin, 2019-07-02:
**Describe the bug**
`pip install mujoco-py` fails on headless ubuntu machine. Mujoco files and key are installed correctly and other mujoco python bindings (such as `dm_control`) work fine.

Could you please help me identify the issue and how to resolve it? Thank you!

**Error Messages**

```
pip install mujoco-py
Collecting mujoco-py
  Using cached https://files.pythonhosted.org/packages/a9/c8/daac2e747a0a516f6a880bf73304bd58e18d8be5cea926e8a1e5938026e6/mujoco-py-2.0.2.2.tar.gz
Requirement already satisfied: glfw>=1.4.0 in /opt/anaconda3/envs/research/lib/python3.7/site-packages (from mujoco-py) (1.8.1)
Requirement already satisfied: numpy>=1.11 in /opt/anaconda3/envs/research/lib/python3.7/site-packages (from mujoco-py) (1.16.2)
Requirement already satisfied: Cython>=0.27.2 in /opt/anaconda3/envs/research/lib/python3.7/site-packages (from mujoco-py) (0.29.6)
Requirement already satisfied: imageio>=2.1.2 in /opt/anaconda3/envs/research/lib/python3.7/site-packages (from mujoco-py) (2.5.0)
Requirement already satisfied: cffi>=1.10 in /opt/anaconda3/envs/research/lib/python3.7/site-packages (from mujoco-py) (1.12.2)
Requirement already satisfied: lockfile>=0.12.2 in /opt/anaconda3/envs/research/lib/python3.7/site-packages (from mujoco-py) (0.12.2)
Requirement already satisfied: pycparser in /opt/anaconda3/envs/research/lib/python3.7/site-packages (from cffi>=1.10->mujoco-py) (2.19)
Building wheels for collected packages: mujoco-py
  Building wheel for mujoco-py (setup.py) ... error
  ERROR: Complete output from command /opt/anaconda3/envs/research/bin/python -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-install-x_7g1960/mujoco-py/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-tl_ec33e --python-tag cp37:
  ERROR: running bdist_wheel
  running build
  Removing old mujoco_py cext /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/cymj_2.0.2.2_37_linuxcpuextensionbuilder_37.so
  Compiling /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/cymj.pyx because it depends on /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/wrappers.pxi.
  [1/1] Cythonizing /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder
  creating /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7
  creating /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp
  creating /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-x_7g1960
  creating /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-x_7g1960/mujoco-py
  creating /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-x_7g1960/mujoco-py/mujoco_py
  creating /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/gl
  gcc -pthread -B /opt/anaconda3/envs/research/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/tmp/pip-install-x_7g1960/mujoco-py/mujoco_py -I/home/misha/.mujoco/mujoco200/include -I/opt/anaconda3/envs/research/lib/python3.7/site-packages/numpy/core/include -I/opt/anaconda3/envs/research/include/python3.7m -c /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/cymj.o -fopenmp -w
  gcc -pthread -B /opt/anaconda3/envs/research/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/tmp/pip-install-x_7g1960/mujoco-py/mujoco_py -I/home/misha/.mujoco/mujoco200/include -I/opt/anaconda3/envs/research/lib/python3.7/site-packages/numpy/core/include -I/opt/anaconda3/envs/research/include/python3.7m -c /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/gl/osmesashim.c -o /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
  creating /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/lib.linux-x86_64-3.7
  creating /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/lib.linux-x86_64-3.7/mujoco_py
  gcc -pthread -shared -B /opt/anaconda3/envs/research/compiler_compat -L/opt/anaconda3/envs/research/lib -Wl,-rpath=/opt/anaconda3/envs/research/lib -Wl,--no-as-needed -Wl,--sysroot=/ /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/cymj.o /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/gl/osmesashim.o -L/home/misha/.mujoco/mujoco200/bin -Wl,-R/home/misha/.mujoco/mujoco200/bin -lmujoco200 -lglewosmesa -lOSMesa -lGL -o /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/lib.linux-x86_64-3.7/mujoco_py/cymj.cpython-37m-x86_64-linux-gnu.so -fopenmp
  /opt/anaconda3/envs/research/compiler_compat/ld: cannot find -lmujoco200
  /opt/anaconda3/envs/research/compiler_compat/ld: cannot find -lglewosmesa
  collect2: error: ld returned 1 exit status
  error: command 'gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
Failed to build mujoco-py
Installing collected packages: mujoco-py
  Running setup.py install for mujoco-py ... error
    ERROR: Complete output from command /opt/anaconda3/envs/research/bin/python -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-install-x_7g1960/mujoco-py/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-_mv2pj8u/install-record.txt --single-version-externally-managed --compile:
    ERROR: running install
    running build
    Removing old mujoco_py cext /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/cymj_2.0.2.2_37_linuxcpuextensionbuilder_37.so
    running build_ext
    building 'mujoco_py.cymj' extension
    gcc -pthread -B /opt/anaconda3/envs/research/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/tmp/pip-install-x_7g1960/mujoco-py/mujoco_py -I/home/misha/.mujoco/mujoco200/include -I/opt/anaconda3/envs/research/lib/python3.7/site-packages/numpy/core/include -I/opt/anaconda3/envs/research/include/python3.7m -c /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/cymj.o -fopenmp -w
    gcc -pthread -B /opt/anaconda3/envs/research/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/tmp/pip-install-x_7g1960/mujoco-py/mujoco_py -I/home/misha/.mujoco/mujoco200/include -I/opt/anaconda3/envs/research/lib/python3.7/site-packages/numpy/core/include -I/opt/anaconda3/envs/research/include/python3.7m -c /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/gl/osmesashim.c -o /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
    gcc -pthread -shared -B /opt/anaconda3/envs/research/compiler_compat -L/opt/anaconda3/envs/research/lib -Wl,-rpath=/opt/anaconda3/envs/research/lib -Wl,--no-as-needed -Wl,--sysroot=/ /tmp/pip-install-x_7g1960/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-in [... cut ...]
**Desktop (please complete the following information):**
```
Distributor ID: Debian
Description:    Debian GNU/Linux 9.9 (stretch)
Release:        9.9
Codename:       stretch
```

**Environment**
 - output of: `echo $LD_LIBRARY_PATH` 

`/usr/local/cuda/lib64:/usr/local/nccl2/lib:/usr/local/cuda/extras/CUPTI/lib64:/home/misha/.mujoco/mujoco200/bin`

 - output of: `echo $HOME`

`/home/misha`

 - output of: `echo $USER`

`misha`

**Additional context**
Tried this on a different machine and same problem occurred

MishaLaskin, 2019-07-13:
any update? running into the same issue on an ubuntu desktop

Wenxuan-Zhou, 2020-02-17:
I have the same issue and I resolved it by downloading mujoco200 again.

namansaxena9, 2021-08-23:
Make sure .mujoco/mujoco200/bin folder should contain all the following files: ['basic', 'compile', 'derivative', 'libglew.so', 'libglewegl.so', 'libglewosmesa.so', 'libglfw.so.3', 'libglfw3.a', 'libmujoco200.so', 'libmujoco200nogl.so', 'mjkey.txt', 'record', 'simulate', 'testspeed', 'testxml']

If all the files are not present redownload the mujoco200 folder from "https://www.roboti.us/index.html" according to your operating system