openai/mujoco-py #432: mujoco_py's MjSim doesn't render (Missing GL version) whereas gym's robotic env works fine

PierreAlexSW, 2019-07-03:
**Describe the bug**
When I try mujoco_py's MjSim and try to render the simulator, I get the classic "ERROR: GLEW initalization error: Missing GL version" error which has been raised by many people, however I failed to find a solution that worked out for me. This is even stranger considering that my gym environments using mujoco work completely fine (see following picture). 
[image]

Here is the easiest code to reproduce the error I get.
**To Reproduce**
import mujoco_py
import os
mj_path, _ = mujoco_py.utils.discover_mujoco()
print(mj_path)
xml_path = os.path.join(mj_path, 'model', 'humanoid.xml')
model = mujoco_py.load_model_from_path(xml_path)
sim = mujoco_py.MjSim(model)
print(sim.data.qpos)
sim.step()
print(sim.data.qpos)
sim.render() 

**Error Messages**
/home/usernmame/.mujoco/mujoco200
[0.  0.  1.4 1.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.
 0.  0.  0.  0.  0.  0.  0.  0.  0.  0. ]
[-1.12164337e-05  7.29847036e-22  1.39975300e+00  9.99999999e-01
  1.80085466e-21  4.45933954e-05 -2.70143345e-20  1.30126513e-19
 -4.63561234e-05 -1.88020744e-20 -2.24492958e-06  4.79357124e-05
 -6.38208396e-04 -1.61130312e-03 -1.37554006e-03  5.54173825e-05
 -2.24492958e-06  4.79357124e-05 -6.38208396e-04 -1.61130312e-03
 -1.37554006e-03 -5.54173825e-05 -5.73572648e-05  7.63833991e-05
 -2.12765194e-05  5.73572648e-05 -7.63833991e-05 -2.12765194e-05]
ERROR: GLEW initalization error: Missing GL version

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04.01 LTS
- Python version: 3.6.7
- Mujoco version: 200
- mujoco_py version: 2.0.2.2

**Environment**
My .bashrc file is the following:

export PATH=$PATH:/usr/local/cuda-9.0/bin
export LD_LIBRARY_PATH=/home/username/.mujoco/mujoco200/bin
export LD_LIBRARY_PATH=/usr/local/cuda-9.0/lib64:$LD_LIBRARY_PATH
export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so:/usr/lib/x86_64-linux-gnu/libGL.so
export MUJOCO_LICENSE_PATH=$HOME/.mujoco/mjkey.txt

**Additional context**
I also tried using the 150 version of Mujoco paired with 1.50 version of mujoco_py but this lead to exactly the same problem.
Thanks for your help!! Let me know if I missed anything worth mentioning to debug this.

ZiwenZhuang, 2019-07-07:
Have you checked your OpenGL version? It seems like the pyopengl only supports OpenGL version v1.1 to 4.4 (refer to [here](http://pyopengl.sourceforge.net/)), although based on your Cuda version, that should not be the problem

PierreAlexSW, 2019-07-07:
> Have you checked your OpenGL version? It seems like the pyopengl only supports OpenGL version v1.1 to 4.4 (refer to [here](http://pyopengl.sourceforge.net/)), although based on your Cuda version, that should not be the problem

Thanks for your answer! Just checked, and it seems my OpenGL is too recent?
OpenGL version string: 4.6.0 NVIDIA 390.116

Should I downgrade my OpenGL?

ZiwenZhuang, 2019-07-07:
> Thanks for your answer! Just checked, and it seems my OpenGL is too recent?
> OpenGL version string: 4.6.0 NVIDIA 390.116
>
> Should I downgrade my OpenGL?

Thank you for the version string. I thought the Nvidia driver version and OpenGL version are connected. That is only a guess based on what I have searched. I haven't tried that yet because I was worrying about reinstalling not only OpenGL, but also Nvidia driver and Cuda.
Now, it seems like it is only needed to downgrade the OpenGL version.

BTW: Have you encountered with other programs which use OpenGL under current settings?

PierreAlexSW, 2019-07-07:
> Thank you for the version string. I thought the Nvidia driver version and OpenGL version are connected. That is only a guess based on what I have searched. I haven't tried that yet because I was [...]

thanks again for the time :) No i've never seen OpenGL give any problem. This is still weird that when I'm using gym (that uses mujoco) I don't have any problem (see above pic), and this only happens when using the MjSim from mujoco_py.  
By the way, I did not have the pyopengl package but it didn't seem to be a problem before, it doesn't look like mujoco_py was depending on it.
If I can dowgrade the OpenGL version without changing the drivers that would be perfect, i'll try that and keep you updated on that.

pvskand, 2019-07-24:
@PierreAlexSW any update on this?