openai/mujoco-py #442: How to I change parameters of the model/environment on the fly?

homangab, 2019-08-01:
For example here https://github.com/rll/rllab/blob/master/vendor/mujoco_models/half_cheetah.xml I am able to change gravity by 

env = gym.make('HalfCheetah-v2')
env.model.opt.gravity[-1] = -4

But how do I change the other physical parameters like friction and visual paramters like color of the grid, brightness etc.?

homangab, 2019-08-01:
This thread https://github.com/openai/gym/issues/234 has some insights but I don;t see how to change the parameters I mentioned above!

gpzhaome, 2020-04-04:
> This thread [openai/gym#234](https://github.com/openai/gym/issues/234) has some insights but I don;t see how to change the parameters I mentioned above!

Hi, I have the same problem. Did you find a solution? I'm using Mujoco 2.0.

hermanjakobsen, 2020-11-14:
I also have the same problem on Mujoco 2.0. Has anyone come up with a fix for this? Looks like the parameters are not writable, as I get the following AttributeError when trying to change `body_mass`: 
```
AttributeError: attribute 'body_mass' of 'mujoco_py.cymj.PyMjModel' objects is not writable
```