openai/mujoco-py #443: Rendering Skin Problem

coffeiersama, 2019-08-06:
I am trying to render a cloth with skin in system Ubuntu 16.04 (or 18.04)
when I just put XML into mujoco200, it renders the skin successfully,
but when I try to run it with codes(env.render), 

it comes to the error: 
**Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)**
and does not create the window

I was just wondering in which way I can use the skin because, without the skin, I cannot see the visual attribute, which is important for me to move on in my project

wutter, 2019-08-09:
Duplicate of #373

coffeiersama, 2019-08-16:
> Duplicate of #373

Hello, Adam. Thanks for all your help. I have followed your suggestion by #373. However, it does not generate a continuous motions with skin. I think there must be something wrong with my setting of mujoco. I guess I am stuck here and do not have any clue ...
Do you mind helping to set it remotely, via like teamviewer? Thanks.