openai/mujoco-py #444: Error while importing mujoco_py

puranjayr96, 2019-08-07:
(tensorflow) Puranjays-MBP:gym puranjayrajvanshi$ python3
Python 3.6.1 (default, Jul 30 2017, 17:49:39) 
[GCC 4.2.1 Compatible Apple LLVM 8.1.0 (clang-802.0.42)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import gym
>>> import mujoco_py
Compiling /Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
creating /Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/generated/_pyxbld_2.0.2.4_36_macextensionbuilder
creating /Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/generated/_pyxbld_2.0.2.4_36_macextensionbuilder/temp.macosx-10.12-x86_64-3.6
creating /Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/generated/_pyxbld_2.0.2.4_36_macextensionbuilder/temp.macosx-10.12-x86_64-3.6/Users
creating /Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/generated/_pyxbld_2.0.2.4_36_macextensionbuilder/temp.macosx-10.12-x86_64-3.6/Users/puranjayrajvanshi
creating /Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/generated/_pyxbld_2.0.2.4_36_macextensionbuilder/temp.macosx-10.12-x86_64-3.6/Users/puranjayrajvanshi/tensorflow
creating /Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/generated/_pyxbld_2.0.2.4_36_macextensionbuilder/temp.macosx-10.12-x86_64-3.6/Users/puranjayrajvanshi/tensorflow/lib
creating /Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/generated/_pyxbld_2.0.2.4_36_macextensionbuilder/temp.macosx-10.12-x86_64-3.6/Users/puranjayrajvanshi/tensorflow/lib/python3.6
creating /Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/generated/_pyxbld_2.0.2.4_36_macextensionbuilder/temp.macosx-10.12-x86_64-3.6/Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages
creating /Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/generated/_pyxbld_2.0.2.4_36_macextensionbuilder/temp.macosx-10.12-x86_64-3.6/Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg
creating /Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/generated/_pyxbld_2.0.2.4_36_macextensionbuilder/temp.macosx-10.12-x86_64-3.6/Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py
creating /Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/generated/_pyxbld_2.0.2.4_36_macextensionbuilder/temp.macosx-10.12-x86_64-3.6/Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/gl
/usr/local/opt/llvm/bin/clang -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/usr/local/opt/llvm/include.bashrc/.zshrc -DONMAC -I/Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py -I/Users/puranjayrajvanshi/.mujoco/mujoco200/include -I/Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/numpy/core/include -I/Users/puranjayrajvanshi/.pyenv/versions/3.6.1/include/python3.6m -c /Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/cymj.c -o /Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/generated/_pyxbld_2.0.2.4_36_macextensionbuilder/temp.macosx-10.12-x86_64-3.6/Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/cymj.o -fopenmp -w
unable to execute '/usr/local/opt/llvm/bin/clang': No such file or directory
Traceback (most recent call last):
  File "/Users/puranjayrajvanshi/.pyenv/versions/3.6.1/lib/python3.6/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/Users/puranjayrajvanshi/.pyenv/versions/3.6.1/lib/python3.6/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/Users/puranjayrajvanshi/.pyenv/versions/3.6.1/lib/python3.6/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/Users/puranjayrajvanshi/.pyenv/versions/3.6.1/lib/python3.6/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command '/usr/local/opt/llvm/bin/clang' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/builder.py", line 503, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/builder.py", line 105, in load_cython_ext
    cext_so_path = builder.build()
  File "/Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/builder.py", line 220, in build
    built_so_file_path = self._build_impl()
  File "/Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/builder.py", line 329, in _build_impl
    so_file_path = super()._build_impl()
  File "/Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/builder.py", line 243, in _build_impl
    dist.run_commands()
  File "/Users/puranjayrajvanshi/.pyenv/versions/3.6.1/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/Users/puranjayrajvanshi/.pyenv/versions/3.6.1/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/Users/puranjayrajvanshi/.pyenv/versions/3.6.1/lib/python3.6/distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/mujoco_py-2.0.2.4-py3.6.egg/mujoco_py/builder.py", line 143, in build_extensions
    build_ext.build_extensions(self)
  File "/Users/puranjayrajvanshi/tensorflow/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 194, in build_extensions
    self.build_extension(ext)
  File "/Users/puranjayrajvanshi/.pyenv/versions/3.6.1/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
    depends=ext.depends)
  File "/Users/puranjayrajvanshi/.pyenv/versions/3.6.1/lib/python3.6/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/Users/puranjayrajvanshi/.pyenv/versions/3.6.1/lib/python3.6/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command '/usr/local/opt/llvm/bin/clang' failed with exit status 1

RussWong, 2019-10-29:
hello man i have the same question as u ,u solved it?could u tell me the solution?

1447866357, 2020-04-20:
i have the same question as u

mbelalsh, 2023-04-26:
Same question