openai/mujoco-py #451: Exception: Please add mujoco library to your PATH in Windows 64bit

kibamin, 2019-08-22:
I installed mujoco_py on python3.7 and windows 10 64bit successfully
But when `import mujoco_py` get this error :

> Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Program Files\JetBrains\PyCharm 2018.1.3\helpers\pydev\_pydev_bundle\pydev_import_hook.py", line 19, in do_import
    module = self._system_import(name, *args, **kwargs)
  File "D:\learning\MS\RL\codes\mujoco\env\lib\site-packages\mujoco_py-2.0.2.5-py3.7.egg\mujoco_py\__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "C:\Program Files\JetBrains\PyCharm 2018.1.3\helpers\pydev\_pydev_bundle\pydev_import_hook.py", line 19, in do_import
    module = self._system_import(name, *args, **kwargs)
  File "D:\learning\MS\RL\codes\mujoco\env\lib\site-packages\mujoco_py-2.0.2.5-py3.7.egg\mujoco_py\builder.py", line 507, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "D:\learning\MS\RL\codes\mujoco\env\lib\site-packages\mujoco_py-2.0.2.5-py3.7.egg\mujoco_py\builder.py", line 79, in load_cython_ext
    "set %s=%s;%%%s%%" % (var, lib_path, var))
**Exception: Please add mujoco library to your PATH:
set PATH=C:\Users\Amin\.mujoco\mujoco200\bin;%PATH%**

I set PATH but take same error again :|

how to fix this error?

chinmaySwami, 2019-09-24:
I was getting a similar error. I am using the path variables MUJOCO_PY_MJKEY_PATH and MUJOCO_PY_MUJOCO_PATH
So for MJKEY variable i set the mujoco200\bin\  path while  for MUJOCO path i set mujoco200\ path
I stopped getting the error above however i am getting a new error which says cannot 
"Could not open activation key file C:\mujoco\mujoco200\bin\mjkey.txt"

So the main problem is because you are not setting the path correctly for the key and for MUJOCO

gokriznastic, 2019-11-15:
@chinmaySwami doing this doesn't solve the issue for my end. I am stuck installing mujoco. Please help.

kushantp58, 2019-11-17:
Even I'm facing this error....Can somebody please help..

chinmaySwami, 2022-01-13:
For windows, We need to create these environment variables under system variables at the same time we also have to add the path C:\Users\Amin.mujoco\mujoco200\bin to the path variable 

NOTE : We do not need the MJKEY variable anymore since mujoco was acquired by deepmind and was made free.

omarabulaziz, 2022-02-05:
@chinmaySwami  can I do it without using Ubuntu? If so, can you please help to do so?

suhibalobeidat, 2022-04-01:
add the key to your bin directory