openai/mujoco-py #452: ERROR: Shadow framebuffer is not complete, error 0x8cd7

seanexplode, 2019-08-27:
**Describe the bug**
When using MjSim.render(), the program returns error

**To Reproduce**
`
>>> import mujoco_py as mj
>>> model = mj.load_model_from_path("./humanoid.xml")
>>> sim = mj.MjSim(model)
>>> sim.render()
ERROR: Shadow framebuffer is not complete, error 0x8cd7

Press Enter to exit ...
`

**Error Messages**
this is the only message I can see
`
ERROR: Shadow framebuffer is not complete, error 0x8cd7
`

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04.3 LTS
 - Python Version 3.7.4
 - Mujoco Version 2.0
 - mujoco-py version 2.0.2.5

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
`/home/sjjung/.mujoco/mujoco200/bin:/usr/local/cuda-10.0/lib64`
 - output of: `echo $HOME`
`/home/sjjung`
 - output of: `echo $USER`
`sjjung`

**Additional context**
Add any other context about the problem here.

when I set LD_PRELOAD properly and run

`env.render()`

in openai gym environment, it works smoothly.

Then I `unset LD_PRELOAD` and run

`sim.render` as above, I get error 0x8cd7

I have Titan Xp and nvidia-driver version is 430.40.

I'd love to hear any opinions and suggestions. Thanks in advance!

seanexplode, 2019-08-28:
For those who are suffering from this issue,

I installed nvidia driver as described in [nvidia official document](https://docs.nvidia.com/cuda/index.html).

This installed a file in /usr/lib/nvidia.

However, mujoco_py only finds /usr/lib/nvidia-[0-9][0-9][0-9].

You can see this code https://github.com/openai/mujoco-py/blob/master/mujoco_py/builder.py#L34

I made a symlink 
`ln -s /usr/lib/nvidia /usr/lib/nvidia-430`

and now it shows new error

```
Found 8 GPUs for rendering. Using device 0.
Traceback (most recent call last):
  File "mujoco_inspect.py", line 5, in <module>
    sim.render()
  File "mjsim.pyx", line 156, in mujoco_py.cymj.MjSim.render
  File "mjsim.pyx", line 163, in mujoco_py.cymj.MjSim.render
  File "mjrendercontext.pyx", line 134, in mujoco_py.cymj.MjRenderContext.render
TypeError: an integer is required
```

Giving proper arguments to `sim.render()` the errors are completely solved :)

closed 2019-08-28 by seanexplode

Cadene, 2024-01-24:
I also fixed this issue by making a symlink, but for me it was:
```
ln -s /usr/local/cuda-12.2 /usr/local/nvidia 
```