openai/mujoco-py #455: Support latest homebrew gcc

samuela, 2019-08-29:
**Describe the bug**
According to https://github.com/openai/mujoco-py/blob/master/mujoco_py/builder.py#L311 only gcc versions 6-8 are support on macOS. Now the latest version of gcc on homebrew is 9, which means that the install process is broken by default and the recommended fix:
```
brew install gcc
```
no longer works.

**To Reproduce**
Install homebrew gcc@9 and then attempt to install mujoco-py.

**Expected behavior**
`mujoco-py` to work with gcc@9.

**Error Messages**
```
Could not find GCC executable.\\n\\n'", '    RuntimeError: Could not find GCC executable.', '    ', '    HINT: On OS X, install GCC with `brew install gcc`. or `port install gcc`
```

**Desktop (please complete the following information):**
 - OS: macOS 10.14.6
 - Python Version 3.7.3
 - Mujoco Version 200
 - mujoco-py version 2.0.2.5

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
 - output of: `echo $HOME`
 - output of: `echo $USER`

```
research 🔥 echo $LD_LIBRARY_PATH

research 🔥 echo $HOME
/Users/skainswo
research 🔥 echo $USER
skainswo
```

**Additional context**
n/a

n-lavrenko, 2019-09-13:
The same error.

davidkiller96, 2019-09-25:
brew install gcc@8

thanks for the advice ~ . @n-lavrenko

RyanRizzo96, 2019-10-10:
Is there still an issue with gcc@9?

wookayin, 2021-07-29:
For those who don't want to pin homebrew `gcc` to an older version for some reason, try setting the `CC` environment variable. It will override the automatic gcc detection logic (which supports up to gcc-8).

```
export CC=/usr/local/bin/gcc-11         # assuming you have gcc@11 installed with homebrew
pip install mujoco-py
```

With this building wheel for `mujoco-py` should work. I don't think you should downgrade your gcc.

Sedherthe, 2024-04-21:
@wookayin This did not work for me. It still shows the same error. Uninstalled the already existing mujoco-py, before re-installing. Anything that I'm missing?