openai/mujoco-py #458: distutils.errors.LinkError: command 'gcc' failed with exit status 1

kdtmac, 2019-09-09:
**Describe the bug**
distutils.errors.LinkError: command 'gcc' failed with exit status 1

**To Reproduce**
I ran python and import mujoco_py, then error .
**Expected behavior**
What can i do

**Error Messages**
Python 3.7.3 (default, Mar 27 2019, 22:11:17) 
[GCC 7.3.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
running build_ext
building 'mujoco_py.cymj' extension
gcc -pthread -B /home/agnes/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Ianaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py -I/home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py -I/home/agnes/.mujoco/mujoco200/include -I/home/agnes/anaconda3/lib/python3.7/site-packages/numpy/core/include -I/home/agnes/anaconda3/include/python3.7m -c /home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py/cymj.c -o /home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py/generated/_pyxbld_2.0.2.5_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -B /home/agnes/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Ianaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py -I/home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py -I/home/agnes/.mujoco/mujoco200/include -I/home/agnes/anaconda3/lib/python3.7/site-packages/numpy/core/include -I/home/agnes/anaconda3/include/python3.7m -c /home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py/gl/osmesashim.c -o /home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py/generated/_pyxbld_2.0.2.5_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py/gl/osmesashim.o -fopenmp -w
gcc -pthread -shared -B /home/agnes/anaconda3/compiler_compat -L/home/agnes/anaconda3/lib -Wl,-rpath=/home/agnes/anaconda3/lib -Wl,--no-as-needed -Wl,--sysroot=/ /home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py/generated/_pyxbld_2.0.2.5_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py/cymj.o /home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py/generated/_pyxbld_2.0.2.5_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py/gl/osmesashim.o -L/home/agnes/.mujoco/mujoco200/bin -Wl,-R/home/agnes/.mujoco/mujoco200/bin -lmujoco200 -lglewosmesa -lOSMesa -lGL -o /home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py/generated/_pyxbld_2.0.2.5_37_linuxcpuextensionbuilder/lib.linux-x86_64-3.7/mujoco_py/cymj.cpython-37m-x86_64-linux-gnu.so -fopenmp
/home/agnes/anaconda3/compiler_compat/ld: cannot find -lGL
collect2: error: ld returned 1 exit status
Traceback (most recent call last):
  File "/home/agnes/anaconda3/lib/python3.7/distutils/unixccompiler.py", line 205, in link
    self.spawn(linker + ld_args)
  File "/home/agnes/anaconda3/lib/python3.7/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/home/agnes/anaconda3/lib/python3.7/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/home/agnes/anaconda3/lib/python3.7/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py/builder.py", line 507, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py/builder.py", line 105, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py/builder.py", line 220, in build
    built_so_file_path = self._build_impl()
  File "/home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py/builder.py", line 272, in _build_impl
    so_file_path = super()._build_impl()
  File "/home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py/builder.py", line 243, in _build_impl
    dist.run_commands()
  File "/home/agnes/anaconda3/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/home/agnes/anaconda3/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/agnes/anaconda3/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/home/agnes/anaconda3/lib/python3.7/distutils/command/build_ext.py", line 340, in run
    self.build_extensions()
  File "/home/agnes/anaconda3/lib/python3.7/site-packages/mujoco_py-2.0.2.5-py3.7.egg/mujoco_py/builder.py", line 143, in build_extensions
    build_ext.build_extensions(self)
  File "/home/agnes/anaconda3/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 194, in build_extensions
    self.build_extension(ext)
  File "/home/agnes/anaconda3/lib/python3.7/distutils/command/build_ext.py", line 559, in build_extension
    target_lang=language)
  File "/home/agnes/anaconda3/lib/python3.7/distutils/ccompiler.py", line 717, in link_shared_object
    extra_preargs, extra_postargs, build_temp, target_lang)
  File "/home/agnes/anaconda3/lib/python3.7/distutils/unixccompiler.py", line 207, in link
    raise LinkError(msg)
distutils.errors.LinkError: command 'gcc' failed with exit status 1

**Desktop (please complete the following information):**
 - OS: ubuntu18
 - Python Version [e.g. 3.6.6]3.7.3
 - Mujoco Version [e.g. 1.50]
 - mujoco-py version [e.g. 1.50.1.59]

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
 - output of: `echo $HOME`
 - output of: `echo $USER`

**Additional context**
Add any other context about the problem here.

closed 2019-09-10 by kdtmac

MishaLaskin, 2019-09-18:
@kdtmac how did you resolve this issue?

kdtmac, 2019-09-18:
> 
> 
> @kdtmac how did you resolve this issue?

another issue also encounter this problem, i follow the comment there. 
https://github.com/openai/mujoco-py/issues/374

MishaLaskin, 2019-09-18:
thanks ill check that out!

SapanaChaudhary, 2020-05-19:
@MishaLaskin Were you able to resolve this issue?