openai/mujoco-py #461: Rendering convex hull

btickell, 2019-09-28:
The simulate executable for mujoco allows for rendering the convex hull version of the scene which is helpful for analyzing the contact model of certain geoms.

Is it possible to do the same thing in `sim.render()` or a similar python setting?

priyankamandikal, 2019-12-14:
Is there any update on this? How do we visualize the convex hull in the renderer?

priyankamandikal, 2019-12-14:
btickell, were you able to figure out how to render the convex meshes?

btickell, 2019-12-15:
I ended up converting my meshes into convex meshes.

priyankamandikal, 2019-12-15:
Using the HACD library?

btickell, 2019-12-28:
Yes. This seems to be the standard approach from people I've talked to. I think rendering the hull would still be a useful feature.

priyankamandikal, 2020-01-04:
After some fiddling around with the code, I got it working with a small fix:
sim = MjSim(model)
viewer = MjViewer(sim)
viewer.vopt.flags[0] = 1

Hope this helps.

right-chan, 2022-09-26:
It was late, but I did it like this.

```python
import mujoco_py
from mujoco_py import load_model_from_path, MjSim, MjViewer
import os

mj_path = mujoco_py.utils.discover_mujoco()
xml_path = os.path.join(mj_path, 'model', 'humanoid.xml')

model = load_model_from_path(xml_path)
sim = MjSim(model)
viewer = MjViewer(sim)

viewer.vopt.flags[0] = True
viewer.render()
```
I found it [here](https://mujoco.readthedocs.io/en/latest/APIreference.html?highlight=mjVIS_CONVEXHULL#mjtvisflag).