openai/mujoco-py #463: Errror when Installing MuJoCo on mac OS Catalina

RyanRizzo96, 2019-10-10:
**Describe the bug**
I just upgraded to Catalina and am trying to install mujoco-py unsuccessfully.

**To Reproduce**
`pip3 install -U 'mujoco-py<2.1,>=2.0'`

**Expected behavior**
Installation completed normally as on Mojave.

**Error Messages**

> Collecting mujoco-py<2.1,>=2.0
>   Using cached https://files.pythonhosted.org/packages/30/d1/a6c45652ef5fc4d02d2dae4a056c0802496ce0a1b5855aacd0b478491fd4/mujoco-py-2.0.2.6.tar.gz
>   Installing build dependencies ... done
>   Missing build requirements in pyproject.toml for mujoco-py<2.1,>=2.0 from https://files.pythonhosted.org/packages/30/d1/a6c45652ef5fc4d02d2dae4a056c0802496ce0a1b5855aacd0b478491fd4/mujoco-py-2.0.2.6.tar.gz#sha256=3736b00a35a4322cf4cd57273c285491ab6b8e1edcb51dc43bbfa1bac945d186.
>   The project does not specify a build backend, and pip cannot fall back to setuptools without 'wheel'.
>   Getting requirements to build wheel ... done
>   Installing backend dependencies ... done
>     Preparing wheel metadata ... done
> Collecting Cython>=0.27.2 (from mujoco-py<2.1,>=2.0)
>   Using cached https://files.pythonhosted.org/packages/62/13/2ed2e2005afb114ffe08fbd25d890e3795ca86cebf11c97ff8877da58f28/Cython-0.29.13-cp37-cp37m-macosx_10_6_intel.macosx_10_9_intel.macosx_10_9_x86_64.macosx_10_10_intel.macosx_10_10_x86_64.whl
> Requirement already satisfied, skipping upgrade: numpy>=1.11 in ./.virtualenvs/baselines_pycharm/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (1.17.2)
> Collecting glfw>=1.4.0 (from mujoco-py<2.1,>=2.0)
> Collecting cffi>=1.10 (from mujoco-py<2.1,>=2.0)
>   Using cached https://files.pythonhosted.org/packages/f0/48/5aa4ea664eba26dd5142558d04762f5065c02220b4665b3f7eecb9bb614e/cffi-1.12.3-cp37-cp37m-macosx_10_9_x86_64.whl
> Collecting lockfile>=0.12.2 (from mujoco-py<2.1,>=2.0)
>   Using cached https://files.pythonhosted.org/packages/c8/22/9460e311f340cb62d26a38c419b1381b8593b0bb6b5d1f056938b086d362/lockfile-0.12.2-py2.py3-none-any.whl
> Collecting imageio>=2.1.2 (from mujoco-py<2.1,>=2.0)
>   Using cached https://files.pythonhosted.org/packages/1a/de/f7f985018f462ceeffada7f6e609919fbcc934acd9301929cba14bc2c24a/imageio-2.6.1-py3-none-any.whl
> Collecting pycparser (from cffi>=1.10->mujoco-py<2.1,>=2.0)
> Collecting pillow (from imageio>=2.1.2->mujoco-py<2.1,>=2.0)
>   Using cached https://files.pythonhosted.org/packages/a4/92/801710e21ede5f583c6bde61b25baa46ed16d9d77d74935e11b9a801cdd7/Pillow-6.2.0-cp37-cp37m-macosx_10_6_intel.macosx_10_9_intel.macosx_10_9_x86_64.macosx_10_10_intel.macosx_10_10_x86_64.whl
> Building wheels for collected packages: mujoco-py
>   Building wheel for mujoco-py (PEP 517) ... error
>   Complete output from command /Users/ryanr/.virtualenvs/baselines_pycharm/bin/python /Users/ryanr/.virtualenvs/baselines_pycharm/lib/python3.7/site-packages/pip-19.0.3-py3.7.egg/pip/_vendor/pep517/_in_process.py build_wheel /var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/tmphbwekt7q:
>   running bdist_wheel
>   running build
>   Removing old mujoco_py cext /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/generated/cymj_2.0.2.6_37_macextensionbuilder_37.so
>   Compiling /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/cymj.pyx because it depends on /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-build-env-pea65rty/overlay/lib/python3.7/site-packages/Cython/Includes/numpy/__init__.pxd.
>   [1/1] Cythonizing /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/cymj.pyx
>   running build_ext
>   building 'mujoco_py.cymj' extension
>   creating /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.6_37_macextensionbuilder
>   creating /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.6_37_macextensionbuilder/temp.macosx-10.14-x86_64-3.7
>   creating /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.6_37_macextensionbuilder/temp.macosx-10.14-x86_64-3.7/private
>   creating /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.6_37_macextensionbuilder/temp.macosx-10.14-x86_64-3.7/private/var
>   creating /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.6_37_macextensionbuilder/temp.macosx-10.14-x86_64-3.7/private/var/folders
>   creating /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.6_37_macextensionbuilder/temp.macosx-10.14-x86_64-3.7/private/var/folders/q7
>   creating /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.6_37_macextensionbuilder/temp.macosx-10.14-x86_64-3.7/private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp
>   creating /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.6_37_macextensionbuilder/temp.macosx-10.14-x86_64-3.7/private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T
>   creating /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.6_37_macextensionbuilder/temp.macosx-10.14-x86_64-3.7/private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9
>   creating /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.6_37_macextensionbuilder/temp.macosx-10.14-x86_64-3.7/private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py
>   creating /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.6_37_macextensionbuilder/temp.macosx-10.14-x86_64-3.7/private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py
>   creating /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.6_37_macextensionbuilder/temp.macosx-10.14-x86_64-3.7/private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/gl
>   /usr/local/bin/gcc-6 -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Tk.framework/Versions/8.5/Headers -DONMAC -Imujoco_py -I/private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py -I/Users/ryanr/.mujoco/mujoco200/include -I/private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-build-env-pea65rty/overlay/lib/python3.7/site-packages/numpy/core/include -I/Users/ryanr/.virtualenvs/baselines_pycharm/include -I/usr/local/Cellar/python/3.7.3/Frameworks/Python.framework/Versions/3.7/include/python3.7m -c /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/cymj.c -o /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.6_37_macextensionbuilder/temp.macosx-10.14-x86_64-3.7/private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/cymj.o -fopenmp -w
>   In file included from /usr/local/Cellar/gcc@6/6.5.0_2/lib/gcc/6/gcc/x86_64-apple-darwin18.5.0/6.5.0/include-fixed/syslimits.h:7:0,
>                    from /usr/local/Cellar/gcc@6/6.5.0_2/lib/gcc/6/gcc/x86_64-apple-darwin18.5.0/6.5.0/include-fixed/limits.h:34,
>                    from /usr/local/Cellar/python/3.7.3/Frameworks/Python.framework/Versions/3.7/include/python3.7m/Python.h:11,
>                    from /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-8hgnmsv9/mujoco-py/mujoco_py/cymj.c:58:
>   /usr/local/Cellar/gcc@6/6.5.0_2/lib/gcc/6/gcc/x86_64-apple-darwin18.5.0/6.5.0/include-fixed/limits.h:168:61: error: no include path in which to search for limits.h
>    #include_next <limits.h>  /* recurse down to the real one */
>                                                                ^
[... cut ...]
>    #include <_stdio.h>
>                       ^
>   compilation terminated.
>   error: command '/usr/local/bin/gcc-6' failed with exit status 1
>   
>   ----------------------------------------
>   Failed building wheel for mujoco-py
>   Running setup.py clean for mujoco-py
> Failed to build mujoco-py
> Could not build wheels for mujoco-py which use PEP 517 and cannot be installed directly
> 

**Desktop:**
 - OS: [10.15]
 - Python Version [3.7.4]
 - Mujoco Version [2.00]
 - mujoco-py version [ ]

cgel, 2019-10-21:
I'm also encountering the same issue.

Jendker, 2019-10-21:
Actually, is there anyone who can install mujoco-py under Catalina?

thepowerfuldeez, 2019-10-22:
Same issue

RyanRizzo96, 2019-10-22:
Hi, similar question here: https://stackoverflow.com/questions/58348607/macos-command-line-tools-and-gcc-suddenly-failing-on-catalina-worked-on-mojave

Try to use OS 10.14 SDK instead of the default SDK. This worked for me.

Jendker, 2019-10-22:
Thanks Ryan. Any hints on how to do it in details?

RyanRizzo96, 2019-10-22:
@Jendker  take a look [here](https://stackoverflow.com/questions/58351899/missing-header-files-on-macos-catalina) too. That question I asked was basically during my digging up on this issue.

These are my header files (Catalina 10.15):

```
/Library/Developer/CommandLineTools/usr/include/c++/v1/stdio.h
/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/sys/stdio.h
/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/stdio.h
/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Kernel.framework/Versions/A/Headers/sys/stdio.h
/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/stdio.h
/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/stdio.h
/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/Kernel.framework/Versions/A/Headers/sys/stdio.h
```

Run `sudo find /Library -name stdio.h ` to see where yours are located.

Mojave 10.14 header files:

```
$ sudo find /Library -name stdio.h 
/Library/Developer/CommandLineTools/usr/include/c++/v1/stdio.h
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/sys/stdio.h
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/stdio.h
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Kernel.framework/Versions/A/Headers/sys/stdio.h
```

As can be seen the SDKs are now split into MACOSX10.14 and 10.15 unlike in Mojave.

So, these were my SDK folders on Catalina:

[image]

Now:
1. Rename `MacOSX.sdk` to `MacOSX_oirig.sdk`
2. Right click on `MacOSX10.14.sdk`
3. Duplicate
4. Rename duplicate folder to `MacOSX.sdk`

Like this we are basically using the previous version's OSX sdk as sysroot. Hope this helps.

Jendker, 2019-10-22:
Quite hacky, but works, thanks Ryan! Would be nice if this issue could be fixed though :)

RyanRizzo96, 2019-10-22:
Glad it worked and saved you some time! 

Would be great if you could mark the answer with an emoji and up vote SO question/answer so that more people could see it helped. 

Hope this is fixed soon!

alexandrubalotescu, 2019-10-23:
@RyanRizzo96 Thanks, that seems to work for other people, but for me it doesn't. My SDK Folders look a bit different from yours:
[image]

My `MacOSX.sdk` seems to be an alias, not a file. Also, renaming `MacOSX10.14.sdk` to `MacOSX_orig.sdk` would mean there will be no `MacOSX10.14.sdk` left to duplicate. Or am I doing something wrong?

RyanRizzo96, 2019-10-23:
@alexandrubalotescu I had a mistake in my steps in the last edit I made. I updated the steps and they should make more sense now.

Rename MacOSX.sdk to MacOSX_orig.sdk (not OSX10.14)

Let me know if this helps.

alexandrubalotescu, 2019-10-23:
@RyanRizzo96 Thank, your solution now makes sense, but unfortunately it still doesn't work for me. I still get this:

```
error: command '/usr/local/bin/gcc-8' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
Failed to build mujoco-py
ERROR: Could not build wheels for mujoco-py which use PEP 517 and cannot be installed directly
```
What gcc did you end up using?

RyanRizzo96, 2019-10-23:
Hmm, so you are now using `MacOSX10.14.sdk` as your `MacOSX.sdk` and still having issues.

Try using gcc6, that's what ended up  working for me. Turns out some gcc versions are not compatible with some other pieces of software.

alexandrubalotescu, 2019-10-23:
Nope, gcc6 doesn't work either :(

RyanRizzo96, 2019-10-23:
Not sure what's going wrong in your case but I pretty much think it has to do with the new command line tools. Did you run `xcode-select --install` after your update to Catalina?

Run that command if you haven't already and try to reinstall gcc. I had to install and reinstall gcc a couple of times.

Jendker, 2019-10-24:
I have again issues with installing mujoco-py under Catalina for some reason. A solution to this issue will be highly appreciated :)

shenzhun, 2019-10-25:
Same issue under Catalina.

Yonv1943, 2019-10-25:
Same issue under Ubuntu 16.04
Python3.7, pip 19.2.3, mujoco-py 2.0.2.7~2.0.2.0
gcc 5.4.0 (Ubuntu~16.04 )

QAQ
```
**ERROR: Failed building wheel for mujoco-py**
Running setup.py clean for mujoco-py
Failed to build mujoco-py
ERROR: Could not build wheels for mujoco-py which use **PEP 517** and cannot be installed directly
```

jkim22, 2019-10-30:
Is this issue solved..?

RyanRizzo96, 2019-10-30:
No don't think so. I found a hacky workaround which I posted above, but nothing released officially as far as I know.

tekeburak, 2019-11-05:
```console
brew uninstall gcc
brew uninstall gcc@7
brew uninstall gcc@8
brew uninstall gcc@9

brew install gcc@6

git clone https://github.com/openai/mujoco-py.git
pip install -e ~/mujoco-py
```
Works on macOS Catalina
**SOLVED!**

alexandrubalotescu, 2019-11-05:
@tekeburak That actually finally worked for me!! Thank you very much! 

> ```
> brew uninstall gcc
> brew uninstall gcc@7
> brew uninstall gcc@9
> 
> brew install gcc@8
> 
> git clone https://github.com/openai/mujoco-py.git
> pip install -e ~/mujoco-py
> ```
> 
> Works on macOS Catalina
> **SOLVED!**

Alpslee, 2019-11-06:
> ```
> ```
> brew uninstall gcc
> brew uninstall gcc@7
> brew uninstall gcc@9
> 
> brew install gcc@8
> 
> git clone https://github.com/openai/mujoco-py.git
> pip install -e ~/mujoco-py
> ```
> 
> 
> Works on macOS Catalina
> SOLVED!
> ```

you saved my life! I just don't get it, it doesn't need mjkey.txt file anymore?

RyanRizzo96, 2019-11-06:
Could be because of the latest Catalina update (10.15.1) and XCode developer tools update that came along with it.

Jendker, 2019-11-06:
Have anyone tried installing mujoco-py directly from pip? I've made downgrade to Mojave so cannot test it.

tranhoangkhuongvn, 2019-11-07:
> ```
> brew uninstall gcc
> brew uninstall gcc@7
> brew uninstall gcc@9
> 
> brew install gcc@8
> 
> git clone https://github.com/openai/mujoco-py.git
> pip install -e ~/mujoco-py
> ```
> 
> Works on macOS Catalina
> **SOLVED!**

Hi, I successfully installed using this step. However trying to import mujoco, it throws out errors as:
`distutils.errors.DistutilsExecError: command '/usr/local/bin/gcc-8' failed with exit status 1`

tekeburak, 2019-11-07:
> Hi, I successfully installed using this step. However trying to import mujoco, it throws out errors as:
> `distutils.errors.DistutilsExecError: command '/usr/local/bin/gcc-8' failed with exit status 1`

I don't know why it's happened but you can try to install mujoco with gcc@6.

Jendker, 2019-11-08:
Mujoco-py can be installed directly with `pip install mujoco-py` on Catalina now, you can close the issue(s).

nkatz565, 2019-11-10:
> ```
> ```
> brew uninstall gcc
> brew uninstall gcc@7
> brew uninstall gcc@9
> 
> brew install gcc@8
> 
> git clone https://github.com/openai/mujoco-py.git
> pip install -e ~/mujoco-py
> ```
> ```

For some reason this didn't solve it for me. Tried with GCC 6 and 8.

Schrominger, 2019-11-15:
> @Jendker take a look [here](https://stackoverflow.com/questions/58351899/missing-header-files-on-macos-catalina) too. That question I asked was basically during my digging up on this issue.
> 
> These are my header files (Catalina 10.15):
> 
> ```
> /Library/Developer/CommandLineTools/usr/include/c++/v1/stdio.h
> /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/sys/stdio.h
> /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/stdio.h
> /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Kernel.framework/Versions/A/Headers/sys/stdio.h
> /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/stdio.h
> /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/stdio.h
> /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/Kernel.framework/Versions/A/Headers/sys/stdio.h
> ```
> 
> Run `sudo find /Library -name stdio.h ` to see where yours are located.
> 
> Mojave 10.14 header files:
> 
> ```
> $ sudo find /Library -name stdio.h 
> /Library/Developer/CommandLineTools/usr/include/c++/v1/stdio.h
> /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/sys/stdio.h
> /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/stdio.h
> /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Kernel.framework/Versions/A/Headers/sys/stdio.h
> ```
> 
> As can be seen the SDKs are now split into MACOSX10.14 and 10.15 unlike in Mojave.
> 
> **TLDR**
> 
> So, these were my SDK folders on Catalina:
> 
> [image]
> 
> 1. Rename `MacOSX.sdk` to `MacOSX_orig.sdk`
> 2. Right click on `MacOSX10.14.sdk`
> 3. Duplicate
> 4. Rename duplicate folder to `MacOSX.sdk`
> 
> Your folder structure should now look like this:
> 
> [image]
> 
> Like this we are basically using the previous version's OSX sdk as sysroot. Hope this helps.

---------------------
OS: [Mac OS Catalina 10.15.1]
Python Version [ 3.7.4]
Mujoco Version [Mujoco200] 
mujoco-py version [2.0.2.8] by ```pip install mujoco_py```

I added `gcc-9` into the `builder.py` file,
the above answer by @RyanRizzo96  fixes the problems for gcc, exited with status 1.
but new errors occur as following,

```shell
Python 3.7.4 (default, Aug 13 2019, 15:17:50) 
[Clang 4.0.1 (tags/RELEASE_401/final)] :: Anaconda, Inc. on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
Compiling /Users/ming/anaconda3/lib/python3.7/site-packages/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /Users/ming/anaconda3/lib/python3.7/site-packages/mujoco_py/cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
creating /Users/ming/anaconda3/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.0.2.8_37_macextensionbuilder
creating /Users/ming/anaconda3/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.0.2.8_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7
creating /Users/ming/anaconda3/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.0.2.8_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/Users
creating /Users/ming/anaconda3/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.0.2.8_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/Users/ming
creating /Users/ming/anaconda3/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.0.2.8_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/Users/ming/anaconda3
creating /Users/ming/anaconda3/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.0.2.8_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/Users/ming/anaconda3/lib
[... 4 lines cut ...]
/usr/local/bin/gcc-9 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Users/ming/anaconda3/include -arch x86_64 -I/Users/ming/anaconda3/include -arch x86_64 -DONMAC -Ianaconda3/lib/python3.7/site-packages/mujoco_py -I/Users/ming/anaconda3/lib/python3.7/site-packages/mujoco_py -I/Users/ming/.mujoco/mujoco200/include -I/Users/ming/anaconda3/lib/python3.7/site-packages/numpy/core/include -I/Users/ming/anaconda3/include/python3.7m -c /Users/ming/anaconda3/lib/python3.7/site-packages/mujoco_py/cymj.c -o /Users/ming/anaconda3/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.0.2.8_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/Users/ming/anaconda3/lib/python3.7/site-packages/mujoco_py/cymj.o -fopenmp -w
/usr/local/bin/gcc-9 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Users/ming/anaconda3/include -arch x86_64 -I/Users/ming/anaconda3/include -arch x86_64 -DONMAC -Ianaconda3/lib/python3.7/site-packages/mujoco_py -I/Users/ming/anaconda3/lib/python3.7/site-packages/mujoco_py -I/Users/ming/.mujoco/mujoco200/include -I/Users/ming/anaconda3/lib/python3.7/site-packages/numpy/core/include -I/Users/ming/anaconda3/include/python3.7m -c /Users/ming/anaconda3/lib/python3.7/site-packages/mujoco_py/gl/dummyshim.c -o /Users/ming/anaconda3/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.0.2.8_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/Users/ming/anaconda3/lib/python3.7/site-packages/mujoco_py/gl/dummyshim.o -fopenmp -w
creating /Users/ming/anaconda3/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.0.2.8_37_macextensionbuilder/lib.macosx-10.9-x86_64-3.7
creating /Users/ming/anaconda3/lib/python3.7/site-packages/mujoco_py/generated/_pyxbld_2.0.2.8_37_macextensionbuilder/lib.macosx-10.9-x86_64-3.7/mujoco_py
[... cut ...]
	/Users/ming/.mujoco/mujoco200/bin/libmujoco200.dylib: code signature in (/Users/ming/.mujoco/mujoco200/bin/libmujoco200.dylib) not valid for use in process using Library Validation: library load disallowed by system policy
	/Users/ming/.mujoco/mujoco200/bin/libmujoco200.dylib: code signature in (/Users/ming/.mujoco/mujoco200/bin/libmujoco200.dylib) not valid for use in process using Library Validation: library load disallowed by system policy
>>> 

```

 **does anyone has solutions or hints about the 'code signatur' or 'Validation' problem ???**

Schrominger, 2019-11-15:
> OS: [Mac OS Catalina 10.15.1]
> Python Version [ 3.7.4]
> Mujoco Version [Mujoco200]
> mujoco-py version [2.0.2.8] by `pip install mujoco_py`
>
> I added `gcc-9` into the `builder.py` file,
> the above answer [...]

well, this maybe a stupid question, I add the related `.dylib` files as security from the Security&Privacy in System Preferences and `import mujoco_py` again, all the [following code](https://github.com/openai/mujoco-py) works without errors ! Thanks !

```python
import mujoco_py
import os
mj_path, _ = mujoco_py.utils.discover_mujoco()
xml_path = os.path.join(mj_path, 'model', 'humanoid.xml')
model = mujoco_py.load_model_from_path(xml_path)
sim = mujoco_py.MjSim(model)

print(sim.data.qpos)
# [0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]

sim.step()
print(sim.data.qpos)
# [-2.09531783e-19  2.72130735e-05  6.14480786e-22 -3.45474715e-06
#   7.42993721e-06 -1.40711141e-04 -3.04253586e-04 -2.07559344e-04
#   8.50646247e-05 -3.45474715e-06  7.42993721e-06 -1.40711141e-04
#  -3.04253586e-04 -2.07559344e-04 -8.50646247e-05  1.11317030e-04
#  -7.03465386e-05 -2.22862221e-05 -1.11317030e-04  7.03465386e-05
#  -2.22862221e-05]
```

RyanRizzo96, 2019-11-15:
@Schrominger , have you tried pip install mujoco-py? @Jendker says this is now working.

Maybe try this command in a new environment, with the original SDK's and see how it goes as it should be easier to install this way.

Schrominger, 2019-11-15:
> @Schrominger , have you tried pip install mujoco-py? @Jendker says this is now working.
> 
> Maybe try this command in a new environment, with the original SDK's and see how it goes as it should be easier to install this way.

Now its OK, thanks~

Schrominger, 2019-11-15:
I make a summary here what I have done.

**system conditions:**
OS: [Mac OS Catalina 10.15.1]
Python Version [ 3.7.4]
Mujoco Version [Mujoco200]
mujoco-py version [2.0.2.8] 

1.   Anaconda-2019 installed already, so the python version=3.7.4.
2. download the `mujoco200 for macos` from [the link](https://www.roboti.us/index.html), copy the unzipped files to `~/.mujoco/mojuco200/`, files in the folder `.mujoco/` showing as 
```shell
[21:49:49] ming@Mac ~/.mujoco $ tree
.
├── mjkey.txt
└── mujoco200
    ├── bin
    │   ├── basic
    │   ├── compile
    │   ├── derivative
    │   ├── libglfw.3.dylib
    │   ├── libglfw3.a
    │   ├── libmujoco200.dylib
    │   ├── libmujoco200nogl.dylib
    │   ├── mjkey.txt
    │   ├── record
    │   ├── simulate
    │   ├── testspeed
    │   └── testxml
    ├── doc
    │   ├── README.txt
    │   └── REFERENCE.txt
    ├── include
    │   ├── glfw3.h
    │   ├── mjdata.h
    │   ├── mjmodel.h
    │   ├── mjrender.h
    │   ├── mjui.h
    │   ├── mjvisualize.h
    │   ├── mjxmacro.h
    │   ├── mujoco.h
    │   ├── uitools.c
    │   └── uitools.h
    ├── model
    │   ├── arm26.xml
    │   ├── carpet.png
    │   ├── cloth.xml
    │   ├── grid1.xml
    │   ├── grid1pin.xml
    │   ├── grid2.xml
    │   ├── grid2pin.xml
    │   ├── hammock.xml
    │   ├── humanoid.xml
    │   ├── humanoid100.xml
    │   ├── loop.xml
    │   ├── marble.png
    │   ├── particle.xml
    │   ├── rope.xml
    │   ├── scene.xml
    │   ├── softbox.xml
    │   ├── softcylinder.xml
    │   ├── softellipsoid.xml
    │   └── sponge.png
    └── sample
        ├── basic.cpp
        ├── compile.cpp
        ├── derivative.cpp
        ├── makefile
        ├── record.cpp
        ├── simulate.cpp
        ├── testspeed.cpp
        └── testxml.cpp

6 directories, 52 files
```

3.  install `mujoco_py` by using `pip`
```shell
[22:07:47] ming@Mac ~ $ pip install mujoco_py
Collecting mujoco_py
  Using cached https://files.pythonhosted.org/packages/44/f8/1c2840d5ed717ac612aba2879dc1ce548451be6e4d32a7fc7c549c127119/mujoco_py-2.0.2.8-py3-none-any.whl
Requirement already satisfied: Cython>=0.27.2 in ./anaconda3/lib/python3.7/site-packages (from mujoco_py) (0.29.13)
Requirement already satisfied: cffi>=1.10 in ./anaconda3/lib/python3.7/site-packages (from mujoco_py) (1.12.3)
Requirement already satisfied: glfw>=1.4.0 in ./anaconda3/lib/python3.7/site-packages (from mujoco_py) (1.8.4)
Requirement already satisfied: fasteners~=0.15 in ./anaconda3/lib/python3.7/site-packages (from mujoco_py) (0.15)
Requirement already satisfied: imageio>=2.1.2 in ./anaconda3/lib/python3.7/site-packages (from mujoco_py) (2.6.0)
Requirement already satisfied: numpy>=1.11 in ./anaconda3/lib/python3.7/site-packages (from mujoco_py) (1.17.2)
Requirement already satisfied: pycparser in ./anaconda3/lib/python3.7/site-packages (from cffi>=1.10->mujoco_py) (2.19)
Requirement already satisfied: monotonic>=0.1 in ./anaconda3/lib/python3.7/site-packages (from fasteners~=0.15->mujoco_py) (1.5)
Requirement already satisfied: six in ./anaconda3/lib/python3.7/site-packages (from fasteners~=0.15->mujoco_py) (1.12.0)
Requirement already satisfied: pillow in ./anaconda3/lib/python3.7/site-packages (from imageio>=2.1.2->mujoco_py) (6.2.0)
Installing collected packages: mujoco-py
Successfully installed mujoco-py-2.0.2.8
```    

4.  edit the [`builder.py` script](https://github.com/openai/mujoco-py/blob/master/mujoco_py/builder.py) to allow the `gcc-9` compiler.    
`~/anaconda3/lib/python3.7/site-packages/mujoco_py/builder.py`    
```python
...     
...            
c_compilers = ['/usr/local/bin/gcc-6',
                           '/usr/local/bin/gcc-7',
                           '/usr/local/bin/gcc-8',
                           '/usr/local/bin/gcc-9', # added here
                           '/opt/local/bin/gcc-mp-6',
                           '/opt/local/bin/gcc-mp-7',
                           '/opt/local/bin/gcc-mp-8']
...   
...      
```

then building it by `import` at the first time.   
```shell
Python 3.7.4 (default, Aug 13 2019, 15:17:50) 
[Clang 4.0.1 (tags/RELEASE_401/final)] :: Anaconda, Inc. on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>>  import mujoco_py    
```

5.   test the code [here](https://github.com/openai/mujoco-py)
```python
>>> import mujoco_py
>>> import os 
>>> mj_path, _ = mujoco_py.utils.discover_mujoco()
>>> xml_path = os.path.join(mj_path, 'model', 'humanoid.xml')
>>> model = mujoco_py.load_model_from_path(xml_path)
>>> sim = mujoco_py.MjSim(model)
>>> print(sim.data.qpos)
[0.  0.  1.4 1.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.
 0.  0.  0.  0.  0.  0.  0.  0.  0.  0. ]
>>> sim.step()
>>> print(sim.data.qpos)
[-1.12164337e-05  7.29847036e-22  1.39975300e+00  9.99999999e-01
  1.80085466e-21  4.45933954e-05 -2.70143345e-20  1.30126513e-19
 -4.63561234e-05 -1.88020744e-20 -2.24492958e-06  4.79357124e-05
 -6.38208396e-04 -1.61130312e-03 -1.37554006e-03  5.54173825e-05
 -2.24492958e-06  4.79357124e-05 -6.38208396e-04 -1.61130312e-03
 -1.37554006e-03 -5.54173825e-05 -5.73572648e-05  7.63833991e-05
 -2.12765194e-05  5.73572648e-05 -7.63833991e-05 -2.12765194e-05]
>>> 

```

 seemed it  works now.

matthiasplappert, 2019-11-25:
Addressed by https://github.com/openai/mujoco-py/pull/485

closed 2019-11-25 by matthiasplappert

RoboKez, 2020-01-09:
If anyone is still having issues I wrote some steps on how to set it up from factory settings...

================================================
MUJOCO-PY Catalina set up guide from factory settings on:

macOS Catalina 10.15.2 ,
anaconda 4.8.1 (python 3.7),
mujoco200, 
mujoco-py  2.0.2.9, 
================================================

# CONDA INSTALL ————————————————————————————————————————

Manually download anaconda 3.7 from https://www.anaconda.com/distribution/

I changed from zsh to bash with chsh -s /bin/bash

conda update conda

source ~/.bash_profile

# XCODE iNSTALL ————————————————————————————————————————

Download Xcode from App Store

Install Xcode tools with xcode-select --install

# MUJOCO INSTALL ————————————————————————————————————————

mkdir .mujoco in your home directory

Download mujoco200_macos to .mujoco (Note: cmd and shift and . Shows hidden files on a Mac)

mujoco200_macos needs renaming as mujoco200

Save your mjkey.txt to .mujoco/mujoco200/bin , this will allow the simulate test to run

Also save your mjkey.txt to .mujoco , this will allow mujoco-py to install in the environment set up step

Test mujoco (not mujoco-py) is installed by running ./simulate ../model/humanoid.xml, you will see a humanoid fall over

# RESTART ————————————————————————————————————————

This probably not required

# INSTALL HOMEBREW ————————————————————————————————————————

/usr/bin/ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"

You may have to agree to Xcode licensing if you haven’t already, if so do that and run the above command again

source ~/.bash_profile  # probably not required

# INSTALL GCC ————————————————————————————————————————

brew install gcc

# MUJOCO-PY ENV SET UP ————————————————————————————————————————

conda create --name=bambi python=3.7

conda activate bambi

Pip list is:
certifi    2019.11.28         
pip        19.3.1             
setuptools 44.0.0.post20200106
wheel      0.33.6  

pip install -U 'mujoco-py<2.1,>=2.0'

# CHECK IT WORKS ————————————————————————————————————————

Pip list is now:
certifi    2019.11.28         
cffi       1.13.2             
Cython     0.29.14            
fasteners  0.15               
glfw       1.9.1              
imageio    2.6.1              
monotonic  1.5                
mujoco-py  2.0.2.9            
numpy      1.18.1             
Pillow     7.0.0              
pip        19.3.1             
pycparser  2.19               
setuptools 44.0.0.post20200106
six        1.13.0             
wheel      0.33.6  

Script the following test.py :

# Reference: https://github.com/openai/mujoco-py
import mujoco_py
import os
mj_path, _ = mujoco_py.utils.discover_mujoco()
xml_path = os.path.join(mj_path, 'model', 'humanoid.xml')
model = mujoco_py.load_model_from_path(xml_path)
sim = mujoco_py.MjSim(model)
print(sim.data.qpos)
sim.step()
print(sim.data.qpos)

If you get something like this it is probably working :)

[0.  0.  1.4 1.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.
 0.  0.  0.  0.  0.  0.  0.  0.  0.  0. ]
[-1.12164337e-05  7.29847036e-22  1.39975300e+00  9.99999999e-01
  1.80085466e-21  4.45933954e-05 -2.70143345e-20  1.30126513e-19
 -4.63561234e-05 -1.88020744e-20 -2.24492958e-06  4.79357124e-05
 -6.38208396e-04 -1.61130312e-03 -1.37554006e-03  5.54173825e-05
 -2.24492958e-06  4.79357124e-05 -6.38208396e-04 -1.61130312e-03
 -1.37554006e-03 -5.54173825e-05 -5.73572648e-05  7.63833991e-05
 -2.12765194e-05  5.73572648e-05 -7.63833991e-05 -2.12765194e-05]

Hope this helps.

wberdanier, 2020-03-16:
Hi, I have been struggling with this problem for days. Nothing above works for me at all, I've tried everything in the above thread to no avail. I continually get 

```
error: command '/usr/local/bin/gcc-8' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
Failed to build mujoco-py
ERROR: Could not build wheels for mujoco-py which use PEP 517 and cannot be installed directly
```

pip install mujoco-py produces the above error;
cloning the repo and running setup.py works at first, but when I import mujoco_py within python, it attempts to compile and produces the above error; 
running pip install -e . within the repo produces the above error. 

I also did RyanRizzo's library hack, and it produces the above error. 

I'm completely out of ideas now. For the record, I can't get gcc to work on the command line in Catalina either, even to compile HelloWorld.cpp. 

Any help appreciated, I'm really bummed I can't get this to work.

wberdanier, 2020-03-16:
The error is enormous, but the output directly above the exit status 1 error is

`/private/var/folders/f9/1bd8dv210tzfrp55jp738j540000gn/T/pip-install-r26mqvy1/mujoco-py/mujoco_py/cymj.c:207139:22: error: unknown type name 'Py_ssize_t'
   static CYTHON_INLINE Py_ssize_t __Pyx_PyIndex_AsSsize_t(PyObject* b) {
                        ^~~~~~~~~~
  /private/var/folders/f9/1bd8dv210tzfrp55jp738j540000gn/T/pip-install-r26mqvy1/mujoco-py/mujoco_py/cymj.c:207139:57: error: expected declaration specifiers or '...' before 'PyObject'
   static CYTHON_INLINE Py_ssize_t __Pyx_PyIndex_AsSsize_t(PyObject* b) {
                                                           ^~~~~~~~
  /private/var/folders/f9/1bd8dv210tzfrp55jp738j540000gn/T/pip-install-r26mqvy1/mujoco-py/mujoco_py/cymj.c:207201:31: error: expected '=', ',', ';', 'asm' or '__attribute__' before '*' token
   static CYTHON_INLINE PyObject * __Pyx_PyBool_FromLong(long b) {
                                 ^
  /private/var/folders/f9/1bd8dv210tzfrp55jp738j540000gn/T/pip-install-r26mqvy1/mujoco-py/mujoco_py/cymj.c:207204:31: error: expected '=', ',', ';', 'asm' or '__attribute__' before '*' token
   static CYTHON_INLINE PyObject * __Pyx_PyInt_FromSize_t(size_t ival) {
                                 ^
  In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/_stdio.h:79,
                   from /usr/local/Cellar/gcc@8/8.4.0/lib/gcc/8/gcc/x86_64-apple-darwin19/8.4.0/include-fixed/stdio.h:78,
                   from /Users/will/anaconda3/include/python3.6m/Python.h:25,
                   from /private/var/folders/f9/1bd8dv210tzfrp55jp738j540000gn/T/pip-install-r26mqvy1/mujoco-py/mujoco_py/cymj.c:58:
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/stdio.h:39:5: error: old-style parameter declarations in prototyped function definition
   int renameat(int, const char *, int, const char *) __OSX_AVAILABLE_STARTING(__MAC_10_10, __IPHONE_8_0);
       ^~~~~~~~
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/stdio.h:39:14: error: parameter name omitted
   int renameat(int, const char *, int, const char *) __OSX_AVAILABLE_STARTING(__MAC_10_10, __IPHONE_8_0);
                ^~~
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/stdio.h:39:19: error: parameter name omitted
   int renameat(int, const char *, int, const char *) __OSX_AVAILABLE_STARTING(__MAC_10_10, __IPHONE_8_0);
                     ^~~~~~~~~~~~
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/stdio.h:39:33: error: parameter name omitted
   int renameat(int, const char *, int, const char *) __OSX_AVAILABLE_STARTING(__MAC_10_10, __IPHONE_8_0);
                                   ^~~
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/stdio.h:39:38: error: parameter name omitted
   int renameat(int, const char *, int, const char *) __OSX_AVAILABLE_STARTING(__MAC_10_10, __IPHONE_8_0);
                                        ^~~~~~~~~~~~
  /private/var/folders/f9/1bd8dv210tzfrp55jp738j540000gn/T/pip-install-r26mqvy1/mujoco-py/mujoco_py/cymj.c:207209: error: expected '{' at end of input
   #endif /* Py_PYTHON_H */`

wberdanier, 2020-03-16:
I've also tried aliasing g++ (which does work for compiling HelloWorld at least) into gcc, i.e. 
`alias /usr/local/bin/gcc-8="g++"`
which produces the same error.

wberdanier, 2020-03-16:
I also looked at #485, specifically uninstalled all other gcc with homebrew and installed gcc@9, and still get

`error: command '/usr/local/bin/gcc-9' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
Failed to build mujoco-py
ERROR: Could not build wheels for mujoco-py which use PEP 517 and cannot be installed directly`

wberdanier, 2020-03-19:
Just wanted to update the thread: uninstalling all older GCC versions via homebrew and then running

`brew install gcc@9`,

which installs GCC 9.3, and then running

`pip install -U mujoco-py<2.1,>=2.0`

finally worked for me. No more compiler errors. Note that you seem to need both mujoco200 and mujocopro150 for openai/spinningup.

In my above response, I had run `brew install gcc` which for some reason did not work, while `brew install gcc@9` does work, when all other versions are uninstalled.

bobyang9, 2020-03-22:
Hi, similar to wberdanier above, I have been stuck with this error for many days.

I've tried everything above as well, and got this error as expected (almost exactly the same as wberdanier's post 7 days prior):

`error: command '/usr/local/bin/gcc-8' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for mujoco-py
Failed to build mujoco-py
ERROR: Could not build wheels for mujoco-py which use PEP 517 and cannot be installed directly`

However, as recommended above, I then used homebrew to uninstall all of the older GCC versions, and then reinstalled gcc@9. However, I'm still getting a similar error:

`error: command '/usr/local/bin/gcc-9' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for mujoco-py
Failed to build mujoco-py
ERROR: Could not build wheels for mujoco-py which use PEP 517 and cannot be installed directly
`

It seems that the above fix using gcc 9 is not working for me either. 

I even tried avoiding using pep517 by using `--no-use-pep517` (since that seemed like the source of the error), but it also failed, as expected:

`error: command '/usr/local/bin/gcc-9' failed with exit status 1`

Additional note: The error message displayed when building the wheel fails is actually very large (thousands of lines); it seems that most lines of code have failed as a result. Most have this error message:

`874 | MJAPI void mju_scl(mjtNum* res, const mjtNum* vec, mjtNum scl, int n);
        |                    ^~~~~~
  /Users/user1/.mujoco/mujoco200/include/mujoco.h:874:45: error: expected ';', ',' or ')' before '*' token
    874 | MJAPI void mju_scl(mjtNum* res, const mjtNum* vec, mjtNum scl, int n);
        |                                             ^
  /Users/user1/.mujoco/mujoco200/include/mujoco.h:877:20: error: expected declaration specifiers or '...' before 'mjtNum'`

piojanu, 2020-06-29:
@bobyang9 see [my solution](https://github.com/openai/mujoco-py/issues/465#issuecomment-651124360). You may find it helpful, but it bypasses gcc (I deleted it and used llvm from brew).

null-id, 2020-07-25:
Agree with @bobyang9 . The problem has not been solved yet. Waiting for a solution.

cyclone923, 2020-09-13:
`(o2p2) yangchen@Chengxis-MacBook-Pro .mujoco % pip install -U 'mujoco-py<2.1,>=2.0'
Collecting mujoco-py<2.1,>=2.0
  Using cached mujoco-py-2.0.2.13.tar.gz (792 kB)
  Installing build dependencies ... done
  WARNING: Missing build requirements in pyproject.toml for mujoco-py<2.1,>=2.0 from https://files.pythonhosted.org/packages/2f/48/b108057c1a23c8da9f4cdc7a7c46ab7cec49c3563c0706d50f2527de6ba0/mujoco-py-2.0.2.13.tar.gz#sha256=d6ae66276b565af9063597fda70683a89c7356290f5ac3961b794ee90ec50eea.
  WARNING: The project does not specify a build backend, and pip cannot fall back to setuptools without 'wheel'.
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
    Preparing wheel metadata ... done
Requirement already satisfied, skipping upgrade: imageio>=2.1.2 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (2.9.0)
Requirement already satisfied, skipping upgrade: glfw>=1.4.0 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (1.12.0)
Requirement already satisfied, skipping upgrade: Cython>=0.27.2 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (0.29.21)
Requirement already satisfied, skipping upgrade: fasteners~=0.15 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (0.15)
Requirement already satisfied, skipping upgrade: numpy>=1.11 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (1.16.1)
Requirement already satisfied, skipping upgrade: cffi>=1.10 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (1.14.2)
Requirement already satisfied, skipping upgrade: pillow in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from imageio>=2.1.2->mujoco-py<2.1,>=2.0) (7.2.0)
Requirement already satisfied, skipping upgrade: six in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from fasteners~=0.15->mujoco-py<2.1,>=2.0) (1.15.0)
Requirement already satisfied, skipping upgrade: monotonic>=0.1 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from fasteners~=0.15->mujoco-py<2.1,>=2.0) (1.5)
Requirement already satisfied, skipping upgrade: pycparser in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from cffi>=1.10->mujoco-py<2.1,>=2.0) (2.20)
Building wheels for collected packages: mujoco-py
  Building wheel for mujoco-py (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /Users/yangchen/opt/miniconda3/envs/o2p2/bin/python /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/tmp2udd9q81
       cwd: /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py
  Complete output (27 lines):
  running bdist_wheel
  running build
  Removing old mujoco_py cext /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/cymj_2.0.2.13_37_macextensionbuilder_37.so
  Compiling /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/cymj.pyx because it depends on /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-build-env-yc969w4s/overlay/lib/python3.7/site-packages/Cython/Includes/libc/string.pxd.
  [1/1] Cythonizing /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/l2
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T
[... cut ...] -x86_64-3.7/mujoco_py/cymj.cpython-37m-darwin.so -fopenmp
  ld: library not found for -lmujoco200
  collect2: error: ld returned 1 exit status
  error: command '/usr/local/bin/gcc-6' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for mujoco-py
Failed to build mujoco-py
ERROR: Could not build wheels for mujoco-py which use PEP 517 and cannot be installed directly
`

Anybody knows how to fix the 
`  ld: library not found for -lmujoco200`
Thanks!

cyclone923, 2020-09-13:
> Agree with @bobyang9 . The problem has not been solved yet. Waiting for a solution.

(o2p2) yangchen@Chengxis-MacBook-Pro .mujoco % 
pip install -U 'mujoco-py<2.1,>=2.0'
Collecting mujoco-py<2.1,>=2.0
  Using cached mujoco-py-2.0.2.13.tar.gz (792 kB)
  Installing build dependencies ... done
  WARNING: Missing build requirements in pyproject.toml for mujoco-py<2.1,>=2.0 from https://files.pythonhosted.org/packages/2f/48/b108057c1a23c8da9f4cdc7a7c46ab7cec49c3563c0706d50f2527de6ba0/mujoco-py-2.0.2.13.tar.gz#sha256=d6ae66276b565af9063597fda70683a89c7356290f5ac3961b794ee90ec50eea.
  WARNING: The project does not specify a build backend, and pip cannot fall back to setuptools without 'wheel'.
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
    Preparing wheel metadata ... done
Requirement already satisfied, skipping upgrade: imageio>=2.1.2 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (2.9.0)
Requirement already satisfied, skipping upgrade: glfw>=1.4.0 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (1.12.0)
Requirement already satisfied, skipping upgrade: Cython>=0.27.2 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (0.29.21)
Requirement already satisfied, skipping upgrade: fasteners~=0.15 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (0.15)
Requirement already satisfied, skipping upgrade: numpy>=1.11 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (1.16.1)
Requirement already satisfied, skipping upgrade: cffi>=1.10 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (1.14.2)
Requirement already satisfied, skipping upgrade: pillow in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from imageio>=2.1.2->mujoco-py<2.1,>=2.0) (7.2.0)
Requirement already satisfied, skipping upgrade: six in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from fasteners~=0.15->mujoco-py<2.1,>=2.0) (1.15.0)
Requirement already satisfied, skipping upgrade: monotonic>=0.1 in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from fasteners~=0.15->mujoco-py<2.1,>=2.0) (1.5)
Requirement already satisfied, skipping upgrade: pycparser in /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages (from cffi>=1.10->mujoco-py<2.1,>=2.0) (2.20)
Building wheels for collected packages: mujoco-py
  Building wheel for mujoco-py (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /Users/yangchen/opt/miniconda3/envs/o2p2/bin/python /Users/yangchen/opt/miniconda3/envs/o2p2/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/tmp2udd9q81
       cwd: /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py
  Complete output (27 lines):
  running bdist_wheel
  running build
  Removing old mujoco_py cext /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/cymj_2.0.2.13_37_macextensionbuilder_37.so
  Compiling /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/cymj.pyx because it depends on /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-build-env-yc969w4s/overlay/lib/python3.7/site-packages/Cython/Includes/libc/string.pxd.
  [1/1] Cythonizing /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/l2
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn
  creating /private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T/pip-install-ts_wwb1k/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.13_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/l2/99n582vs6fj7kzgd3ysj2l2m0000gn/T
[... cut ...] _macextensionbuilder/lib.macosx-10.9-x86_64-3.7/mujoco_py/cymj.cpython-37m-darwin.so -fopenmp

 ld: library not found for -lmujoco200
 collect2: error: ld returned 1 exit status
 error: command '/usr/local/bin/gcc-6' failed with exit status 1
 ----------------------------------------
ERROR: Failed building wheel for mujoco-py
Failed to build mujoco-py
ERROR: Could not build wheels for mujoco-py which use PEP 517 and cannot be installed directly

Anybody nows how to solve this 'ld' issue?
Thanks.

raymondchua, 2020-12-27:
For me, it was the security and privacy issues on the dylib files on mujoco (namely, libglfw.3, libmujoco200 and libmujoco200nogl). I have to manually open the files to override the security permissions that came with macOS mojave. After which, I followed @RoboKez 's steps which helped!

quinnliu, 2021-01-10:
@raymondchua Thanks for the suggestion. How did you manually open the files? Did you use chmod to change permissions?

quinnliu, 2021-01-10:
@ new all new readers. I've tried all of the above strategies on MacOS 10.15.7 inside a python 3.7.3 virtual environment with pip 20.3.3 && nothing has worked. Would appreciate an actual solution from the founders of this repo for MacOS 10.15.7 Thanks

raymondchua, 2021-01-11:
> @raymondchua Thanks for the suggestion. How did you manually open the files? Did you use chmod to change permissions on MacOS 10.15.7?

I just did right click and choose open. You should see a window opening up informing you on the security issues.

mg19950, 2023-04-03:
`/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c: In function '__pyx_f_9mujoco_py_4cymj_9PyMjModel__set':
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c:32017:78: error: 'mjModel {aka struct _mjModel}' has no member named 'key_mpos'; did you mean 'key_qpos'?
   __pyx_t_1 = ((PyObject *)__pyx_f_9mujoco_py_4cymj__wrap_mjtNum_2d(__pyx_v_p->key_mpos, __pyx_v_p->nkey, (3 * __pyx_v_p->nmocap))); if (unlikely(!__pyx_t_1)) __PYX_ERR(4, 1494, __pyx_L1_error)
                                                                              ^~
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c:32032:78: error: 'mjModel {aka struct _mjModel}' has no member named 'key_mquat'; did you mean 'body_quat'?
   __pyx_t_1 = ((PyObject *)__pyx_f_9mujoco_py_4cymj__wrap_mjtNum_2d(__pyx_v_p->key_mquat, __pyx_v_p->nkey, (4 * __pyx_v_p->nmocap))); if (unlikely(!__pyx_t_1)) __PYX_ERR(4, 1495, __pyx_L1_error)
                                                                              ^~
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c: In function '__pyx_pf_9mujoco_py_4cymj_12PyMjvPerturb_7active2___get__':
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c:70069:53: error: 'mjvPerturb {aka struct _mjvPerturb}' has no member named 'active2'; did you mean 'active'?
   __pyx_t_1 = __Pyx_PyInt_From_int(__pyx_v_self->ptr->active2); if (unlikely(!__pyx_t_1)) __PYX_ERR(4, 3352, __pyx_L1_error)
                                                     ^~
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c: In function '__pyx_pf_9mujoco_py_4cymj_12PyMjvPerturb_7active2_2__set__':
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c:70124:20: error: 'mjvPerturb {aka struct _mjvPerturb}' has no member named 'active2'; did you mean 'active'?
   __pyx_v_self->ptr->active2 = __pyx_v_x;
                    ^~
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c: In function '__pyx_f_9mujoco_py_4cymj_10PyMjvScene__set':
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c:76679:80: error: 'mjvScene {aka struct _mjvScene}' has no member named 'framergb'; did you mean 'camera'?
   __pyx_t_1 = ((PyObject *)__pyx_f_9mujoco_py_4cymj__wrap_float_1d((&(__pyx_v_p->framergb[0])), 3)); if (unlikely(!__pyx_t_1)) __PYX_ERR(4, 3743, __pyx_L1_error)
                                                                                ^~
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c: In function '__pyx_pf_9mujoco_py_4cymj_10PyMjvScene_10framewidth___get__':
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c:77386:53: error: 'mjvScene {aka struct _mjvScene}' has no member named 'framewidth'
   __pyx_t_1 = __Pyx_PyInt_From_int(__pyx_v_self->ptr->framewidth); if (unlikely(!__pyx_t_1)) __PYX_ERR(4, 3774, __pyx_L1_error)
                                                     ^~
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c: In function '__pyx_pf_9mujoco_py_4cymj_10PyMjvScene_10framewidth_2__set__':
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c:77441:20: error: 'mjvScene {aka struct _mjvScene}' has no member named 'framewidth'
   __pyx_v_self->ptr->framewidth = __pyx_v_x;
                    ^~
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c: In function '__pyx_pf_9mujoco_py_4cymj_11PyMjvFigure_9linewidth___get__':
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c:78821:34: error: incompatible type for argument 1 of 'PyFloat_FromDouble'
   __pyx_t_1 = PyFloat_FromDouble(__pyx_v_self->ptr->linewidth); if (unlikely(!__pyx_t_1)) __PYX_ERR(4, 3878, __pyx_L1_error)
                                  ^~~~~~~~~~~~
In file included from /Library/Frameworks/Python.framework/Versions/3.11/include/python3.11/Python.h:55:0,
                 from /Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c:63:
/Library/Frameworks/Python.framework/Versions/3.11/include/python3.11/floatobject.h:39:23: note: expected 'double' but argument is of type 'float *'
 PyAPI_FUNC(PyObject*) PyFloat_FromDouble(double);
                       ^~~~~~~~~~~~~~~~~~
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c: In function '__pyx_pf_9mujoco_py_4cymj_11PyMjvFigure_9linewidth_2__set__':
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c:78876:32: error: assignment to expression with array type
   __pyx_v_self->ptr->linewidth = __pyx_v_x;
                                ^
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c: In function '__pyx_pf_9mujoco_py_4cymj_11PyMjvFigure_12legendoffset___get__':
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c:79007:53: error: 'mjvFigure {aka struct _mjvFigure}' has no member named 'legendoffset'; did you mean 'legendoff'?
   __pyx_t_1 = __Pyx_PyInt_From_int(__pyx_v_self->ptr->legendoffset); if (unlikely(!__pyx_t_1)) __PYX_ERR(4, 3886, __pyx_L1_error)
                                                     ^~
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c: In function '__pyx_pf_9mujoco_py_4cymj_11PyMjvFigure_12legendoffset_2__set__':
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c:79062:20: error: 'mjvFigure {aka struct _mjvFigure}' has no member named 'legendoffset'; did you mean 'legendoff'?
   __pyx_v_self->ptr->legendoffset = __pyx_v_x;
                    ^~
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c: In function '__pyx_pf_9mujoco_py_4cymj_11PyMjvFigure_7subplot___get__':
/Users/mansigupta/Easy21/mujoco-py/mujoco_py/cymj.c:79100:53: error: 'mjvFigure {aka struct _mjvFigure}' has no member named 'subplot'
[... cut ...]
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/setuptools/_distutils/ccompiler.py", line 599, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/setuptools/_distutils/unixccompiler.py", line 188, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command '/usr/local/bin/gcc-6' failed with exit code 1`

I am getting this error when trying to import mujoco_py on catalina. I installed mujoco200