openai/mujoco-py #464: AttributeError: module 'mujoco_py' has no attribute 'utils'

superggb, 2019-10-11:
**Describe the bug**
    Traceback (most recent call last):
    File "test_mujoco.py", line 3, in <module>
    mj_path, _ = mujoco_py.utils.discover_mujoco()

**To Reproduce**
I have installed gym and mujoco and mujoco_py all on the step of the web
I can run the gym test successfully and run the mujoco example sucessfully too.
But when I run the test-script in the Code web ,it appeared the BUG

**Error Messages**
AttributeError: module 'mujoco_py' has no attribute 'utils'

**Desktop (please complete the following information):**
 - OS: [ubuntu 16.04]
 - Python Version [ 3.6.8]
 - Mujoco Version [MJPRO131]
 - mujoco-py version [0.5.7]

**Additional context**
I use the software Version according a project's code, so I think I have successfully install mujoco and gym.
What's the problem with mujoco-py.......

leiyue812, 2021-05-14:
I have exact the same issue. Did you solve them?

alex-rudenkiy, 2022-06-04:
import mujoco_py
from mujoco_py import utils #<-------
import os

mj_path = utils.discover_mujoco()
xml_path = os.path.join(mj_path, 'model', 'humanoid.xml')
model = mujoco_py.load_model_from_path(xml_path)
sim = mujoco_py.MjSim(model)
print(sim.data.qpos)

sim.step()
print(sim.data.qpos)