openai/mujoco-py #465: Command line tools and gcc errors during installation on macOS Catalina

RyanRizzo96, 2019-10-11:
**Describe the bug**
Command Line tools and gcc are giving a multitude of errors when trying to install MuJoCo. I suspect with high confidence that this is an issue from the command line tools on my macOS Catalina since I have installed mujoco previously on Mojave and multiple ubuntu machines.

**To Reproduce**
pip3 install 'mujoco-py<2.1,>=2.0' on Catalina

**Expected behavior**
Installs successfully.

**Error Messages**
  > /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/resource.h:443:34:
    > error: expected declaration specifiers before
    > '__OSX_AVAILABLE_STARTING'    int     getiopolicy_np(int, int)
    > __OSX_AVAILABLE_STARTING(__MAC_10_5, __IPHONE_2_0);
    >                                     ^~~~~~~~~~~~~~~~~~~~~~~~

and

      /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:144:1: error: expected declaration specifiers before 'div_t'
       div_t  div(int, int) __pure2;
       ^~~~~
and 

```
In file included from /Users/ryanr/.mujoco/mujoco200/include/mujoco.h:42:0,
                   from /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-4_wykvnc/mujoco-py/mujoco_py/gl/glshim.h:4,
                   from /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-4_wykvnc/mujoco-py/mujoco_py/gl/dummyshim.c:1:
  /Users/ryanr/.mujoco/mujoco200/include/mjui.h:54:3: error: storage class specified for parameter 'mjtButton'
   } mjtButton;
     ^~~~~~~~~
  /Users/ryanr/.mujoco/mujoco200/include/mjui.h:66:3: error: storage class specified for parameter 'mjtEvent'
   } mjtEvent;
     ^~~~~~~~
  /Users/ryanr/.mujoco/mujoco200/include/mjui.h:89:3: error: storage class specified for parameter 'mjtItem'
   } mjtItem;
     ^~~~~~~
  /Users/ryanr/.mujoco/mujoco200/include/mjui.h:93:15: error: storage class specified for parameter 'mjfItemEnable'
   typedef int (*mjfItemEnable)(int category, void* data);
                 ^~~~~~~~~~~~~
  /Users/ryanr/.mujoco/mujoco200/include/mjui.h:100:5: error: expected specifier-qualifier-list before 'mjrRect'
       mjrRect rect[mjMAXUIRECT];  // rectangles (index 0: entire window)
       ^~~~~~~
```

**Desktop (please complete the following information):**
 - OS: [e.g. macOS 10.15]
 - Python Version [e.g. 3.7]
 - Mujoco Version [e.g. 2.00]
 - mujoco-py version [e.g. 2.0.5.2]

andyk, 2019-11-21:
Bump. I'm having same issues. Thanks @RyanRizzo96 for doing all the legwork for reporting this so far.

RyanRizzo96, 2019-11-22:
#254

RyanRizzo96, 2019-11-22:
Have you seen issue #463?

Jendker, 2019-11-25:
Million thanks @matthiasplappert! This closes this issue and all the related issues with MacOS Catalina created by @RyanRizzo96

closed 2019-11-25 by RyanRizzo96

YangRui2015, 2020-01-21:
Same problem. I used Python 3.6.6, mujoco-py 2.0.2.9, mujoco200, gcc 8.2 and my MuJoCo-py was installed by source. When I imported mujoco_py, I get the same error. However, after I updated gcc to 9.2.0, it disappeared.

The following settings works for me.

OS: [MacOS Catallina 10.15.2]
Python Version [3.6.6]
Mujoco Version [mujoco200]
mujoco-py version [2.0.2.9]
gcc version [9.2.0]

piojanu, 2020-04-02:
I have the same problem with gcc 9.3.0 :( Help!

### Configuration:

OS: [MacOS Catallina 10.15.3]
Python Version [3.6.9/3.7.7]
Mujoco Version [mujoco200]
mujoco-py version [2.0.2.9]
gcc version [9.3.0]

### Trace:

```
➜ pip install mujoco-py
Collecting mujoco-py
  Using cached mujoco-py-2.0.2.9.tar.gz (777 kB)
  Installing build dependencies ... done
  WARNING: Missing build requirements in pyproject.toml for mujoco-py from https://files.pythonhosted.org/packages/a2/30/21abd0cf2734bf5f34a7a8967789b12dee55f1e51e9c1c60af1cba549123/mujoco-py-2.0.2.9.tar.gz#sha256=6ae20ca9509203758f5e30a7a4019cb2d581b6d40dc2c2669dbe3229cfdf05e8.
  WARNING: The project does not specify a build backend, and pip cannot fall back to setuptools without 'wheel'.
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
    Preparing wheel metadata ... done
```
.
.
.
```
Building wheels for collected packages: mujoco-py
  Building wheel for mujoco-py (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /Users/piotr/anaconda3/envs/SpinUp/bin/python /Users/piotr/anaconda3/envs/SpinUp/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/tmphk6wwt2c
       cwd: /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py
  Complete output (2377 lines):
  running bdist_wheel
  running build
  Removing old mujoco_py cext /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py/generated/cymj_2.0.2.9_37_macextensionbuilder_37.so
  Compiling /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py/cymj.pyx because it depends on /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-build-env-xp2haxum/overlay/lib/python3.7/site-packages/Cython/Includes/numpy/__init__.pxd.
  [1/1] Cythonizing /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.9_37_macextensionbuilder
  creating /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.9_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7
  creating /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.9_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private
  creating /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.9_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var
  creating /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.9_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders
  creating /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.9_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/9l
  creating /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.9_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn
  creating /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.9_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T
  creating /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.9_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493
  creating /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.9_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py
  creating /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.9_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py
  creating /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.9_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py/gl
  /usr/local/bin/gcc-9 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Users/piotr/anaconda3/envs/SpinUp/include -arch x86_64 -I/Users/piotr/anaconda3/envs/SpinUp/include -arch x86_64 -DONMAC -Imujoco_py -I/private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py -I/Users/piotr/.mujoco/mujoco200/include -I/private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-build-env-xp2haxum/overlay/lib/python3.7/site-packages/numpy/core/include -I/Users/piotr/anaconda3/envs/SpinUp/include/python3.7m -c /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py [... cut ...]
    443 | int     getiopolicy_np(int, int) __OSX_AVAILABLE_STARTING(__MAC_10_5, __IPHONE_2_0);
        |                             ^~~
  /private/var/folders/9l/r3_05sd15p7fn3vf7t8wd47m0000gn/T/pip-install-wkh0_493/mujoco-py/mujoco_py/gl/dummyshim.c:37: error: expected '{' at end of input
     37 | }
        |
  error: command '/usr/local/bin/gcc-9' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for mujoco-py
Failed to build mujoco-py
ERROR: Could not build wheels for mujoco-py which use PEP 517 and cannot be installed directly
```

ahmarz, 2020-05-11:
@piojanu any updates ? I have the same issue

piojanu, 2020-05-13:
@ahmarz I just gave up.

piojanu, 2020-06-29:
@ahmarz I found a solution. Simply remove gcc (I couldn't make it work, I couldn't downgrade to some working version, I simply gave up with it, for your sanity health delete it) and follow [this](https://github.com/openai/mujoco-py#youre-on-macos-and-you-see-clang-error-unsupported-option--fopenmp) WA.

Step by step (based on WA from the link):
1. Uninstall gcc: `brew uninstall gcc`
You may need to run it a couple of times if you have more than one version.
2. Install clang/llvm: `brew install llvm boost hdf5`
I don't know if `boost` and `hdf5` are required, but I installed it as ordered.
3. Add this to your `.bashrc`/`.zshrc`:
```
export PATH="/usr/local/opt/llvm/bin:$PATH"
export CC="/usr/local/opt/llvm/bin/clang"
export CXX="/usr/local/opt/llvm/bin/clang++"
export CXX11="/usr/local/opt/llvm/bin/clang++"
export CXX14="/usr/local/opt/llvm/bin/clang++"
export CXX17="/usr/local/opt/llvm/bin/clang++"
export CXX1X="/usr/local/opt/llvm/bin/clang++"
export LDFLAGS="-L/usr/local/opt/llvm/lib"
export CPPFLAGS="-I/usr/local/opt/llvm/include"
```
4. Don't forget to source your `.bashrc`/`.zshrc` after editing it and make sure your python environment is activated.
5. Try to install mujoco-py again:
```
$ pip install -U 'mujoco-py<2.1,>=2.0'
$ python -c "import mujoco_py"
```

Voilà!

null-id, 2020-07-25:
Hi, I used your solution on MacOS Catalina. But still facing a problem like https://github.com/openai/mujoco-py/issues/495 @piojanu

piojanu, 2020-07-25:
@null-id I used Anaconda with Python 3.7.7. Moreover, I don't think these warnings from the issue are a problem. I see them too and I'm still able to use MuJoCo OpenAI Gym environments. See my ipython session example:
```
Python 3.7.7 (default, Mar 26 2020, 10:32:53)
Type 'copyright', 'credits' or 'license' for more information
IPython 7.13.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import mujoco_py
objc[99575]: Class GLFWApplicationDelegate is implemented in both /Users/piotr/.mujoco/mjpro150/bin/libglfw.3.dylib (0x11781f778) and /Users/piotr/anaconda3/envs/SpinUp/lib/python3.7/site-packages/glfw-1.11.0-py3.7-macosx-10.9-x86_64.egg/glfw/libglfw.3.dylib (0x1178f36e8). One of the two will be used. Which one is undefined.
objc[99575]: Class GLFWWindowDelegate is implemented in both /Users/piotr/.mujoco/mjpro150/bin/libglfw.3.dylib (0x11781f700) and /Users/piotr/anaconda3/envs/SpinUp/lib/python3.7/site-packages/glfw-1.11.0-py3.7-macosx-10.9-x86_64.egg/glfw/libglfw.3.dylib (0x1178f3710). One of the two will be used. Which one is undefined.
objc[99575]: Class GLFWContentView is implemented in both /Users/piotr/.mujoco/mjpro150/bin/libglfw.3.dylib (0x11781f7a0) and /Users/piotr/anaconda3/envs/SpinUp/lib/python3.7/site-packages/glfw-1.11.0-py3.7-macosx-10.9-x86_64.egg/glfw/libglfw.3.dylib (0x1178f3760). One of the two will be used. Which one is undefined.
objc[99575]: Class GLFWWindow is implemented in both /Users/piotr/.mujoco/mjpro150/bin/libglfw.3.dylib (0x11781f818) and /Users/piotr/anaconda3/envs/SpinUp/lib/python3.7/site-packages/glfw-1.11.0-py3.7-macosx-10.9-x86_64.egg/glfw/libglfw.3.dylib (0x1178f37d8). One of the two will be used. Which one is undefined.

In [2]: import gym

In [3]: env = gym.make('HalfCheetah-v2')

In [4]: env.reset()
Out[4]:
array([ 0.0228056 , -0.01409458,  0.07844352,  0.04524467,  0.09728677,
        0.04553284, -0.07546735, -0.04468718, -0.01352164,  0.18370665,
       -0.15241323, -0.02871676, -0.05958594, -0.08736752,  0.17731656,
        0.0478807 , -0.03618564])
```

null-id, 2020-07-25:
Thanks @piojanu . I just realized that my real problem is using a wrong license file. Now it's OK!

quinnliu, 2021-01-10:
@piojanu Your solution appears to be working on MacOS 10.15.7 with python 3.7.3 in virtual environment && pip 20.3.3 

No other solutions I've found have worked. I do not know why but I think uninstalling all versions of gcc several times was the missing step that prevented your strategy from working the first time.