openai/mujoco-py #473: PID control for shadow robot like hand does not work

Jendker, 2019-10-25:
**Describe the bug**
The P and PID control does not behave correctly on the prepared model. P control does not behave like position control, and PID behaves in the same wrong way as P control.

**To Reproduce**
Please use prepared repository: https://github.com/Jendker/pid_shadow_robot The setup from repository was created according to the file: https://github.com/openai/mujoco-py/blob/83759c2e20df965dcf727aa79e590a0483aaea0f/mujoco_py/tests/test_pid.py

**Expected behavior**
For the P control we should observe the same behaviour as for position control. The PID should keep the fingers in place and prevent fingers falling due to the gravity influence.

**Error Messages**
None

**Desktop (please complete the following information):**
 - OS: Ubunut 18.04
 - Python Version: 3.6.8
 - Mujoco Version: 2.0
 - mujoco-py version 2.0.2.7

**Environment**
 - output of: `echo $LD_LIBRARY_PATH` /home/jedrzej/catkin_ws/devel/lib:/opt/ros/melodic/lib:/home/jedrzej/.mujoco/mujoco200/bin:/home/jedrzej/.mujoco/mujoco200/bin
 - output of: `echo $HOME` /home/jedrzej
 - output of: `echo $USER` jedrzej

**Additional context**
Issue mentioned in https://github.com/openai/mujoco-py/pull/462

bayesian, 2019-10-25:
you have to set biastype="user" and gaintype="user", as our pid implementation use user-defined callback: mjcb_act_gain and mjcb_act_bias to implement full PID control.

you can read more about mujoco actuator model here: http://www.mujoco.org/book/XMLreference.html#actuator

change your xml here: https://github.com/Jendker/pid_shadow_robot/blob/master/DAPG_relocate_P_control.xml#L80
and try it out, I will close out after you verify it works.

bayesian, 2019-10-25:
I debugged a bit your branch, I think that you need to change settings here
https://github.com/Jendker/pid_shadow_robot/blob/master/dependencies/Adroit/resources/joint_position_actuation.xml

Jendker, 2019-10-25:
Thank you for your answer @bayesian !

I am sorry, I have added to many files, even some which were not used at all, so the example was not easy to follow.
Now the file which you have mentioned has been removed. Could you please have a look at this issue again? https://github.com/Jendker/pid_shadow_robot

Jendker, 2019-10-28:
Should simplify the model even further for easier debugging?

bayesian, 2019-10-28:
@Jendker, I could repro from your branch, but not sure about the cause. I have some hypotheses but no much time to debug through. 

I am fairly certain that is not the issue in mujoco-py, as we have tested PID quite thoroughly inside our team. Switch from position to P only and passed all internal unit tests; trained new control policy with full PID and physically tested new policy. The included unit test test_pid.py also tested out that behavior.

Jendker, 2019-10-28:
Thank you @bayesian, and yes I have played out a bit with the unit tests and they look just impeccable. I am unable to find where the problem is...

Could you maybe share your shadow robot hand model so that it will be easier for me to debug this issue for me?

bayesian, 2019-10-28:
@Jendker sorry that I could not share with you the shadow hand model, and it should be standard. I think the problem is most likely in your mujoco world model file.  The behavior like the whole hand dropping indicates that too, fixing that perhaps could help solve your issue.

bayesian, 2019-11-07:
@Jendker Hey, after I add forcerange into your actuator like

<general forcerange="-100 100" ctrllimited="true" ctrlrange="-1 1" gaintype="user" biastype="user" gainprm="1"/>   
<general forcerange="-1000 1000" ctrllimited="true" gaintype="user" biastype="user" gainprm="500"/>   

then it is kind of working, but seems to move much slower

Jendker, 2019-11-07:
Thank you very much for taking your time and debugging the problem! It solved the issue :)
I'm happy that I will be able to use the PID directly using mujoco-py, it makes things way easier and cleaner.

Now I only need to find the proper PID parameters for Shadow Robot hand :D

bayesian, 2019-11-07:
let me know how & what you find. :)

closed 2019-11-07 by bayesian