openai/mujoco-py #474: [cannot install mujoco-py: ]cannot find -lGL

SieRaX, 2019-11-03:
Hello, I am trying to install mujoco-py in my Linux LTS 18.04 but having serious errors.
I followed all the instructions written in the Readme.md 
```
 matthewcool14@ubuntu001:~/Main/mujoco-py$ pip3 install -U 'mujoco-py<2.1,>=2.0'
Collecting mujoco-py<2.1,>=2.0

```
but got this kind of error

```
Collecting mujoco-py<2.1,>=2.0
  Using cached https://files.pythonhosted.org/packages/c3/a1/00f5482d6325565f35a81707b8c6b3e364bdc8f7b9487c15c18781e1432a/mujoco-py-2.0.2.7.tar.gz
  Installing build dependencies ... done
  WARNING: Missing build requirements in pyproject.toml for mujoco-py<2.1,>=2.0 from https://files.pythonhosted.org/packages/c3/a1/00f5482d6325565f35a81707b8c6b3e364bdc8f7b9487c15c18781e1432a/mujoco-py-2.0.2.7.tar.gz#sha256=2a294006c66ed225c3409b9d60a91afeb7756f96fecf3d8c22116d00adf6d82a.
  WARNING: The project does not specify a build backend, and pip cannot fall back to setuptools without 'wheel'.
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
    Preparing wheel metadata ... done
Requirement already satisfied, skipping upgrade: glfw>=1.4.0 in /home/matthewcool14/anaconda3/envs/my_env/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (1.8.3)
Requirement already satisfied, skipping upgrade: numpy>=1.11 in /home/matthewcool14/anaconda3/envs/my_env/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (1.17.2)
Collecting fasteners~=0.15
  Using cached https://files.pythonhosted.org/packages/18/bd/55eb2d6397b9c0e263af9d091ebdb756b15756029b3cededf6461481bc63/fasteners-0.15-py2.py3-none-any.whl
Requirement already satisfied, skipping upgrade: imageio>=2.1.2 in /home/matthewcool14/anaconda3/envs/my_env/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (2.5.0)
Requirement already satisfied, skipping upgrade: Cython>=0.27.2 in /home/matthewcool14/anaconda3/envs/my_env/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (0.29.13)
Requirement already satisfied, skipping upgrade: cffi>=1.10 in /home/matthewcool14/anaconda3/envs/my_env/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (1.12.3)
Requirement already satisfied, skipping upgrade: six in /home/matthewcool14/anaconda3/envs/my_env/lib/python3.7/site-packages (from fasteners~=0.15->mujoco-py<2.1,>=2.0) (1.12.0)
Collecting monotonic>=0.1
  Using cached https://files.pythonhosted.org/packages/ac/aa/063eca6a416f397bd99552c534c6d11d57f58f2e94c14780f3bbf818c4cf/monotonic-1.5-py2.py3-none-any.whl
Requirement already satisfied, skipping upgrade: pillow in /home/matthewcool14/anaconda3/envs/my_env/lib/python3.7/site-packages (from imageio>=2.1.2->mujoco-py<2.1,>=2.0) (6.2.0)
Requirement already satisfied, skipping upgrade: pycparser in /home/matthewcool14/anaconda3/envs/my_env/lib/python3.7/site-packages (from cffi>=1.10->mujoco-py<2.1,>=2.0) (2.19)
Building wheels for collected packages: mujoco-py
  Building wheel for mujoco-py (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/matthewcool14/anaconda3/envs/my_env/bin/python /home/matthewcool14/anaconda3/envs/my_env/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /tmp/tmp3jlgnyvr
       cwd: /tmp/pip-install-bjmdj8fw/mujoco-py
  Complete output (22 lines):
  running bdist_wheel
  running build
  Removing old mujoco_py cext /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/generated/cymj_2.0.2.7_37_linuxcpuextensionbuilder_37.so
  Compiling /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/cymj.pyx because it depends on /tmp/pip-build-env-q5v466o4/overlay/lib/python3.7/site-packages/Cython/Includes/numpy/__init__.pxd.
  [1/1] Cythonizing /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxcpuextensionbuilder
  creating /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7
  creating /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp
  creating /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-bjmdj8fw
  creating /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-bjmdj8fw/mujoco-py
  creating /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py
  creating /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/gl
  gcc -pthread -B /home/matthewcool14/anaconda3/envs/my_env/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py -I/home/matthewcool14/.mujoco/mujoco200/include -I/tmp/pip-build-env-q5v466o4/overlay/lib/python3.7/site-packages/numpy/core/include -I/home/matthewcool14/anaconda3/envs/my_env/include/python3.7m -c /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/cymj.o -fopenmp -w
  gcc -pthread -B /home/matthewcool14/anaconda3/envs/my_env/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py -I/home/matthewcool14/.mujoco/mujoco200/include -I/tmp/pip-build-env-q5v466o4/overlay/lib/python3.7/site-packages/numpy/core/include -I/home/matthewcool14/anaconda3/envs/my_env/include/python3.7m -c /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/gl/osmesashim.c -o /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
  creating /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxcpuextensionbuilder/lib.linux-x86_64-3.7
  creating /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxcpuextensionbuilder/lib.linux-x86_64-3.7/mujoco_py
  gcc -pthread -shared -B /home/matthewcool14/anaconda3/envs/my_env/compiler_compat -L/home/matthewcool14/anaconda3/envs/my_env/lib -Wl,-rpath=/home/matthewcool14/anaconda3/envs/my_env/lib -Wl,--no-as-needed -Wl,--sysroot=/ /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/cymj.o /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/gl/osmesashim.o -L/home/matthewcool14/.mujoco/mujoco200/bin -Wl,-R/home/matthewcool14/.mujoco/mujoco200/bin -lmujoco200 -lglewosmesa -lOSMesa -lGL -o /tmp/pip-install-bjmdj8fw/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxcpuextensionbuilder/lib.linux-x86_64-3.7/mujoco_py/cymj.cpython-37m-x86_64-linux-gnu.so -fopenmp
  /home/matthewcool14/anaconda3/envs/my_env/compiler_compat/ld: cannot find -lGL
  collect2: error: ld returned 1 exit status
  error: command 'gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
Failed to build mujoco-py
ERROR: Could not build wheels for mujoco-py which use PEP 517 and cannot be installed directly

```
I googled possible solutions but none of it worked. what is lGL and how can I solve this issue?

Jendker, 2019-11-04:
Have you tried building in venv virtual environment, not in anaconda environment? If works with venv try to ask the anaconda people for help.

ThitherShore, 2019-11-08:
> Have you tried building in venv virtual environment, not in anaconda environment? If works with venv try to ask the anaconda people for help.

I am using vitrualenv, on Ubuntu 16.04. But unfortunately facing the same issue "...Could not build wheels for mujoco-py which use PEP 517...".

Jendker, 2019-11-08:
@ThitherShore Have you followed the steps from README: "Ubuntu installation troubleshooting"? 
Always try to read Readme first before submitting the issue. Quite often it has the answer to you problem.

RyanRizzo96, 2019-11-24:
Same issue here with Conda.

sroubert, 2019-12-03:
is there anything for this? this is extremely frustrating. macOS.

rominamir, 2019-12-25:
I am facing a same problem with windows 10.

jianghaoyuan1994, 2020-01-13:
I have this problem too in mujoco-py1.5

seyeeet, 2020-08-25:
any solution?

tinyzqh, 2020-10-26:
i fixed it using：
`apt install libgl1-mesa-glx --reinstall`

reference：https://github.com/openai/mujoco-py/issues/374

BeBraveBeCurious, 2022-05-17:
Is there any recommended solution without root or sudo? Thank you.

hlhllh, 2022-05-17:
This works for me:
 `sudo apt-get install libgl1-mesa-dev`