openai/mujoco-py #475: SET PATH ERROR in Windows 10 64bit

Alok-Baba, 2019-11-07:
1. I have downloaded and unzip mujoco200 at \userpath\.mojoco. 
2. I have  obtained license and mjkey.txt and copied it to \userpath\.mujoco  as well as 
     \userpath\.mujoco\mujoco200\bin\
3. I have added environment variable name PATH = C:\userpath\.mujoco\mujoco200\bin;%PATH%
4. Alongside this, I have tried
 >set MUJOCO_PY_MJKEY_PATH=C:\path\to\.mujoco\mjkey.txt
 >set MUJOCO_PY_MUJOCO_PATH=C:\path\to\.mujoco\mujoco200\bin

5. And the rest of procedure of cloning this git, and installing openai

Yet I only able to run classic gym environments, whenever I try to run any robotics environment it shows " add mujoco to your library error".

[image]

The error shown is

[image]

 I don't get it even I have added the PATH as new variable and also in command promt, why it is still showing this error.

Any help is highly appreciated.

dorox1, 2019-11-12:
I don't have an answer, but I have the same issue. I installed MuJoCo, have the license key and everything, but get that exact error when I try to run a MuJoCo program:

> Exception: Please add mujoco library to your PATH:
> set PATH=C:\Users\USERNAME\.mujoco\mujoco200\bin;%PATH%

I have tried several different batch configurations, and have put the MuJoCo mjkey.txt file in several different directories. My user environment variables are as follows:

**MUJOCO_PY_MJKEY_PATH** = C:\path\to\.mujoco\mujoco200\bin\mjkey.txt
**MUJOCO_PY_MUJOCO_PATH** = C:\path\to\.mujoco\mujoco200\bin
**Path** = C:\Users\USERNAME\.mujoco\mujoco200\bin;%PATH%

I have also tried:

**MUJOCO_PY_MJKEY_PATH** = C:\path\to\.mujoco\mjkey.txt
**MUJOCO_PY_MUJOCO_PATH** = C:\path\to\.mujoco\mujoco200\bin
**Path** = C:\path\to\.mujoco\mujoco200\bin;%PATH% - (or the previous USERNAME one)

Just adding this to the list of things that have been tried.

thavlik, 2019-12-26:
The python wheel currently expects you to have mjpro150 and not mujuco200. Your paths are otherwise set correctly. Check the download page at https://www.roboti.us/index.html

chuoru, 2020-03-25:
I have successfully install mujoco-py on Windows 10. In my opinion, your problem might be:

1) Window-supported mujoco-py version stopped at 1.50.1.68 as it was written in README. So we should use that version. And if using this version, we need to download mjpro150 from https://www.roboti.us/index.html. 

2) About the SET PATH errors, I just copied and pasted the recommended from the error which was
set PATH=C:\Users\tranthanh\.mujoco\mjpro150\bin;%PATH% , into the terminal.

KomanechiL, 2020-04-26:
Hello, I followed serveral tutorials to install Mujoco on WIndows 10:

- https://medium.com/@sayanmndl21/install-openai-gym-with-box2d-and-mujoco-in-windows-10-e25ee9b5c1d5
- https://github.com/openai/mujoco-py/issues/253

But none of these worked for me.
 
Here are my steps that i made

1. I downloaded Microsoft Visual C++ build tools 2019

2. I downloaded `mjpro150 win64` from https://www.roboti.us/index.html
 

3. created `%userprofile%\.mujoco\` and put mjpro150 into the folder.
  So my folder structure looks as following:
- .mujoco
    - mjkey.txt
     - mjpro150
         - bin
             - mjkey.txt
             - ...
         - doc
         - include
         - model
         - sample
 
4. I created my virtual enviroment in my projectfolder.
 `> virtualenv env`
 

5. I cloned the repository and 
 ```
$git clone https://github.com/openai/mujoco-py.git
$git checkout tag/1.50.1.0
 ```
 

6. I activated my virtual enviroment and installed gym
 ```
> & env/Scripts/activate
> pip install gym
> SET PATH=%userprofile%\.mujoco\mjpro150\bin;$Env:PATH;
```
 `> echo $Env:PATH ` contained  the path `%userprofile%\.mujoco\mjpro150\bin`
 
 
7. Preparations to install mujoco
```
cd mujoco-py
python -m pip install --upgrade setuptools #upgrade setuptools
pip install -r requirements.txt #install requirements
pip install -r requirements.dev.txt #install dev requirements
```

8. I replaced all the instances of
`isinstance(addr, (int, np.int32, np.int64))`
with
`hasattr(addr, '__int__')`
in these files
 ```
\path\to\mujoco-py\scripts\gen_wrappers.py
and
\path\to\mujoco-py\mujoco_py\generated\wrappers.pxi
```
 
9. Now I tried to recompile it
`> python -c "import mujoco_py" #force compile mujoco_py`
 
**And then this Error occured :  
`ImportError: DLL load failed while importing cymj:`**
_Note the error occured even if I did not made the changes in Step 8._
 

10. I ignored this error and continued with
`>python setup.py install`
and it displayed no Error
 

11. However I executed the following code 
 ```
import gym # openAi gym
from gym import envs
 
env = gym.make('FetchReach-v1')
env.reset()
for _ in range(1000):
     env.render()
     env.step(env.action_space.sample()) # take a random action
env.close()
 ```
 

and got:
**`CompileError: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.25.28610\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2`**

Can someone tell which steps I have done wrong?

paulaxa1, 2020-11-21:
After you set the path, if you still have the problem, restart your computer. That fixed it for me. Restart it, don't shut down and open again.

AOD555, 2021-02-10:
> I have successfully install mujoco-py on Windows 10. In my opinion, your problem might be:
>
> 1. Window-supported mujoco-py version stopped at 1.50.1.68 as it was written in README. So we should use [...]

Literally open up the terminal and paste the set path, and then it should work

xuxiaoxia8385, 2021-03-22:
> After you set the path, if you still have the problem, restart your computer. That fixed it for me. Restart it, don't shut down and open again.

This also works for me! Thanks very much!

KindleKin, 2021-07-06:
> Hello, I followed serveral tutorials to install Mujoco on WIndows 10:
> 
> * https://medium.com/@sayanmndl21/install-openai-gym-with-box2d-and-mujoco-in-windows-10-e25ee9b5c1d5
> * [What is the step by step process to install mujoco-py in Windows 10 without installing Anaconda? #253](https://github.com/openai/mujoco-py/issues/253)
> 
> But none of these worked for me.
> 
> Here are my steps that i made
> 
> 1. I downloaded Microsoft Visual C++ build tools 2019
> 2. I downloaded `mjpro150 win64` from https://www.roboti.us/index.html
> 3. created `%userprofile%\.mujoco\` and put mjpro150 into the folder.
>    So my folder structure looks as following:
> 
> * .mujoco
>   
>   * mjkey.txt
>   * mjpro150
>     
>     * bin
>       
>       * mjkey.txt
>       * ...
>     * doc
>     * include
>     * model
>     * sample
> 
> 1. I created my virtual enviroment in my projectfolder.
>    `> virtualenv env`
> 2. I cloned the repository and
> 
> ```
> $git clone https://github.com/openai/mujoco-py.git
> $git checkout tag/1.50.1.0
> ```
> 
> 1. I activated my virtual enviroment and installed gym
> 
> ```
> > & env/Scripts/activate
> > pip install gym
> > SET PATH=%userprofile%\.mujoco\mjpro150\bin;$Env:PATH;
> ```
> 
> `> echo $Env:PATH ` contained the path `%userprofile%\.mujoco\mjpro150\bin`
> 
> 1. Preparations to install mujoco
> 
> ```
> cd mujoco-py
> python -m pip install --upgrade setuptools #upgrade setuptools
> pip install -r requirements.txt #install requirements
> pip install -r requirements.dev.txt #install dev requirements
> ```
> 
> 1. I replaced all the instances of
>    `isinstance(addr, (int, np.int32, np.int64))`
>    with
>    `hasattr(addr, '__int__')`
>    in these files
> 
> ```
> \path\to\mujoco-py\scripts\gen_wrappers.py
> and
> \path\to\mujoco-py\mujoco_py\generated\wrappers.pxi
> ```
> 
> 1. Now I tried to recompile it
>    `> python -c "import mujoco_py" #force compile mujoco_py`
> 
> **And then this Error occured : `ImportError: DLL load failed while importing cymj:`**
> _Note the error occured even if I did not made the changes in Step 8._
> 
> 1. I ignored this error and continued with
>    `>python setup.py install`
>    and it displayed no Error
> 2. However I executed the following code
> 
> ```
> import gym # openAi gym
> from gym import envs
> 
> env = gym.make('FetchReach-v1')
> env.reset()
> for _ in range(1000):
>     env.render()
>     env.step(env.action_space.sample()) # take a random action
> env.close()
> ```
> 
> and got:
> **`CompileError: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.25.28610\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2`**
> 
> Can someone tell which steps I have done wrong?

Hi bro, have you sloved this problem?

I got same error as you...

KomanechiL, 2021-07-06:
> Hi bro, have you sloved this problem?
>
> I got same error as you...

> Hi bro, have you sloved this problem?
>
> I got same error as you...

unfortunately not. I moved to pybullet for my project

grumm9690, 2021-10-17:
> unfortunately not. I moved to pybullet for my project

Have anyone solved this final error? I need mujoco for my openai-gym project

jannish-git, 2021-11-15:
> Hello, I followed serveral tutorials to install Mujoco on WIndows 10:
>
>     * https://medium.com/@sayanmndl21/install-openai-gym-with-box2d-and-mujoco-in-windows-10-e25ee9b5c1d5
>
>     * [What is the [...]

I have found a solution for the problem after step 9, the importerror: load dll.
According to [this post](https://stackoverflow.com/questions/41365446/how-to-resolve-importerror-dll-load-failed-on-python) Python loads only DLLs from trusted folders since version 3.8. Therefore, one has to add the folder where mujoco is installed to the dll_list:
`os.add_dll_directory("....... //.mujoco//mjpro150//bin")`

jannish-git, 2021-11-15:
I could fix the rebuild-error
`Import error. Trying to rebuild mujoco_py. running build_ext building 'mujoco_py.cymj' extension`
by simple adding the dll BEFORE importing gym