openai/mujoco-py #477: Installing gym[mujoco] requires mujoco150 (not mujoco200)

TomasMerva, 2019-11-14:
Hello, I am trying to install mujoco_py and gym[mujoco, robotics] at new pc, although it cannot be installed because I have always been getting error about `missing Mujoco`. I installed mujoco200 and the gym[mujoco, robotics] is missing mujoco150. Is there any way how to solve this? At others pc I have not had this problem.
 
```
You appear to be missing MuJoCo.  We expected to find the file here: /home/tomas/.mujoco /mjpro150
This package only provides python bindings, the library must be installed separately.    
Please follow the instructions on the README to install MuJoCo
```

Python: 3.6.8
Mujoco: 200 
mujoco_py: downloaded from github (using git clone)

Thank you in advance.

buptwcy, 2019-11-15:
The latest version of gym Removed mujoco >= 2.0 support due to lack of tests.

2019-11-08 (v0.15.4)

        Added multiple env wrappers (thanks @zuoxingdong and @hartikainen!)

        Removed mujoco >= 2.0 support due to lack of tests

TomasMerva, 2019-11-15:
Thank you, 0.15.3 works :)

Urheen, 2019-11-22:
Hi, how do you install mujoco-py? I follow the instruction in the README but see errors. Could you provide your instructions?

TomasMerva, 2019-11-26:
I am sorry that it took me so long to answer you. So there is the procedure for successfully installation (tested today):
1. sudo apt install libosmesa6-dev libgl1-mesa-glx libglfw3
2. sudo apt install libglew-dev libglfw3-dev
3. sudo apt install patchelf
4. git clone https://github.com/openai/mujoco-py.git
5. cd mujoco-py
6. pip3 install --no-cache-dir e .
7. export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so
8. sudo apt install gym[all]==0.15.3

Before installing gym[mujoco, robotics] I was getting an error. However, installing gym[mujoco, robotics] have fixed this error.
Hope it will help

JayanthRR, 2021-07-28:
Is there a reason to prefer one version of mujoco over the other?

DanielTakeshi, 2021-08-03:
If this version does not support MuJoCo 2.0, then it seems like this README is confusing:

[image]