openai/mujoco-py #481: Missing path to your environment variable.

RyanRizzo96, 2019-11-22:
**Describe the bug**
mujoco-py installation not recognising that there is in fact the correct path in my environment variable.

When I run echo `$LD_LIBRARY_PATH` from inside the virtual env
`
(tensorflow_p36) ubuntu@ip-172-XX-XX-XXX:~$ echo $LD_LIBRARY_PATH`

I get `/home/ubuntu/.mujoco/mujoco200/bin`

**Expected behavior**
Installation as expected.

**Error Messages**
```
  Building wheel for mujoco-py (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-einoco3j/mujoco-py/setup.py'"'"'; __file__='"'"'/tmp/pip-install-einoco3j/mujoco-py/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-wn8gxwtt --python-tag cp36
       cwd: /tmp/pip-install-einoco3j/mujoco-py/
  Complete output (35 lines):
  running bdist_wheel
  running build
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-einoco3j/mujoco-py/setup.py", line 45, in <module>
      tests_require=read_requirements_file('requirements.dev.txt'),
    File "/usr/local/lib/python3.6/dist-packages/setuptools/__init__.py", line 143, in setup
      return distutils.core.setup(**attrs)
    File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
      self.run_command(cmd)
    File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/usr/local/lib/python3.6/dist-packages/wheel/bdist_wheel.py", line 192, in run
      self.run_command('build')
    File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/tmp/pip-install-einoco3j/mujoco-py/setup.py", line 29, in run
      import mujoco_py  # noqa: force build
    File "/tmp/pip-install-einoco3j/mujoco-py/mujoco_py/__init__.py", line 3, in <module>
      from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
    File "/tmp/pip-install-einoco3j/mujoco-py/mujoco_py/builder.py", line 503, in <module>
      cymj = load_cython_ext(mujoco_path)
    File "/tmp/pip-install-einoco3j/mujoco-py/mujoco_py/builder.py", line 69, in load_cython_ext
      _ensure_set_env_var("LD_LIBRARY_PATH", lib_path)
    File "/tmp/pip-install-einoco3j/mujoco-py/mujoco_py/builder.py", line 118, in _ensure_set_env_var
      var_name, var_name, lib_path))
  Exception:
  Missing path to your environment variable.
  Current values LD_LIBRARY_PATH=
  Please add following line to .bashrc:
  export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/ubuntu/.mujoco/mujoco200/bin
  ----------------------------------------
  ERROR: Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
Failed to build mujoco-py
Installing collected packages: mujoco-py
    Running setup.py install for mujoco-py ... error
```

**Desktop (please complete the following information):**
 - OS: [Ubuntu 18.04]
 - Python Version [e.g. 3.6.5]
 - Mujoco Version [e.g. 2.00]
 - mujoco-py version [e.g. 2.0.2.0]

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`: /home/ubuntu/.mujoco/mujoco200/bin

closed 2019-11-23 by RyanRizzo96

yunzhe-tao, 2020-03-11:
Hi @RyanRizzo96, have you figured out the solution? I have exactly the same issue. Thanks

Marwa215, 2020-03-20:
Hi @RyanRizzo96, I have exactly the same problem, if you get the solution please let me know!!

piojanu, 2020-04-03:
Hi! The same problem here, I'm using virtualenv. I've tried to install it on MacOS and failed. Now I started from clean Ubuntu Bionic VM and get stuck here... This is really frustrating.

bainro, 2020-09-11:
I had the same problem. It went away when I ran in a normal shell, not a tmux one.