openai/mujoco-py #483: Successfully installed but ModuleNotFoundError ?!

RyanRizzo96, 2019-11-24:
**Describe the bug**

Installs successfully but cannot find the module. Why?

```
(test_env) ubuntu@ip-172-31-33-90:~$ pip3 install -U 'mujoco-py<2.1,>=2.0'
Collecting mujoco-py<2.1,>=2.0
  Using cached https://files.pythonhosted.org/packages/44/f8/1c2840d5ed717ac612aba2879dc1ce548451be6e4d32a7fc7c549c127119/mujoco_py-2.0.2.8-py3-none-any.whl
Requirement already satisfied, skipping upgrade: fasteners~=0.15 in ./anaconda3/lib/python3.6/site-packages (from mujoco-py<2.1,>=2.0) (0.15)
Requirement already satisfied, skipping upgrade: numpy>=1.11 in ./anaconda3/lib/python3.6/site-packages (from mujoco-py<2.1,>=2.0) (1.16.4)
Requirement already satisfied, skipping upgrade: imageio>=2.1.2 in ./anaconda3/lib/python3.6/site-packages (from mujoco-py<2.1,>=2.0) (2.3.0)
Requirement already satisfied, skipping upgrade: glfw>=1.4.0 in ./anaconda3/lib/python3.6/site-packages (from mujoco-py<2.1,>=2.0) (1.8.4)
Requirement already satisfied, skipping upgrade: cffi>=1.10 in ./anaconda3/lib/python3.6/site-packages (from mujoco-py<2.1,>=2.0) (1.12.3)
Requirement already satisfied, skipping upgrade: Cython>=0.27.2 in ./anaconda3/lib/python3.6/site-packages (from mujoco-py<2.1,>=2.0) (0.28.2)
Requirement already satisfied, skipping upgrade: six in ./anaconda3/lib/python3.6/site-packages (from fasteners~=0.15->mujoco-py<2.1,>=2.0) (1.12.0)
Requirement already satisfied, skipping upgrade: monotonic>=0.1 in ./anaconda3/lib/python3.6/site-packages (from fasteners~=0.15->mujoco-py<2.1,>=2.0) (1.5)
Requirement already satisfied, skipping upgrade: pycparser in ./anaconda3/lib/python3.6/site-packages (from cffi>=1.10->mujoco-py<2.1,>=2.0) (2.19)
Installing collected packages: mujoco-py
Successfully installed mujoco-py-2.0.2.8

```
**To Reproduce**
`pip3 install -U 'mujoco-py<2.1,>=2.0'`

**Expected behavior**
A clear and concise description of what you expected to happen.

**Error Messages**

> Python 3.6.9 |Anaconda, Inc.| (default, Jul 30 2019, 19:07:31) 
> [GCC 7.3.0] on linux
> Type "help", "copyright", "credits" or "license" for more information.
> >>> import mujoco_py
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
> ModuleNotFoundError: No module named 'mujoco_py'

**Desktop (please complete the following information):**
 - OS: [e.g. macOS 10.13.6]
 - Python Version [e.g. 3.6.9]
 - Mujoco Version [e.g. 2.00]
 - mujoco-py version [e.g. 2.0.2.8]

Marwa215, 2020-03-20:
Ihave the same problem, @RyanRizzo96 did you solve this problem?

RyanRizzo96, 2020-03-21:
This environment worked for me:

`absl-py==0.7.1
astor==0.8.0
atari-py==0.2.3
atomicwrites==1.3.0
attrs==19.1.0
cffi==1.12.3
Click==7.0
cloudpickle==1.2.1
colorama==0.4.3
coverage==4.5.4
cycler==0.10.0
Cython==0.29.12
dill==0.3.1.1
filelock==3.0.12
future==0.17.1
gast==0.2.2
glfw==1.8.2
glob2==0.7
google-pasta==0.1.7
grpcio==1.22.0
gym==0.15.3
h5py==2.9.0
hunter==3.1.3
imageio==2.5.0
importlib-metadata==0.18
joblib==0.13.2
Keras-Applications==1.0.8
Keras-Preprocessing==1.1.0
kiwisolver==1.1.0
lockfile==0.12.2
manhole==1.6.0
Markdown==3.1.1
matplotlib==3.1.1
mock==3.0.5
more-itertools==7.1.0
mpi4py==3.0.2
mujoco-py==2.0.2.2
numpy==1.16.4
opencv-python==4.1.0.25
opt-einsum==3.1.0
packaging==19.0
pandas==0.24.2
Pillow==6.1.0
pluggy==0.12.0
progressbar2==3.47.0
protobuf==3.8.0
py==1.8.0
pycparser==2.19
pygit2==0.28.2
pyglet==1.3.2
pyparsing==2.4.0
pytest==5.0.1
pytest-cov==2.8.1
python-dateutil==2.8.0
python-utils==2.3.0
pytz==2019.1
scipy==1.3.0
seaborn==0.10.0
six==1.12.0
tensorboard==1.14.0
tensorflow==1.14.0
tensorflow-estimator==1.14.0
termcolor==1.1.0
tqdm==4.32.2
vulture==1.1
wcwidth==0.1.7
Werkzeug==0.15.4
wrapt==1.11.2
zipp==0.5.2`