openai/mujoco-py #484: How to load my own robot(urdf) model to mujoco?

YanhuaZhang516, 2019-11-24:
**Describe the bug**
Hi, I am new to mujoco, and I have a robot model which is urdf file. 
At first, I add
 "<mujoco>
        <compiler meshdir="../meshes/"/>
</mujoco>"
in the beginning of my urdf file.
And then I use command "./compile model.urdf model.mjb", and also use "./compile model.urdf model.xml" to get mjb and xml file.
But the contents in those files are empty, Nothing is convert.

The contents in xml file are like this:
"
<mujoco model="MuJoCo Model">
    <compiler angle="radian" meshdir="../meshes/" />
    <size njmax="500" nconmax="100" />
    <worldbody />
</mujoco>
"
That is all in my xml file.
Has anyone idea how to load urdf robot model to Mujoco? 
Thanks a lot!!!

gokhansolak, 2019-11-26:
Hi Yanhua,

I would suggest [verifying the URDF file](http://wiki.ros.org/urdf#Verification) first. 

I think this issue is more about Mujoco than Mujoco-py. You can also consider the Mujoco forum to seek help.

wangcongrobot, 2020-01-15:
Hi, there is an example of how to create a Mujoco model from urdf.
[https://github.com/wangcongrobot/dual_ur5_husky_mujoco](https://github.com/wangcongrobot/dual_ur5_husky_mujoco)

ashBabu, 2020-05-10:
@YanhuaZhang516 : I think that adding of the tag ```< mujoco> < /mujoco >``` is forcing the ```./compile``` to behave the way it's behaving. I have tried with another urdf without mujoco tag and had no problem. Obviously I had to add manually mujoco specific things like ``` <option > </option> ``` afterwards

closed 2020-08-09 by YanhuaZhang516