openai/mujoco-py #488: distutils.errors.LinkError: command 'gcc' failed with exit status 1 on Import

mehrotrasan16, 2019-12-01:
python3 gymtest.py # this imports mujoco and tries to make an environment
Choosing the latest nvidia driver: /usr/lib/nvidia-430, among ['/usr/lib/nvidia-375', '/usr/lib/nvidia-430']
Choosing the latest nvidia driver: /usr/lib/nvidia-430, among ['/usr/lib/nvidia-375', '/usr/lib/nvidia-430']
running build_ext
building 'mujoco_py.cymj' extension
gcc -pthread -B /home/sanket/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/home/sanket/.mujoco/mujoco-py/mujoco_py -I/home/sanket/.mujoco/mujoco200/include -I/home/sanket/anaconda3/lib/python3.7/site-packages/numpy/core/include -I/home/sanket/.mujoco/mujoco-py/mujoco_py/vendor/egl -I/home/sanket/anaconda3/include/python3.7m -c /home/sanket/.mujoco/mujoco-py/mujoco_py/cymj.c -o /home/sanket/.mujoco/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxgpuextensionbuilder/temp.linux-x86_64-3.7/home/sanket/.mujoco/mujoco-py/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -B /home/sanket/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/home/sanket/.mujoco/mujoco-py/mujoco_py -I/home/sanket/.mujoco/mujoco200/include -I/home/sanket/anaconda3/lib/python3.7/site-packages/numpy/core/include -I/home/sanket/.mujoco/mujoco-py/mujoco_py/vendor/egl -I/home/sanket/anaconda3/include/python3.7m -c /home/sanket/.mujoco/mujoco-py/mujoco_py/gl/eglshim.c -o /home/sanket/.mujoco/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxgpuextensionbuilder/temp.linux-x86_64-3.7/home/sanket/.mujoco/mujoco-py/mujoco_py/gl/eglshim.o -fopenmp -w
gcc -pthread -shared -B /home/sanket/anaconda3/compiler_compat -L/home/sanket/anaconda3/lib -Wl,-rpath=/home/sanket/anaconda3/lib -Wl,--no-as-needed -Wl,--sysroot=/ /home/sanket/.mujoco/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxgpuextensionbuilder/temp.linux-x86_64-3.7/home/sanket/.mujoco/mujoco-py/mujoco_py/cymj.o /home/sanket/.mujoco/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxgpuextensionbuilder/temp.linux-x86_64-3.7/home/sanket/.mujoco/mujoco-py/mujoco_py/gl/eglshim.o -L/home/sanket/.mujoco/mujoco200/bin -Wl,-R/home/sanket/.mujoco/mujoco200/bin -lmujoco200 -lglewegl -o /home/sanket/.mujoco/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.7_37_linuxgpuextensionbuilder/lib.linux-x86_64-3.7/mujoco_py/cymj.cpython-37m-x86_64-linux-gnu.so -fopenmp
/home/sanket/anaconda3/compiler_compat/ld: cannot find -lmujoco200
/home/sanket/anaconda3/compiler_compat/ld: cannot find -lglewegl
collect2: error: ld returned 1 exit status
Traceback (most recent call last):
File "/home/sanket/anaconda3/lib/python3.7/distutils/unixccompiler.py", line 205, in link
self.spawn(linker + ld_args)
File "/home/sanket/anaconda3/lib/python3.7/distutils/ccompiler.py", line 909, in spawn
spawn(cmd, dry_run=self.dry_run)
File "/home/sanket/anaconda3/lib/python3.7/distutils/spawn.py", line 36, in spawn
_spawn_posix(cmd, search_path, dry_run=dry_run)
File "/home/sanket/anaconda3/lib/python3.7/distutils/spawn.py", line 159, in _spawn_posix
% (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "gymtest.py", line 3, in
from gym.envs.mujoco import mujoco_env
File "/home/sanket/anaconda3/lib/python3.7/site-packages/gym/envs/mujoco/init.py", line 1, in
from gym.envs.mujoco.mujoco_env import MujocoEnv
File "/home/sanket/anaconda3/lib/python3.7/site-packages/gym/envs/mujoco/mujoco_env.py", line 12, in
import mujoco_py
File "/home/sanket/.mujoco/mujoco-py/mujoco_py/init.py", line 3, in
from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
File "/home/sanket/.mujoco/mujoco-py/mujoco_py/builder.py", line 506, in
cymj = load_cython_ext(mujoco_path)
File "/home/sanket/.mujoco/mujoco-py/mujoco_py/builder.py", line 105, in load_cython_ext
cext_so_path = builder.build()
File "/home/sanket/.mujoco/mujoco-py/mujoco_py/builder.py", line 221, in build
built_so_file_path = self._build_impl()
File "/home/sanket/.mujoco/mujoco-py/mujoco_py/builder.py", line 291, in _build_impl
so_file_path = super()._build_impl()
File "/home/sanket/.mujoco/mujoco-py/mujoco_py/builder.py", line 244, in _build_impl
dist.run_commands()
File "/home/sanket/anaconda3/lib/python3.7/distutils/dist.py", line 966, in run_commands
self.run_command(cmd)
File "/home/sanket/anaconda3/lib/python3.7/distutils/dist.py", line 985, in run_command
cmd_obj.run()
File "/home/sanket/anaconda3/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
_build_ext.build_ext.run(self)
File "/home/sanket/anaconda3/lib/python3.7/distutils/command/build_ext.py", line 340, in run
self.build_extensions()
File "/home/sanket/.mujoco/mujoco-py/mujoco_py/builder.py", line 144, in build_extensions
build_ext.build_extensions(self)
File "/home/sanket/anaconda3/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 194, in build_extensions
self.build_extension(ext)
File "/home/sanket/anaconda3/lib/python3.7/distutils/command/build_ext.py", line 559, in build_extension
target_lang=language)
File "/home/sanket/anaconda3/lib/python3.7/distutils/ccompiler.py", line 717, in link_shared_object
extra_preargs, extra_postargs, build_temp, target_lang)
File "/home/sanket/anaconda3/lib/python3.7/distutils/unixccompiler.py", line 207, in link
raise LinkError(msg)
distutils.errors.LinkError: command 'gcc' failed with exit status 1

I get that it is because
/home/sanket/anaconda3/compiler_compat/ld: cannot find -lmujoco200
/home/sanket/anaconda3/compiler_compat/ld: cannot find -lglewegl
are missing but i can't find too much about what they are and how to fix it.

xiaohythu, 2019-12-08:
I met the same problem with you, did you fix it?

mehrotrasan16, 2019-12-09:
Not yet, I know the error seems to be because of these missing libraries, but i'm not sure what to add or remove here.

/home/sanket/anaconda3/compiler_compat/ld: cannot find -lmujoco200
/home/sanket/anaconda3/compiler_compat/ld: cannot find -lglewegl

pengPeterpeng, 2020-09-26:
I face the same problem

namansaxena9, 2021-08-23:
Make sure .mujoco/mujoco200/bin folder should contain all the following files: ['basic', 'compile', 'derivative', 'libglew.so', 'libglewegl.so', 'libglewosmesa.so', 'libglfw.so.3', 'libglfw3.a', 'libmujoco200.so', 'libmujoco200nogl.so', 'mjkey.txt', 'record', 'simulate', 'testspeed', 'testxml']

If all the files are not present redownload the mujoco200 folder from "https://www.roboti.us/index.html" according to your operating system.

CounKer, 2022-01-01:
I face this problem on ubuntu too, and i find it is because the mujoco i download is for windows. I fixed this problem by download the right mujoco.