openai/mujoco-py #489: Successful installation but canot import mujoco_py on ubuntu 18

PhysicalElective2, 2019-12-07:
**Describe the bug**
I have add LD_LIBRARY_PATH ,but it always can't

>>> import mujoco_py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/shaoxi/.local/lib/python3.6/site-packages/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/shaoxi/.local/lib/python3.6/site-packages/mujoco_py/builder.py", line 510, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/shaoxi/.local/lib/python3.6/site-packages/mujoco_py/builder.py", line 69, in load_cython_ext
    _ensure_set_env_var("LD_LIBRARY_PATH", lib_path)
  File "/home/shaoxi/.local/lib/python3.6/site-packages/mujoco_py/builder.py", line 119, in _ensure_set_env_var
    var_name, var_name, lib_path))
Exception: 
Missing path to your environment variable. 
Current values LD_LIBRARY_PATH=
Please add following line to .bashrc:
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/shaoxi/.mujoco/mujoco200/bin

**To Reproduce**
Steps to reproduce the behavior.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Error Messages**
Including more/longer error messages gives us more information to work with.

**Desktop (please complete the following information):**
 - OS: ubuntu 18LTS
 - Python Version [e.g. 3.6.6]
 - Mujoco Version [e.g. 1.50]
 - mujoco-py version 2.00

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
:/home/shaoxi/.mujoco/mujoco200/bin
 - output of: `echo $HOME`
 - output of: `echo $USER`

**Additional context**
Add any other context about the problem here.

surssur, 2020-04-02:
did you solve this problem?
after add this line "export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/sur/.mujoco/mujoco200/bin
"
but still can't find this environment during run py_file, i don't kown why

sush1996, 2020-04-06:
I had a similar issue. I added LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/sur/.mujoco/mujoco200/bin in my etc/environment

littlebirdoflargehead, 2020-04-11:
> I had a similar issue. You can resolve it by adding LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/sur/.mujoco/mujoco200/bin in your /etc/environment

Do you mean that I should add the whole sentence "LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/sur/.mujoco/mujoco200/bin in your /etc/environment" in the file .bashrc ?

Or I need to add "LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/sur/.mujoco/mujoco200/bin" in some other place that you mentioned /etc/environment ?

sush1996, 2020-04-14:
The latter. "/etc/environment" is basically a text file and you usually add your environment variables in there.