openai/mujoco-py #492: ERROR: Could not build wheels for mujoco-py which use PEP 517 and cannot be installed directly

soerendip, 2019-12-11:
Trying to install with pip in a fresh conda env with installed gym.
```

channels:
  - defaults
dependencies:
  - _libgcc_mutex=0.1=main
  - ca-certificates=2019.11.27=0
  - certifi=2019.11.28=py38_0
  - libedit=3.1.20181209=hc058e9b_0
  - libffi=3.2.1=hd88cf55_4
  - libgcc-ng=9.1.0=hdf63c60_0
  - libstdcxx-ng=9.1.0=hdf63c60_0
  - ncurses=6.1=he6710b0_1
  - openssl=1.1.1d=h7b6447c_3
  - pip=19.3.1=py38_0
  - python=3.8.0=h0371630_2
  - readline=7.0=h7b6447c_5
  - setuptools=42.0.2=py38_0
  - sqlite=3.30.1=h7b6447c_0
  - tk=8.6.8=hbc83047_0
  - wheel=0.33.6=py38_0
  - xz=5.2.4=h14c3975_4
  - zlib=1.2.11=h7b6447c_3
  - pip:
    - cloudpickle==1.2.2
    - future==0.18.2
    - gym==0.15.4
    - numpy==1.17.4
    - opencv-python==4.1.2.30
    - pyglet==1.3.2
    - scipy==1.3.3
    - six==1.13.0
```

soerendip, 2019-12-11:
I tried:

    pip install mujoco_py

    pip install -U 'mujoco-py<2.1,>=2.0'

Also downgrading to python 3.6 did not help.

closed 2019-12-12 by soerendip

Sixzero, 2019-12-15:
Same probelm here on Mac.

rbadkund, 2019-12-17:
Same, on MacOS Catalina

rominamir, 2019-12-18:
same problem with windows 10

niccolomenicori, 2020-01-04:
Same problem in linux, trying to install pupil-detectors

jacekplocharczyk, 2020-01-10:
The same on Ubuntu 18.04.
Why it is closed?

iecaser, 2020-02-08:
Same problem, ArchLinux

prinsherbert, 2020-02-10:
Same on Google Colab.

Spec122, 2020-02-11:
Same on Windows 10

Midnight93, 2020-02-11:
Same on Colab

astooke, 2020-02-26:
From Ubuntu 18, installed fresh on a couple machines with varied experiences over a few weeks...one worked with `pip install mujoco-py=2.0.2.7`; another worked by cloning the repo and installing; but yet another had successful cloned-repo install but ran into the gcc error on importing mujoco_py (like in #501), which was finally resolved by `apt install gcc`, which put a new gcc-7 on the machine, then `pip install mujoco-py` gave hints through a series of missing libraries like patchelf (`apt-get install patchelf`) and -lGL (`apt-get install libglu1-mesa-dev mesa-common-dev`), and finally success.

Not sure that reproduces, but hope it helps!

Rm4n, 2020-03-13:
Updating the gcc worked for me; OS Catalina

elnazsn1988, 2020-03-15:
> From Ubuntu 18, installed fresh on a couple machines with varied experiences over a few weeks...one worked with `pip install mujoco-py=2.0.2.7`; another worked by cloning the repo and installing; [...]

does anyone know how to fix this on Google Cloud Platform? have exact same error.

LittleChild, 2020-04-02:
I don't konw this is right or not...but, it worked for me.
details see here, point 4, 5
[https://github.com/reinforcement-learning-kr/pg_travel/wiki/Installing-Mujoco-py-on-Linux](url)
first download MuJoCo-pro 150 version, see more on the website above.
and then Install mujoco_py
`$ git clone https://github.com/openai/mujoco-py.git`
`$ cd mujoco-py`
`$ python3 setup.py install`

janblumenkamp, 2020-04-04:
Same problem for me with the most recent release (mujoco 2.0.2.9). `pip install mujoco_py==2.0.2.8` worked.

emuemuJP, 2020-04-16:
> Same problem for me with the most recent release (mujoco 2.0.2.9). `pip install mujoco_py==2.0.2.8` worked.

It worked for me ! Thank you!
Note that i use Mac OS Mojave.

SungmanHong, 2020-05-11:
> Same problem for me with the most recent release (mujoco 2.0.2.9). `pip install mujoco_py==2.0.2.8` worked.

It worked for me ! Thank you!
Note that i use Win10

chelseas, 2020-07-21:
> I don't konw this is right or not...but, it worked for me.
> details see here, point 4, 5
> https://github.com/reinforcement-learning-kr/pg_travel/wiki/Installing-Mujoco-py-on-Linux
> first download MuJoCo-pro 150 version, see more on the website above.
> and then Install mujoco_py
> `$ git clone https://github.com/openai/mujoco-py.git`
> `$ cd mujoco-py`
> `$ python3 setup.py install`

Thanks! In conjunction with this:
https://github.com/ethz-asl/reinmav-gym/issues/35

Worked for me on ubuntu 16

quinnliu, 2021-01-10:
`pip install mujoco_py==2.0.2.8` also worked for me but in a `test.py` file I'm still not able to run `import mujoco_py` in my virtual environment where mujoco_py installed successfully. Any advice?

Error on `python3 test.py` 

```
  File "test.py", line 1, in <module>
    import mujoco_py
  File "/Users/qliu/Documents/agihome.com/robosuite_env/lib/python3.7/site-packages/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Users/qliu/Documents/agihome.com/robosuite_env/lib/python3.7/site-packages/mujoco_py/builder.py", line 506, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/Users/qliu/Documents/agihome.com/robosuite_env/lib/python3.7/site-packages/mujoco_py/builder.py", line 105, in load_cython_ext
    cext_so_path = builder.build()
  File "/Users/qliu/Documents/agihome.com/robosuite_env/lib/python3.7/site-packages/mujoco_py/builder.py", line 221, in build
    built_so_file_path = self._build_impl()
  File "/Users/qliu/Documents/agihome.com/robosuite_env/lib/python3.7/site-packages/mujoco_py/builder.py", line 334, in _build_impl
    so_file_path = super()._build_impl()
  File "/Users/qliu/Documents/agihome.com/robosuite_env/lib/python3.7/site-packages/mujoco_py/builder.py", line 244, in _build_impl
    dist.run_commands()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/Users/qliu/Documents/agihome.com/robosuite_env/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/command/build_ext.py", line 340, in run
    self.build_extensions()
  File "/Users/qliu/Documents/agihome.com/robosuite_env/lib/python3.7/site-packages/mujoco_py/builder.py", line 144, in build_extensions
    build_ext.build_extensions(self)
  File "/Users/qliu/Documents/agihome.com/robosuite_env/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/command/build_ext.py", line 449, in build_extensions
    self._build_extensions_serial()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/command/build_ext.py", line 474, in _build_extensions_serial
    self.build_extension(ext)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/command/build_ext.py", line 534, in build_extension
    depends=ext.depends)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command '/usr/local/opt/llvm/bin/clang' failed with exit status 1
```

hjw-1014, 2021-01-22:
> Same problem for me with the most recent release (mujoco 2.0.2.9). `pip install mujoco_py==2.0.2.8` worked.

work on MacOs, thank you!

carlee0, 2021-02-18:
> Same problem for me with the most recent release (mujoco 2.0.2.9). `pip install mujoco_py==2.0.2.8` worked.

Installation worked. But I got CompileError when trying to import the module. T.T

codenie, 2021-02-23:
> Same problem for me with the most recent release (mujoco 2.0.2.9). `pip install mujoco_py==2.0.2.8` worked.

Thank you! This version works.

janblumenkamp, 2021-02-23:
@soerendip This issue should really be re-opened. Why have you closed it in the first place? Not sure if the authors of the package are even aware of the problem since newer packages are released continuously (the most recent is 2.0.2.13). Just downgrading to a much older version is only a workaround and might cause problems in the future.

tao2020, 2021-03-24:
> @soerendip This issue should really be re-opened. Why have you closed it in the first place? Not sure if the authors of the package are even aware of the problem since newer packages are released [...]

Wish the authors of the package be aware of this issue. 
Installing the latest mujoco-py as of March 24, 2021 still has the same issue. Need to downgrade to 2.0.2.8 as a workaround.

VaiSuliafu, 2021-06-08:
Looks like as of 6/8/21, the default pip install does not work. Still need to downgrade to 2.0.2.8 as a workaround.

JayanthRR, 2021-07-07:
Downgrading to 2.0.2.8 allowed me to install, however I received an error when I try to import `mujoco_py`. The following worked finally.

`sudo ln -s /usr/lib/x86_64-linux-gnu/libGL.so.1 /usr/lib/x86_64-linux-gnu/libGL.so`

NishanthJKumar, 2021-08-15:
I struggled with this for a while because I needed `2.0.2.13` (I'm on Linux Ubuntu 20.04). Definitely start by following [the README's troubleshooting guide](https://github.com/openai/mujoco-py#ubuntu-installtion-troubleshooting). In my case, I was still getting this issue, and realized the root problem was that I didn't have the `dev` version of Python (Ubuntu 20.04 comes pre-installed with python 3.8, but I'd manually installed python 3.6 and created a virtualenv, where I was trying to install `mujoco-py` without python 3.6-dev). I was able to solve my issues by installing python 3.6-dev, but in general, make sure you have the dev version of whatever python you're using!

amirhossein-nhn, 2021-08-25:
I have been working on this issue for a while. I tried all of the solutions mentioned in this page. However, none of them worked!

danielstankw, 2021-10-06:
why is this closed? The installation of the mujoco-py is pain in the ass

AdaChambers, 2022-01-08:
For me, It couldn't find .mujoco/mujoco200_linux/bin/libmujoco200.so. But I checked it out and found that I already had a libmujoco210.so. So I simply used
  `ln -s libmujoco210.so libmujoco200.so` 
And it worked like a miracle on my machine with Ubuntu21.10.

SungmanHong, 2022-01-10:
pip install mujoco_py==2.0.2.8