openai/mujoco-py #493: MjSim rendering doesn't use GPU

prasoongoyal, 2019-12-16:
I'm running body_interaction.py example file on a remote server. Here's the code after making changes to run in headless mode:

```
#!/usr/bin/env python3
"""
Example of how bodies interact with each other. For a body to be able to
move it needs to have joints. In this example, the "robot" is a red ball
with X and Y slide joints (and a Z slide joint that isn't controlled).
On the floor, there's a cylinder with X and Y slide joints, so it can
be pushed around with the robot. There's also a box without joints. Since
the box doesn't have joints, it's fixed and can't be pushed around.
"""
from mujoco_py import load_model_from_xml, MjSim, MjViewer
import math
import os

MODEL_XML = """
<?xml version="1.0" ?>
<mujoco>
    <option timestep="0.005" />
    <worldbody>
        <body name="robot" pos="0 0 1.2">
            <joint axis="1 0 0" damping="0.1" name="slide0" pos="0 0 0" type="slide"/>
            <joint axis="0 1 0" damping="0.1" name="slide1" pos="0 0 0" type="slide"/>
            <joint axis="0 0 1" damping="1" name="slide2" pos="0 0 0" type="slide"/>
            <geom mass="1.0" pos="0 0 0" rgba="1 0 0 1" size="0.15" type="sphere"/>
			<camera euler="0 0 0" fovy="40" name="rgb" pos="0 0 2.5"></camera>
        </body>
        <body mocap="true" name="mocap" pos="0.5 0.5 0.5">
			<geom conaffinity="0" contype="0" pos="0 0 0" rgba="1.0 1.0 1.0 0.5" size="0.1 0.1 0.1" type="box"></geom>
			<geom conaffinity="0" contype="0" pos="0 0 0" rgba="1.0 1.0 1.0 0.5" size="0.2 0.2 0.05" type="box"></geom>
		</body>
        <body name="cylinder" pos="0.1 0.1 0.2">
            <geom mass="1" size="0.15 0.15" type="cylinder"/>
            <joint axis="1 0 0" name="cylinder:slidex" type="slide"/>
            <joint axis="0 1 0" name="cylinder:slidey" type="slide"/>
        </body>
        <body name="box" pos="-0.8 0 0.2">
            <geom mass="0.1" size="0.15 0.15 0.15" type="box"/>
        </body>
        <body name="floor" pos="0 0 0.025">
            <geom condim="3" size="1.0 1.0 0.02" rgba="0 1 0 1" type="box"/>
        </body>
    </worldbody>
    <actuator>
        <motor gear="2000.0" joint="slide0"/>
        <motor gear="2000.0" joint="slide1"/>
    </actuator>
</mujoco>
"""

model = load_model_from_xml(MODEL_XML)
sim = MjSim(model)
img = sim.render(600, 600, device_id=1)
t = 0
while True:
    sim.data.ctrl[0] = math.cos(t / 10.) * 0.01
    sim.data.ctrl[1] = math.sin(t / 10.) * 0.01
    t += 1
    sim.step()
    img = sim.render(600, 600, device_id=1)
    if t > 100 and os.getenv('TESTING') is not None:
        break

```

Even though I specify device_id=1 in sim.render(), the process doesn't use the GPU. Neither does it throw any errors; it just falls back to using the CPU for rendering.
Any suggestions?

Thank you!

SudeepDasari, 2020-02-29:
Maybe you're having the same issue I had? Check out #512

wookayin, 2020-07-16:
Make sure mujoco_py is using a shared module built with GPU:

```
>>> import mujoco_py
>>> mujoco_py.cymj
```

If it prints something like mujoco_py/generated/cymj_2.0.2.9_37_linux**gpu**extensionbuilder_37.so, you are good to go. Otherwise, you may need to rebuild it:

```
$ pip uninstall mujoco_py
$ pip install mujoco_py --no-binary mujoco_py    # force rebuilding shared modules
```

You need to have `libopengl-dev`, `libglew-dev`, etc. installed. Also, make sure that `/usr/lib/nvidia-000` directory exists (you may need to create on your own) -- this is a bug of openai mujoco_py. Refer to: https://github.com/openai/mujoco-py/issues/408#issuecomment-547026798

prasoongoyal, 2020-07-26:
Thanks for the suggestions! I'll try those out.
For now, I was able to get it work using a hack -- I installed dm_control and imported mujoco from dm_control:
`from dm_control import mujoco`
That fixed the issue.

closed 2020-07-26 by prasoongoyal

wangcongrobot, 2020-09-22:
> Make sure mujoco_py is using a shared module built with GPU:
> 
> ```
> >>> import mujoco_py
> >>> mujoco_py.cymj
> ```
> 
> If it prints something like mujoco_py/generated/cymj_2.0.2.9_37_linux**gpu**extensionbuilder_37.so, you are good to go. Otherwise, you may need to rebuild it:
> 
> ```
> $ pip uninstall mujoco_py
> $ pip install mujoco_py --no-binary mujoco_py    # force rebuilding shared modules
> ```
> 
> You need to have `libopengl-dev`, `libglew-dev`, etc. installed. Also, make sure that `/usr/lib/nvidia-000` directory exists (you may need to create on your own) -- this is a bug of openai mujoco_py. Refer to: [#408 (comment)](https://github.com/openai/mujoco-py/issues/408#issuecomment-547026798)
Hi @wookayin. I got the output like this: mujoco_py/generated/cymj_2.0.2.13_37_linux**cpu**extensionbuilder_37.so, and even I  rebuild it, I cannot get the **gpu** version. Do you know how to deal with this?

wangcongrobot, 2020-09-22:
> Make sure mujoco_py is using a shared module built with GPU:
>
> ```
> ```
>
> If it prints something like mujoco_py/generated/cymj_2.0.2.9_37_linux**gpu**extensionbuilder_37.so, you are good to go. [...]

Hi @wookayin. I got the output like this: mujoco_py/generated/cymj_2.0.2.13_37_linux**cpu**extensionbuilder_37.so, and even I rebuild it, I cannot get the gpu version. Do you know how to deal with this?

wenbin-hu, 2021-03-18:
> Hi @wookayin. I got the output like this: mujoco_py/generated/cymj_2.0.2.13_37_linux**cpu**extensionbuilder_37.so, and even I rebuild it, I cannot get the gpu version. Do you know how to deal with [...]

Hi @wangcongrobot , I had the same issue with you. Did you solve this problem?

Amir-Arsalan, 2021-03-31:
@prasoongoyal How do you check whether or not MuJoCo is running stuff on the GPU?

HareshKarnan, 2021-04-16:
^ you can check the output of nvidia-smi. It should show a "G" to the left of the python3.6 process.