openai/mujoco-py #494: fresh install not find gcc

emimarz, 2019-12-17:
**Describe the bug**
fresh install won't work

**To Reproduce**
$ pip3 install -U 'mujoco-py<2.1,>=2.0'
$ python3
import mujoco_py

**Expected behavior**
work

**Error Messages**
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/bla/ai/mujoco-py/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Users/bla/ai/mujoco-py/mujoco_py/builder.py", line 507, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/Users/bla/ai/mujoco-py/mujoco_py/builder.py", line 105, in load_cython_ext
    cext_so_path = builder.build()
  File "/Users/bla/ai/mujoco-py/mujoco_py/builder.py", line 220, in build
    built_so_file_path = self._build_impl()
  File "/Users/bla/ai/mujoco-py/mujoco_py/builder.py", line 324, in _build_impl
    'Could not find GCC executable.\n\n'
RuntimeError: Could not find GCC executable.

**Desktop (please complete the following information):**
 - OS: macOS 10.13.6
 - Python Version  3.7.4

**Additional context**
pip3 install -U 'mujoco-py<2.1,>=2.0'
Requirement already up-to-date: mujoco-py<2.1,>=2.0 in /Users/bla/ai/test1/lib/python3.7/site-packages (2.0.2.9)
Requirement already satisfied, skipping upgrade: glfw>=1.4.0 in /Users/bla/ai/test1/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (1.8.7)
Requirement already satisfied, skipping upgrade: Cython>=0.27.2 in /Users/bla/ai/test1/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (0.29.14)
Requirement already satisfied, skipping upgrade: imageio>=2.1.2 in /Users/bla/ai/test1/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (2.6.1)
Requirement already satisfied, skipping upgrade: numpy>=1.11 in /Users/bla/ai/test1/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (1.17.4)
Requirement already satisfied, skipping upgrade: cffi>=1.10 in /Users/bla/ai/test1/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (1.13.2)
Requirement already satisfied, skipping upgrade: fasteners~=0.15 in /Users/bla/ai/test1/lib/python3.7/site-packages (from mujoco-py<2.1,>=2.0) (0.15)
Requirement already satisfied, skipping upgrade: pillow in /Users/bla/ai/test1/lib/python3.7/site-packages (from imageio>=2.1.2->mujoco-py<2.1,>=2.0) (6.2.1)
Requirement already satisfied, skipping upgrade: pycparser in /Users/bla/ai/test1/lib/python3.7/site-packages (from cffi>=1.10->mujoco-py<2.1,>=2.0) (2.19)
Requirement already satisfied, skipping upgrade: monotonic>=0.1 in /Users/bla/ai/test1/lib/python3.7/site-packages (from fasteners~=0.15->mujoco-py<2.1,>=2.0) (1.5)
Requirement already satisfied, skipping upgrade: six in /Users/bla/ai/test1/lib/python3.7/site-packages (from fasteners~=0.15->mujoco-py<2.1,>=2.0) (1.13.0)
You are using pip version 19.0.3, however version 19.3.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

brew info gcc
gcc: stable 9.2.0 (bottled), HEAD
GNU compiler collection
https://gcc.gnu.org/
/usr/local/Cellar/gcc/9.2.0_2 (1,462 files, 289.8MB) *
  Poured from bottle on 2019-12-17 at 01:50:58
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/gcc.rb
==> Dependencies
Required: gmp ✔, isl ✔, libmpc ✔, mpfr ✔
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 116,918 (30 days), 325,989 (90 days), 1,051,333 (365 days)
install-on-request: 60,799 (30 days), 170,142 (90 days), 509,330 (365 days)
build-error: 0 (30 days)

I try clone the repository install from there and I have the same error

bibbygoodwin, 2020-09-16:
I have the same issue:
Mac OS 10.14.4
Python 3.7.6

brew info gcc
gcc: stable 10.2.0 (bottled), HEAD
GNU compiler collection
https://gcc.gnu.org/
/usr/local/Cellar/gcc/9.1.0 (1,542 files, 333.9MB) *
  Poured from bottle on 2019-05-08 at 21:00:16
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/gcc.rb
License: GPL-3.0
==> Dependencies
Required: gmp ✔, isl ✘, libmpc ✔, mpfr ✘

TMmichi, 2020-12-26:
I've been facing the same issue till I found the solution.
Building gcc version 8 from the source within brew solved this issue.

```brew install gcc@8```

will give you an error saying installing gcc8 via brew is only possible from the source, and provide a command that builds it with the source. Try that, it solved mine!