openai/mujoco-py #495: import mujoco throw me an error

emimarz, 2019-12-19:
**Describe the bug**
>>> import mujoco_py
give me an error

**To Reproduce**
$ pip3 install -U 'mujoco-py<2.1,>=2.0'
$ python3
import mujoco_py

**Expected behavior**
work

**Error Messages**
>>> import mujoco_py
objc[25979]: Class GLFWApplicationDelegate is implemented in both /Users/blablabla/.mujoco/mujoco200/bin/libglfw.3.dylib (0x111838778) and /usr/local/anaconda3/lib/python3.7/site-packages/glfw/libglfw.3.dylib (0x11d6586d0). One of the two will be used. Which one is undefined.
objc[25979]: Class GLFWWindowDelegate is implemented in both /Users/blablabla/.mujoco/mujoco200/bin/libglfw.3.dylib (0x111838700) and /usr/local/anaconda3/lib/python3.7/site-packages/glfw/libglfw.3.dylib (0x11d6586f8). One of the two will be used. Which one is undefined.
objc[25979]: Class GLFWContentView is implemented in both /Users/blablabla/.mujoco/mujoco200/bin/libglfw.3.dylib (0x1118387a0) and /usr/local/anaconda3/lib/python3.7/site-packages/glfw/libglfw.3.dylib (0x11d658748). One of the two will be used. Which one is undefined.
objc[25979]: Class GLFWWindow is implemented in both /Users/blablabla/.mujoco/mujoco200/bin/libglfw.3.dylib (0x111838818) and /usr/local/anaconda3/lib/python3.7/site-packages/glfw/libglfw.3.dylib (0x11d6587c0). One of the two will be used. Which one is undefined.

**Desktop (please complete the following information):**
 - OS: macOS 10.13.6
 - Python Version  3.7.4
 - mujoco-py version 2.0.2.9

**Additional context**
pip3 search GLFW
glfw-toolbox (1.0.1)  - GLFW toolbox
glfw-cffi (0.2.0)     - Foreign Function Interface wrapper for GLFW v3.x
glfw (1.8.7)          - A ctypes-based wrapper for GLFW3.
  INSTALLED: 1.8.7 (latest)
pyglfw (0.2.2)        - Python bindings for the GLFW library
cyglfw3 (3.1.0.2)     - Python bindings for GLFW 3+ using Cython
piety (0.1.0)         - A modern curses compatible terminal built using Vulcan and GLFW

shangwenyao, 2020-01-10:
I have the same issue too

cying9, 2020-01-13:
I have this issue as well

chisness, 2020-02-22:
I also have this issue, has anyone found a solution?

aktivhoon, 2020-04-02:
I think this isn't the best solution, but this worked for me:

`rm ~/.mujoco/mujoco200/bin/ligblfw*`

This will remove GLFW from mujoco directoy and thus make us to skip the error message.
However it doesn't seem to be the best solution since its quite risky to remove libraries from the original mujoco directory.

aktivhoon, 2020-04-02:
I finally noticed this is a unique problem in MacOS.
Just forget about my previous comment and **change your default python** to which you have installed **using homebrew**. This will solve the problem properly.

apandey99, 2020-05-25:
@aktivhoon  Could you provide more info on your solution? I am facing same the problem.

null-id, 2020-07-25:
What if I use Anaconda and don't want to change my default python to homebrew version? @aktivhoon

null-id, 2020-07-25:
@aktivhoon Besides, your ```rm``` solution is also not working.

null-id, 2020-07-25:
Anyway, this is just a warning. My code can run normally.

raul-openai, 2020-09-01:
Hi everyone.

Unfortunately I don't think there is an easy actual fix for this, since both mujoco and pip will end up with the library. However, **you should be able to change mujoco's folder's lib to point to the env you want to use**.

For example with a symlink to the env's:

1. Locate the python env's dylib:
```
(tests) ~/Projects/tests/python $ locate libglfw.3.dylib
/Users/username/.mujoco/mjpro150/bin/libglfw.3.dylib
/Users/username/.mujoco/mujoco200/bin/libglfw.3.dylib
/Users/username/.pyenv/versions/3.7.3/envs/tests/lib/python3.7/site-packages/glfw/libglfw.3.dylib
```

2. Go to mujoco's bin folder:
```
(tests) ~/Projects/tests/python $ cd ~/.mujoco/mujoco200/bin
~/.mujoco/mujoco200/bin $ ls
MUJOCO_LOG.TXT         compile                libglfw.3.dylib        libmujoco200.dylib     mjkey.txt              simulate               testxml
basic                  derivative             libglfw3.a             libmujoco200nogl.dylib record                 testspeed
```

3. Move the offending dylib to a backup file, and create a symlink to point to your env / python.
```
~/.mujoco/mujoco200/bin $ mv libglfw.3.dylib libglfw.3.dylib.unused
~/.mujoco/mujoco200/bin $ ln -s /Users/username/.pyenv/versions/3.7.3/envs/tests/lib/python3.7/site-packages/glfw/libglfw.3.dylib libglfw.3.dylib
```

Result:
```
~/.mujoco/mujoco200/bin $ ls -las libglfw.3*
  0 lrwxr-xr-x  1 username  staff      93 Sep  1 14:04 libglfw.3.dylib -> /Users/username/.pyenv/versions/3.7.3/envs/tests/lib/python3.7/site-packages/glfw/libglfw.3.dylib
280 -rwxr-xr-x  1 username  staff  140368 Apr 16 17:38 libglfw.3.dylib.unused
```
This should be a good workaround, since now both mujoco and python env use the same library.
Please, let me know if it doesn't work or any other issues arise.

Thank you.

closed 2020-09-10 by raul-openai

wanxinjin, 2022-03-27:
I am with macOS Monterey. I have also encountered the issue:
`~/.mujoco/mujoco210/bin/simulate ~/.mujoco/mujoco210/model/humanoid.xml
dyld[14477]: Library not loaded: libglfw.3.dylib`

I fixed this problem by running: brew install glfw

17780537256, 2022-07-15:
> 

Can you describe the operation process in detail?

> I finally noticed this is an unique problem in MacOS.
> Just forget about my previous comment and **change your default python** to which you have installed **using homebrew**. This will solve the problem properly.

Can you describe the operation process in detail?