openai/mujoco-py #500: [manjaro machine] command 'gcc' failed with exit status 1

cying9, 2020-01-01:
**Describe the bug**
Hi, I'm installing the mujoco_py with a Manjaro machine, and I meet a problem of command 'gcc' failed with exit status 1. Here is more details.
`
>>> import mujoco_py
running build_ext
building 'mujoco_py.cymj' extension
gcc -pthread -B /home/cying/anaconda3/envs/gymlab/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Ianaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py -I/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py -I/home/cying/.mujoco/mujoco200/include -I/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/numpy/core/include -I/home/cying/anaconda3/envs/gymlab/include/python3.7m -c /home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/cymj.c -o /home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/generated/_pyxbld_2.0.2.9_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -B /home/cying/anaconda3/envs/gymlab/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Ianaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py -I/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py -I/home/cying/.mujoco/mujoco200/include -I/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/numpy/core/include -I/home/cying/anaconda3/envs/gymlab/include/python3.7m -c /home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/gl/osmesashim.c -o /home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/generated/_pyxbld_2.0.2.9_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/gl/osmesashim.o -fopenmp -w
gcc -pthread -shared -B /home/cying/anaconda3/envs/gymlab/compiler_compat -L/home/cying/anaconda3/envs/gymlab/lib -Wl,-rpath=/home/cying/anaconda3/envs/gymlab/lib -Wl,--no-as-needed -Wl,--sysroot=/ /home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/generated/_pyxbld_2.0.2.9_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/cymj.o /home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/generated/_pyxbld_2.0.2.9_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/gl/osmesashim.o -L/home/cying/.mujoco/mujoco200/bin -Wl,-R/home/cying/.mujoco/mujoco200/bin -lmujoco200 -lglewosmesa -lOSMesa -lGL -o /home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/generated/_pyxbld_2.0.2.9_37_linuxcpuextensionbuilder/lib.linux-x86_64-3.7/mujoco_py/cymj.cpython-37m-x86_64-linux-gnu.so -fopenmp
/home/cying/anaconda3/envs/gymlab/compiler_compat/ld: /home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/generated/_pyxbld_2.0.2.9_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/cymj.o: unable to initialize decompress status for section .debug_info
/home/cying/anaconda3/envs/gymlab/compiler_compat/ld: /home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/generated/_pyxbld_2.0.2.9_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/cymj.o: unable to initialize decompress status for section .debug_info
/home/cying/anaconda3/envs/gymlab/compiler_compat/ld: /home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/generated/_pyxbld_2.0.2.9_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/cymj.o: unable to initialize decompress status for section .debug_info
/home/cying/anaconda3/envs/gymlab/compiler_compat/ld: /home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/generated/_pyxbld_2.0.2.9_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/cymj.o: unable to initialize decompress status for section .debug_info
/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/generated/_pyxbld_2.0.2.9_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/cymj.o: file not recognized: file format not recognized
collect2: error: ld returned 1 exit status
Traceback (most recent call last):
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/distutils/unixccompiler.py", line 205, in link
    self.spawn(linker + ld_args)
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/distutils/ccompiler.py", line 910, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/builder.py", line 510, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/builder.py", line 105, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/builder.py", line 221, in build
    built_so_file_path = self._build_impl()
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/builder.py", line 273, in _build_impl
    so_file_path = super()._build_impl()
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/builder.py", line 244, in _build_impl
    dist.run_commands()
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/distutils/command/build_ext.py", line 340, in run
    self.build_extensions()
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/mujoco_py-2.0.2.9-py3.7.egg/mujoco_py/builder.py", line 144, in build_extensions
    build_ext.build_extensions(self)
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/distutils/command/build_ext.py", line 449, in build_extensions
    self._build_extensions_serial()
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/distutils/command/build_ext.py", line 474, in _build_extensions_serial
    self.build_extension(ext)
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/distutils/command/build_ext.py", line 559, in build_extension
    target_lang=language)
[... cut ...]
  File "/home/cying/anaconda3/envs/gymlab/lib/python3.7/distutils/unixccompiler.py", line 207, in link
    raise LinkError(msg)
distutils.errors.LinkError: command 'gcc' failed with exit status 1'

I also read some solutions but can not solve my problem

**To Reproduce**
Install the mujoco_py from source code downloaded from github and run import mujoco_py in python cosole/

**Desktop (please complete the following information):**
 - OS: [Manjaro 18.1.5]
 - Python Version [3.7.5]
 - Mujoco Version [mujoco200]
 - mujoco-py version [2.0.2.9]

PurgingPanda, 2020-01-11:
``` pip install mujoco-py --no-use-pep517 --ignore-installed```
and then
```sudo ln -s /usr/bin/gcc /usr/local/bin/gcc-9```` seems to have fixed it on my manjaro instance.

PurgingPanda, 2020-01-11:
This solution worked for me: https://github.com/mrjbq7/ta-lib/issues/268

YangRui2015, 2020-01-21:
Same problem. I used Python 3.6.6, mujoco-py 2.0.2.9, mujoco200, gcc 8.2 and my MuJoCo-py was installed by source. When I imported mujoco_py, I get the same error. However, after I updated gcc to 9.2.0, it disappeared.

The following settings are ok for me.

- OS: [MacOS Catallina 10.15.2]
- Python Version [3.6.6]
- Mujoco Version [mujoco200]
- mujoco-py version [2.0.2.9]
- gcc version [9.2.0]

closed 2020-02-01 by cying9

Peter-zds, 2021-05-13:
I have tried `sudo ln -s /usr/lib/x86_64-linux-gnu/libGL.so.1 /usr/lib/x86_64-linux-gnu/libGL.so ` in the readme.md and it works