openai/mujoco-py #501: I think I have a quite complex question. I followed the guidance to install everything.Finally, I python setup.py install and it told me that everything is satisfied but when I input import mujoco_py, there was an error: distutils.error.LinkError:command:command 'gcc' failed with exit status 1. And the detail is that

suyunjiephoebe, 2020-01-13:
**Describe the bug**
I followed the guidance to install everything.Finally, I python setup.py install and it told me that everything is satisfied but when I input import mujoco_py, there was an error: distutils.error.LinkError:command:command 'gcc' failed with exit status 1. 

**To Reproduce**
The above step shows me that my gcc version is 7.2.0 
When I use gcc -v to check it, it was 5.4.0
I guessed it might be a problem but I am not sure
**Expected behavior**
Hope you can help me successfully install the mujoco-py and I don't might to use team viewer

**Error Messages**
./anaconda3/lib/python3.6/distutils/unixccompiler.py line199 in link raise LinkErorr(msg)
distutils.error.LinkError:command:command 'gcc' failed with exit status 1. 

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 16.04]
 - Python Version [e.g. 3.6.5]
 - Mujoco Version [e.g. 2.0]
 - mujoco-py version [e.g. 2.0.2.9]

Tarrandurham, 2020-01-13:
I get the same kind of thing on MacOs

suyunjiephoebe, 2020-01-13:
> I get the same kind of thing on MacOs
Oh, I sincerely hope that if you have any solution for this problem, you would tell me too! Thank you so much!!

gregbruss, 2020-03-04:
Same thing on most MacOS

wberdanier, 2020-03-16:
Same on MacOS

wberdanier, 2020-03-19:
Just wanted to update the thread: uninstalling all older GCC versions via homebrew and then running

`brew install gcc@9`,

which installs GCC 9.3, and then running

`pip install -U mujoco-py<2.1,>=2.0`

finally worked for me. No more compiler errors. Note that you seem to need both mujoco200 and mujocopro150 for openai/spinningup.

ww-Jingle, 2020-04-19:
Hi, did you solve this problem? I got the same problem in Linux.

suyunjiephoebe, 2020-04-24:
No I didn’t. I think my problem may be due to the virtual machine. I use virtual machine to operate Linux. And finally I choose box 2D to simulate 

| |
suyunjie0824
邮箱：suyunjie0824@163.com
|

Signature is customized by Netease Mail Master

On 04/20/2020 03:31, ww-Jingle wrote:

Hi, did you solve this problem? I got the same problem in Linux.