openai/mujoco-py #503: NameError: name '_raise_glfw_errors_as_exceptions' is not defined

DevMaelFranceschetti, 2020-01-20:
Hello,
after installing mujoco i encountered this error on spyder :

_`import gym
env = gym.make('Hopper-v2')
env.reset()
for _ in range(1000):
    env.render()
    env.step(env.action_space.sample()) # take a random action
env.close()`_

Output error :

>   File "/home/orion/.local/lib/python3.7/site-packages/glfw/__init__.py", line 840, in set_error_callback
>     if previous_callback is not None and previous_callback[0] != _raise_glfw_errors_as_exceptions:
> 
> NameError: name '_raise_glfw_errors_as_exceptions' is not defined

How could I solve it ?
Thanks ;)

DevMaelFranceschetti, 2020-01-20:
By adding "_raise_glfw_errors_as_exceptions = False" on the header of ~/.local/lib/python3.7/site-packages/glfw/__init__.py" it solved the problem and allowed me to run my python script on the Mujoco environment. But is it a good solution ? ...

clj003, 2020-01-21:
Also got this error yesterday. Not sure why it is happening. Following thread.

AllanSCosta, 2020-01-22:
Same issue here.