openai/mujoco-py #504: TypeError: 'numpy.intc' object is not iterable

pedagil, 2020-02-11:
**Describe the bug**
I'm able to use env that requires mujoco such as 'CartPole-v1', but can't render any of Fetch robotic env. This is likely something like [`TypeError:'numpy.int32' object is not iterable` for make('FetchReach-v1') #324](https://github.com/openai/mujoco-py/issues/324) . Fix [fix advanced mujoco envs for windows path separation #1220](https://github.com/openai/gym/pull/1220) Does not work for me

**To Reproduce**
import gym # openAi gym
from gym import envs

env = gym.make('FetchSlide-v1')
env.reset()
for _ in range(1000):
    env.render()
    env.step(env.action_space.sample()) # take a random action
env.close()

**Error Messages**
Traceback (most recent call last):
  File "C:/Users/petrosp/PycharmProjects/FetchReach/FetchReach.py", line 4, in <module>

    env = gym.make('FetchSlide-v1')

  File "C:\Users\petrosp\Anaconda3\envs\MainEnv\lib\site-packages\gym\envs\registration.py", line 156, in make
    return registry.make(id, **kwargs)

  File "C:\Users\petrosp\Anaconda3\envs\MainEnv\lib\site-packages\gym\envs\registration.py", line 101, in make
    env = spec.make(**kwargs)

  File "C:\Users\petrosp\Anaconda3\envs\MainEnv\lib\site-packages\gym\envs\registration.py", line 73, in make
    env = cls(**_kwargs)

  File "C:\Users\petrosp\Anaconda3\envs\MainEnv\lib\site-packages\gym\envs\robotics\fetch\slide.py", line 24, in __init__
    initial_qpos=initial_qpos, reward_type=reward_type)

  File "C:\Users\petrosp\Anaconda3\envs\MainEnv\lib\site-packages\gym\envs\robotics\fetch_env.py", line 48, in __init__
    initial_qpos=initial_qpos)

  File "C:\Users\petrosp\Anaconda3\envs\MainEnv\lib\site-packages\gym\envs\robotics\robot_env.py", line 36, in __init__
    self._env_setup(initial_qpos=initial_qpos)

  File "C:\Users\petrosp\Anaconda3\envs\MainEnv\lib\site-packages\gym\envs\robotics\fetch_env.py", line 172, in _env_setup
    self.sim.data.set_joint_qpos(name, value)

  File "mujoco_py\generated/wrappers.pxi", line 2547, in mujoco_py.cymj.PyMjData.set_joint_qpos
TypeError: 'numpy.intc' object is not iterable

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Python Version 3.7.5
 - Mujoco Version [e.g. 1.50]
 - mujoco-py version [e.g. 1.50.1.68]

ysd007, 2020-03-13:
I also have this problem.

TneitaP, 2020-04-29:
Same problem happened in:

- Windows 10
-Python Version 3.7.6
- Mujoco200
- mujoco-py  2.0.2.9

TneitaP, 2020-04-29:
@pedagil I cloned the current master of mujoco-py, changed all `isinstance(addr, (int, np.int32, np.int64))  `|->   `hasattr(addr, '__int__')` in `mujoco-py\mujoco_py\generated\wrappers.pxi` as discussed at  [TypeError:'numpy.int32' object is not iterable](https://github.com/openai/mujoco-py/issues/324). 
Finally, I used `pip install -e .` in the cloned mujoco-py mater repo, and successfully created the hand related repo, e.g.`env = gym.make('HandManipulatePenTouchSensorsDense-v0')` 
[image]

leey127, 2020-04-29:
I have the same problem, on Windows 10. Seems that the solutions for [TypeError:'numpy.int32' object is not iterable.](https://github.com/openai/mujoco-py/issues/324) can not work for this issue.

Iskandor, 2020-05-16:
> @pedagil I cloned the current master of mujoco-py, changed all `isinstance(addr, (int, np.int32, np.int64)) `|-> `hasattr(addr, '__int__')` in `mujoco-py\mujoco_py\generated\wrappers.pxi` as [...]

Worked for me for environment FetchReach-v1. Thanks a lot.

KindleKin, 2020-07-18:
> @pedagil I cloned the current master of mujoco-py, changed all `isinstance(addr, (int, np.int32, np.int64)) `|-> `hasattr(addr, '__int__')` in `mujoco-py\mujoco_py\generated\wrappers.pxi` as [...]

Where is your setup.py of mujoco-py ? How to pip install -e .

KindleKin, 2020-07-19:
> Where is your setup.py of mujoco-py ? How to pip install -e .

Oh，I've got it！

paulaxa1, 2020-11-22:
> @pedagil I cloned the current master of mujoco-py, changed all `isinstance(addr, (int, np.int32, np.int64)) `|-> `hasattr(addr, '__int__')` in `mujoco-py\mujoco_py\generated\wrappers.pxi` as [...]

I have no words. Not all heroes wear capes. Thank you.

2000222, 2022-02-13:
> **Describe the bug** I'm able to use env that requires mujoco such as 'CartPole-v1', but can't render any of Fetch robotic env. This is likely something like [`TypeError:'numpy.int32' object is not iterable` for make('FetchReach-v1') #324](https://github.com/openai/mujoco-py/issues/324) . Fix [fix advanced mujoco envs for windows path separation #1220](https://github.com/openai/gym/pull/1220) Does not work for me
> 
> **To Reproduce** import gym # openAi gym from gym import envs
> 
> env = gym.make('FetchSlide-v1') env.reset() for _ in range(1000): env.render() env.step(env.action_space.sample()) # take a random action env.close()
> 
> **Error Messages** Traceback (most recent call last): File "C:/Users/petrosp/PycharmProjects/FetchReach/FetchReach.py", line 4, in
> 
> ```
> env = gym.make('FetchSlide-v1')
> ```
> 
> File "C:\Users\petrosp\Anaconda3\envs\MainEnv\lib\site-packages\gym\envs\registration.py", line 156, in make return registry.make(id, **kwargs)
> 
> File "C:\Users\petrosp\Anaconda3\envs\MainEnv\lib\site-packages\gym\envs\registration.py", line 101, in make env = spec.make(**kwargs)
> 
> File "C:\Users\petrosp\Anaconda3\envs\MainEnv\lib\site-packages\gym\envs\registration.py", line 73, in make env = cls(**_kwargs)
> 
> File "C:\Users\petrosp\Anaconda3\envs\MainEnv\lib\site-packages\gym\envs\robotics\fetch\slide.py", line 24, in **init** initial_qpos=initial_qpos, reward_type=reward_type)
> 
> File "C:\Users\petrosp\Anaconda3\envs\MainEnv\lib\site-packages\gym\envs\robotics\fetch_env.py", line 48, in **init** initial_qpos=initial_qpos)
> 
> File "C:\Users\petrosp\Anaconda3\envs\MainEnv\lib\site-packages\gym\envs\robotics\robot_env.py", line 36, in **init** self._env_setup(initial_qpos=initial_qpos)
> 
> File "C:\Users\petrosp\Anaconda3\envs\MainEnv\lib\site-packages\gym\envs\robotics\fetch_env.py", line 172, in _env_setup self.sim.data.set_joint_qpos(name, value)
> 
> File "mujoco_py\generated/wrappers.pxi", line 2547, in mujoco_py.cymj.PyMjData.set_joint_qpos TypeError: 'numpy.intc' object is not iterable
> 
> **Desktop (please complete the following information):**
> 
> * OS: Windows 10
> * Python Version 3.7.5
> * Mujoco Version [e.g. 1.50]
> * mujoco-py version [e.g. 1.50.1.68]

Can not solve the problem even with mujoco 150, 200 or 210. If there exist any new solution ......