openai/mujoco-py #507: gym+mujoco finish execute raise OpenGL.error.NullFunctionError

yaoxunji, 2020-02-15:
I try to test gym+mujoco using this code
```
import gym
env = gym.make('Walker2d-v1')
for i_episode in range(20):
    observation = env.reset()
    for t in range(100):
        env.render()
        action = env.action_space.sample()
        observation, reward, done, info = env.step(action)
        if done:
            print("Episode {} finished after {} timesteps".format(i_episode,t+1))
            break
```
when finish execute it raise 

> OpenGL.error.NullFunctionError: Attempt to call an undefined function glIsFramebuffer, check for bool(glIsFramebuffer) before calling

all error:
```
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "D:\Anaconda\envs\mujoco\lib\site-packages\gym\utils\closer.py", line 67, in close
    closeable.close()
  File "D:\Anaconda\envs\mujoco\lib\site-packages\gym\core.py", line 167, in close
    self.render(close=True)
  File "D:\Anaconda\envs\mujoco\lib\site-packages\gym\core.py", line 153, in render
    return self._render(mode=mode, close=close)
  File "D:\Anaconda\envs\mujoco\lib\site-packages\gym\envs\mujoco\mujoco_env.py", line 105, in _render
    self._get_viewer().finish()
  File "D:\Anaconda\envs\mujoco\lib\site-packages\mujoco_py\mjviewer.py", line 327, in finish
    if gl.glIsFramebuffer(self._fbo):
  File "D:\Anaconda\envs\mujoco\lib\site-packages\OpenGL\platform\baseplatform.py", line 425, in __call__
    self.__name__, self.__name__,
OpenGL.error.NullFunctionError: Attempt to call an undefined function glIsFramebuffer, check for bool(glIsFramebuffer) before calling
Exception ignored in: <bound method Env.__del__ of <TimeLimit<Walker2dEnv<Walker2d-v1>>>>
Traceback (most recent call last):
  File "D:\Anaconda\envs\mujoco\lib\site-packages\gym\core.py", line 202, in __del__
  File "D:\Anaconda\envs\mujoco\lib\site-packages\gym\core.py", line 169, in close
  File "D:\Anaconda\envs\mujoco\lib\site-packages\gym\core.py", line 288, in _close
  File "D:\Anaconda\envs\mujoco\lib\site-packages\gym\core.py", line 167, in close
  File "D:\Anaconda\envs\mujoco\lib\site-packages\gym\core.py", line 153, in render
  File "D:\Anaconda\envs\mujoco\lib\site-packages\gym\envs\mujoco\mujoco_env.py", line 105, in _render
  File "D:\Anaconda\envs\mujoco\lib\site-packages\mujoco_py\mjviewer.py", line 325, in finish
  File "D:\Anaconda\envs\mujoco\lib\site-packages\mujoco_py\glfw.py", line 808, in destroy_window
OSError: exception: access violation reading 0x0000000000000000
Exception ignored in: <bound method Env.__del__ of <gym.envs.mujoco.walker2d.Walker2dEnv object at 0x000001FE7D20A7B8>>
Traceback (most recent call last):
  File "D:\Anaconda\envs\mujoco\lib\site-packages\gym\core.py", line 202, in __del__
  File "D:\Anaconda\envs\mujoco\lib\site-packages\gym\core.py", line 167, in close
  File "D:\Anaconda\envs\mujoco\lib\site-packages\gym\core.py", line 153, in render
  File "D:\Anaconda\envs\mujoco\lib\site-packages\gym\envs\mujoco\mujoco_env.py", line 105, in _render
  File "D:\Anaconda\envs\mujoco\lib\site-packages\mujoco_py\mjviewer.py", line 325, in finish
  File "D:\Anaconda\envs\mujoco\lib\site-packages\mujoco_py\glfw.py", line 808, in destroy_window
OSError: exception: access violation reading 0x0000000000000000
```
my configuration as follows and in win10 and python 3.6
```
mujoco-py            0.5.7
gym         0.9.1
PyOpenGL            3.1.5
```

leiyue812, 2021-05-12:
I have the same issue. Did you find out the solutions? Thanks.

csimo005, 2023-03-26:
Searching a similar error brought me here, so in case anyone lands here. I fixed my error by adding env.close() and the end of my training. I noticed that @yaoxunji did not include it at the end of their training. Hope this helps.