openai/mujoco-py #510: Spawn the window of the offscreen context out of focus

iSaran, 2020-02-20:
I am creating an offscreen context using the following:

```python        
offscreen = MjRenderContextOffscreen(sim, 0)
```

This creates a window which spawns in front of the other windows (in focus) that I have open. Is this the intended behavior? 

I have a setup in which I create multiple mujoco sessions overtime and I run them in the same pc, which I am working on. This means that the spawning windows interrupt my workflow. One fix is to minimize the window right after the context construction with:

```python
glfw.iconify_window(offscreen.opengl_context.window)
```
This is not a correct solution though because even momentarily the focus goes to the Mujoco window.

Can I create this context with its window spawning out of focus from the very start?

My mujoco-py version is 1.50.1.68.

iSaran, 2020-03-05:
It seems that `MjRenderContextOffscreen` does not really create an offscreen context and this is why it creates these annoying windows. If I change [the constructor of MjRenderContextOffscreen](https://github.com/openai/mujoco-py/blob/d73ce6e91d096b74da2a2fcb0a4164e10db5f641/mujoco_py/mjrendercontext.pyx#L279) from `cinit` to `init` then it properly creates an offscreen context.

For now I directly create an instance of the base class, in order to create an offscreen context properly:

```python
offscreen = MjRenderContext(sim=self.sim, device_id=0, offscreen=True, opengl_backend='glfw')

gonultasbu, 2020-05-27:
I also haven't been able to solve this problem. Super annoying.

kracon7, 2020-05-27:
@iSaran 
Your approach worked for me. Now I can render offscreen images without having to spawn a empty mj_viewer window. 

For anyone who is interested, here's what I did:
I'm using Ubuntu 16.04 and mujoco-py 2.0.2.5
After initializing the sim object
```
from mujoco_py import MjSim, MjViewer, load_model_from_path, MjRenderContext 

model = load_model_from_path(xml_path)
sim = MjSim(model)
```
If I want to render images without visualizing the simulation using MjViewer()
Previously I need to do this:
```
viewer = MjViewer(sim)
img = sim.render(w, h, camera_name, depth=True)
```
This will initialize an empty window for rendering in ''window' mode, otherwise sim.render() alone couldn't work.

Now after change the constructor of MjRenderContextOffscreen from cinit to init, I did this:
```
offscreen = MjRenderContext(sim=sim, device_id=0, offscreen=True, opengl_backend='glfw')
offscreen.render(100, 100, camera_id=-1)
img = offscreen.read_pixels(100, 100, depth=True)
```
It doesn't spawn any extra window, the whole rendering process is at the backend.