openai/mujoco-py #513: libOpenGL import error

kangtinglee, 2020-03-01:
Following a fresh install of mujoco from both pip and directly from cloning this repo, I always get this error on my first launch of mujoco. I have tried searching for similar problems but am unable to find any solutions. Has anyone encountered this error? How did you get around it?

```>>> import mujoco_py
Import error. Trying to rebuild mujoco_py.
running build_ext
building 'mujoco_py.cymj' extension
gcc -pthread -B /home/minerva/anaconda3/envs/pytorch/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Ianaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py -I/home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py -I/home/minerva/.mujoco/mujoco200/include -I/home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/numpy/core/include -I/home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py/vendor/egl -I/home/minerva/anaconda3/envs/pytorch/include/python3.6m -c /home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py/cymj.c -o /home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.9_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -B /home/minerva/anaconda3/envs/pytorch/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Ianaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py -I/home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py -I/home/minerva/.mujoco/mujoco200/include -I/home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/numpy/core/include -I/home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py/vendor/egl -I/home/minerva/anaconda3/envs/pytorch/include/python3.6m -c /home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py/gl/eglshim.c -o /home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.9_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py/gl/eglshim.o -fopenmp -w
gcc -pthread -shared -B /home/minerva/anaconda3/envs/pytorch/compiler_compat -L/home/minerva/anaconda3/envs/pytorch/lib -Wl,-rpath=/home/minerva/anaconda3/envs/pytorch/lib -Wl,--no-as-needed -Wl,--sysroot=/ /home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.9_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py/cymj.o /home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.9_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py/gl/eglshim.o -L/home/minerva/.mujoco/mujoco200/bin -Wl,-R/home/minerva/.mujoco/mujoco200/bin -lmujoco200 -lglewegl -o /home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.9_36_linuxgpuextensionbuilder/lib.linux-x86_64-3.6/mujoco_py/cymj.cpython-36m-x86_64-linux-gnu.so -fopenmp
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py/builder.py", line 510, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py/builder.py", line 106, in load_cython_ext
    mod = load_dynamic_ext('cymj', cext_so_path)
  File "/home/minerva/anaconda3/envs/pytorch/lib/python3.6/site-packages/mujoco_py/builder.py", line 125, in load_dynamic_ext
    return loader.load_module()
ImportError: /usr/lib/x86_64-linux-gnu/libOpenGL.so.0: undefined symbol: _glapi_tls_Current
```

divyanshj16, 2021-07-27:
I am stuck on this error, did you solve it?

djsamseng, 2021-11-13:
I was able to solve this when I had this error installing mujoco-py via 
```bash
$ pip3 install git+https://github.com/openai/mujoco-py`
```

## TLDR: Solution
```bash
$ sudo apt-get reinstall libglvnd0
```

## Explanation
```bash
$ pip3 install git+https://github.com/openai/mujoco-py
        ImportError: /usr/lib/x86_64-linux-gnu/libOpenGL.so.0: undefined symbol: _glapi_tls_Current
# Failed install - look into libOpenGL.so.0
$ ldd /usr/lib/x86_64-linux-gnu/libOpenGL.so.0
	linux-vdso.so.1 (0x00007ffe1bb68000)
	libGLdispatch.so.0 => /usr/lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007f1e06623000)
	libc.so.6 => /usr/lib/x86_64-linux-gnu/libc.so.6 (0x00007f1e06431000)
	libdl.so.2 => /usr/lib/x86_64-linux-gnu/libdl.so.2 (0x00007f1e0642b000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f1e06942000)
$ nm -D /usr/lib/x86_64-linux-gnu/libGLdispatch.so.0 | grep tls
# Look for the missing symbol - not present in libGLdispatch.so.0
$ sudo apt-get reinstall libglvnd0
# Reinstall libglvnd0 which provides libGLdispatch.so.0
$ nm -D /usr/lib/x86_64-linux-gnu/libGLdispatch.so.0 | grep tls
0000000000000000 D _glapi_tls_Current
$ pip3 install git+https://github.com/openai/mujoco-py
# Success!
```

PeppaCat, 2022-01-07:
How about on Ubuntu16.04, there's no such pkg.
```
E: Unable to locate package libglvnd0
```