openai/mujoco-py #516: Accessing geom segid for semantic segmentation?

danfeiX, 2020-03-17:
It looks like the segid property defined at https://github.com/openai/mujoco-py/blob/master/mujoco_py/generated/wrappers.pxi#L3530 is not exposed through the API. I would like to parse the semantic segmentation map returned from the renderer. To do that, I would need the segid for each geom. Is there a way I could access that? thanks!

ehu-ai, 2020-09-02:
After a lot of googling and trial, I have a solution for myself. Logging this for any future users who are interested in obtaining a segmentation map. 

1. Set the `segmentation=True` in the `render` function for the mujoco environment.

 2. This will return a (W, H, 2) array. I found a reference to this output in the deepmind continuous control suite camera [code.](https://github.com/deepmind/dm_control/blob/9e0fe0f0f9713a2a993ca78776529011d6c5fbeb/dm_control/mujoco/engine.py#L754)
> this is a (height, width, 2) int32 numpy array where the first channel contains the integer ID of the object at each pixel, and the second channel contains the corresponding object type (a value in the `mjtObj` enum). Background pixels are labeled (-1, -1).

However, this comment is **incorrect**. It seems like they switched the channel definitions, or I am just misunderstanding what object type and integer ID of object means. From my experimentation, it seems like the first channel contains the object type, and the second channel contains the object id.

3. Using the first channel mask, we can figure out what type of object it is (mesh, geom, body, etc.) by comparing the mask value with the mujoco enums. You can see the enum list [here](http://www.mujoco.org/book/APIreference.html#mjtObj). They are also accessible through mujoco_py.

4. Once you know what the object is, you can use id2name or the id to do what you want with the object. In my case, I wanted to color all the named objects with the prefix "robot" red, since I want to segment my robot.

Here is my example code for creating an image segmentation where the robot geoms are red.

```
from mujoco_py.generated import const

img = np.zeros(512, 512, 3)
seg = env.render(segmentation=True)
types = seg[:, :, 0]
ids = seg[:, :, 1]
geoms = types == const.OBJ_GEOM
geoms_ids = np.unique(obj_ids[geoms])
for i in geoms_ids:
    name = env.sim.model.geom_id2name(i)
    if "robot" in name:
        img[ids == i] = (255, 0, 0)

```

denmanorwatCDS, 2024-12-04:
In order to remove outlines of the table and what seems to be ball, it is also helpful to set 
```
<visual>
    	<quality offsamples="0"/>
</visual>
```
In your .xml file (must be child of <mujoco> tag, i.e.):
```
<mujoco>
	<visual>
    	<quality offsamples="0"/>
  	</visual>
...
</mujoco>
```