openai/mujoco-py #518: Fail to build wheel

gthd, 2020-03-25:
**Describe the bug**
Building wheel for mujoco-py error

**To Reproduce**
Install package that depends on mujoco-py

**Expected behavior**
Package installation succeeds, wheel can be built successfully for mujoco-py

**Error Messages**
  Building wheel for mujoco-py (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/george/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-6u4syqfx/mujoco-py/setup.py'"'"'; __file__='"'"'/tmp/pip-install-6u4syqfx/mujoco-py/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-hr2hy3w1
       cwd: /tmp/pip-install-6u4syqfx/mujoco-py/
  Complete output (20 lines):
  running bdist_wheel
  running build
  Removing old mujoco_py cext /tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/generated/cymj_2.0.2.2_37_linuxcpuextensionbuilder_37.so
  Compiling /tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/cymj.pyx because it depends on /home/george/lib/python3.7/site-packages/Cython/Includes/numpy/__init__.pxd.
  [1/1] Cythonizing /tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/cymj.pyx
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder
  creating /tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7
  creating /tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp
  creating /tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-6u4syqfx
  creating /tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-6u4syqfx/mujoco-py
  creating /tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py
  creating /tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/gl
  gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py -I/home/george/.mujoco/mujoco200/include -I/home/george/lib/python3.7/site-packages/numpy/core/include -I/home/george/include -I/usr/local/include/python3.7m -c /tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/cymj.o -fopenmp -w
  /tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/cymj.c:644:10: fatal error: mjmodel.h: No such file or directory
   #include "mjmodel.h"
            ^~~~~~~~~~~
  compilation terminated.
  error: command 'gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
Failed to build mujoco-py
Installing collected packages: mujoco-py, robosuite
    Running setup.py install for mujoco-py ... error
    ERROR: Command errored out with exit status 1:
     command: /home/george/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-6u4syqfx/mujoco-py/setup.py'"'"'; __file__='"'"'/tmp/pip-install-6u4syqfx/mujoco-py/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-ukj0t070/install-record.txt --single-version-externally-managed --compile --install-headers /home/george/include/site/python3.7/mujoco-py
         cwd: /tmp/pip-install-6u4syqfx/mujoco-py/
    Complete output (11 lines):
    running install
    running build
    Removing old mujoco_py cext /tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/generated/cymj_2.0.2.2_37_linuxcpuextensionbuilder_37.so
    running build_ext
    building 'mujoco_py.cymj' extension
    gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py -I/home/george/.mujoco/mujoco200/include -I/home/george/lib/python3.7/site-packages/numpy/core/include -I/home/george/include -I/usr/local/include/python3.7m -c /tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_37_linuxcpuextensionbuilder/temp.linux-x86_64-3.7/tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/cymj.o -fopenmp -w
    /tmp/pip-install-6u4syqfx/mujoco-py/mujoco_py/cymj.c:644:10: fatal error: mjmodel.h: No such file or directory
     #include "mjmodel.h"
              ^~~~~~~~~~~
    compilation terminated.
    error: command 'gcc' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/george/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-6u4syqfx/mujoco-py/setup.py'"'"'; __file__='"'"'/tmp/pip-install-6u4syqfx/mujoco-py/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-ukj0t070/install-record.txt --single-version-externally-managed --compile --install-headers /home/george/include/site/python3.7/mujoco-py Check the logs for full command output.

**Desktop (please complete the following information):**
 - OS: Ubuntu18.04
 - Python Version 3.7.7
 - Mujoco Version 200
 - mujoco-py version 2.00

**Environment**
 - output of: `echo $LD_LIBRARY_PATH` /usr/local/cuda-10.0/lib64:/home/george/Desktop/V-REP_PRO_EDU_V3_6_2_Ubuntu18_04:/home/george/.mujoco/mujoco200/bin
 - output of: `echo $HOME`  /home/george
 - output of: `echo $USER` george

SammyRossberg, 2020-03-26:
I had the same issue. Post to Reddit to see if they could help.

gthd, 2020-03-27:
Did you manage to solve it ? How ?

HailinRen, 2020-03-29:
I think they removed the mujoco 2.0 support....

realiti4, 2020-06-21:
I am having the same issue with 2.0.

ylan01, 2020-12-30:
> I had the same issue. Post to Reddit to see if they could help.

Can you send a link? thank you.