openai/mujoco-py #520: Depth conversion from OpenGL zbuffer

kracon7, 2020-03-26:
Hi,
I was trying to do 3d reconstruction within MuJoCo. I managed to get rgb and depth images by calling ```sim.render(width=w, height=h, camera_name=camera_name, depth=depth)```
As far as I have learnt about the depth rendering in MuJoCo, it uses OpenGL which convert real depth value into zbuffer. So the returned depth images have all values within [0,1]. 
 [See this link in MuJoCo forum](http://www.mujoco.org/forum/index.php?threads/ds525-or-kinect-camera-for-depth-information-in-mujoco-simulation.3718/#post-4771) 
[see the wikipage about the zbuffering](https://en.wikipedia.org/wiki/Z-buffering)
[image]

The near and far plane is defined by /visual/map/znear and zfar in the XML file. I tested a few senarios with prospective transformation and therefore a non-linear transformation. However, none of the converted real depth value is correct.
Could anyone tell me what is the correct conversion from the zbuffer to real depth value? Thanks a lot.

tkelestemur, 2020-03-27:
You can take a look at the dm_control's [implementation](https://github.com/deepmind/dm_control/blob/master/dm_control/mujoco/engine.py#L734) to convert depth values between [0, 1] to actual values.

kracon7, 2020-03-27:
@tkelestemur
Cool. I get it. My calculation on near and far plane was incorrect. 
Thanks a lot.

closed 2020-03-27 by kracon7

karanchahal, 2021-03-07:
Hello @kracon7 , can you tell me how to you calculated the values for near and far planes ?

geyang, 2022-09-22:
Just want to show: @tkelestemur 's link is pointing to this version: https://github.com/deepmind/dm_control/blob/409a0579c7f874f16d74ebfd7ef4cb5c959881b8/dm_control/mujoco/engine.py#L734

```python
    if depth:
      # Get the distances to the near and far clipping planes.
      extent = self._physics.model.stat.extent
      near = self._physics.model.vis.map_.znear * extent
      far = self._physics.model.vis.map_.zfar * extent
      # Convert from [0 1] to depth in meters, see links below:
      # http://stackoverflow.com/a/6657284/1461210
      # https://www.khronos.org/opengl/wiki/Depth_Buffer_Precision
      image = near / (1 - self._depth_buffer * (1 - near / far))
```