openai/mujoco-py #530: MjViewer recording shadows

a-z-e-r-i-l-a, 2020-04-19:
When I press "V" on keyboard to record a video of the simulation, a shadow screen is added to the gui which does not go away unless I press V again to stop recording. This shadow screen also gets recorded. Example:

Is there a way to prevent this?

[image]

stefanwanckel, 2021-05-07:
Are your surfaces really reflective? is your light source focal?

a-z-e-r-i-l-a, 2021-05-07:
> Are your surfaces really reflective? is your light source focal?

no only when I press V to start recording.

y8han, 2023-02-07:
Hi, I have solved this problem. The issue is that when you press V for recording, Mujuco start to overwrite the original screen resolution but somehow did not clear the original one. So you saw the shadow images in the video. The solution is pretty simple, you just have to comment out the overwriting codes:
In mjviewer.py, line 217 - line 220:
_# resolution = np.array(resolution)
# resolution = resolution * min(1000 / np.min(resolution), 1)
# resolution = resolution.astype(np.int32)
# resolution -= resolution % 16_
Hope this helps.