openai/mujoco-py #535: Apply external forces to the model.

daichi-3110, 2020-05-10:
Hello.

I want to apply external forces to the model.
Is there a way to apply an external force to the model?

kevincheng3, 2020-05-17:
add actuator/motor

justinberi, 2021-12-01:
For `xfrc_applied` where is the force applied?

justinberi, 2021-12-01:
To answer my own question. I believe it is applied at the position defined by `pos` in the `body` tag in the xml.

aknh9189, 2021-12-03:
Based on [this](https://roboti.us/forum/index.php?threads/applying-external-force-at-a-position.2199/), `xfrc_applied` is applied at the center of mass for the body.

fcycq, 2024-10-23:
If you want to apply force on object in a short time, just add it to data->xfrc_applied, this work on next step only, xfrc_applied() will apply the force forever.